ChristianTremblay/BAC0 #203: CRON service on rasp pi

bbartling, 2020-06-30:
Hello,

I apologize for posting this... But I was curious to ask if there would be anything in BAC0 running a scheduled task from CRON service on a rasp pi that would error out a script...?

If I run this from terminal works just fine... (default version of Python on the rasp pi == 3.7) but thru CRON service some exception happens where I use a `numpy.nan` if BAC0 exceptions...

Sorry alot code here.. But basically the CRON service runs this file every 15 minutes to perform a BACnet read request of a sensor and also grab some weather data from an API. Script also sends it to [this HTTP server I am experimenting](http://rcxreporter.pythonanywhere.com/) with as well as save to sqlite..

But what I am noticing thru the server (web URL above) is the `numpy.nan`'s are coming thru for the BACnet read request on the `meter_reading`.. On the server when there is a value shown for `metering_reading`, this is because I am running the file manually thru terminal..

Would anyone have any logging services to try ??

For some reason this portion of the code some exception happens when using CRON but not when running the file thru terminal...
```
try:
    elctricMeterReading = bacnet.read('201:2 analogValue 300 presentValue')
    print(f'elctricMeterReading {elctricMeterReading} kW')
    allStuff['meter_reading'] = elctricMeterReading

except:
    allStuff['meter_reading'] = numpy.nan
    print("issues with BACnet reading!!! NaN")

bacnet.disconnect()
print("BACnet disconnected!!")
```

This is the full script:

```
#!/usr/bin/env python3

import requests, json, time, BAC0, numpy
from datetime import datetime
import pandas as pd
from sqlalchemy import create_engine

bacnet = BAC0.lite()

api_key = "zzz"
base_url = "http://api.openweathermap.org/data/2.5/weather?"
city_name = "duluth"

allStuff = {}
stamp = time.time()

try:
    complete_url = base_url + "appid=" + api_key + "&q=" + city_name + "&units=imperial"
    response = requests.get(complete_url)

    # convert json format data into
    q = response.json()

    #json keys
    y = q["main"]
    w = q["wind"]
    c = q["clouds"]

    current_temperature = y["temp"]
    current_wind_speed = w["speed"]
    current_wind_deg = w["deg"]
    current_humidiy = y["humidity"]
    current_pressure = y["pressure"]
    current_cloud = c["all"]

    # store the value of "weather"
    # key in variable z
    z = q["weather"]

    # store the value corresponding
    # to the "description" key at
    # the 0th index of z
    weather_description = z[0]["description"]

    allStuff['Date'] = stamp
    allStuff['current_temperature'] = current_temperature
    allStuff['current_wind_speed'] = current_wind_speed
    allStuff['current_wind_deg'] = current_wind_deg
    allStuff['current_humidiy'] = current_humidiy
    allStuff['current_pressure'] = current_pressure
    allStuff['current_cloud'] = current_cloud
    allStuff['weather_description'] = weather_description

except:
    allStuff['Date'] = stamp
    allStuff['current_temperature'] = numpy.nan
    allStuff['current_wind_speed'] = numpy.nan
    allStuff['current_wind_deg'] = numpy.nan
    allStuff['current_humidiy'] = numpy.nan
    allStuff['current_pressure'] = numpy.nan
    allStuff['current_cloud'] = numpy.nan
    allStuff['weather_description'] = numpy.nan
    print("issues with weather API reading!!! NaN")

try:
    elctricMeterReading = bacnet.read('201:2 analogValue 300 presentValue')
    print(f'elctricMeterReading {elctricMeterReading} kW')
    allStuff['meter_reading'] = elctricMeterReading

except:
    allStuff['meter_reading'] = numpy.nan
    print("issues with BACnet reading!!! NaN")

bacnet.disconnect()
print("BACnet disconnected!!")

try:
    print("sending this dict",allStuff)
    #r = requests.post('https://httpbin.org/post', data=stuff, timeout=5)
    r = requests.post('http://rcxreporter.pythonanywhere.com/datainform', data=allStuff, timeout=5)
    print("Status code: ", r.status_code)
    print("Printing Entire Post Request")
    print(r.text)
except:
    print('Isses with sending data to cloud!')

try:
    allStuff['Date'] = datetime.fromtimestamp(allStuff['Date'])
    master_data = pd.DataFrame(allStuff,index=[0])
    master_data = master_data.set_index('Date')

    engine = create_engine('sqlite:///save_pandas_pi.db', echo=True)
    sqlite_connection = engine.connect()
    sqlite_table = "all_data"
    master_data.to_sql(sqlite_table, sqlite_connection, if_exists='append')
    sqlite_connection.close()
    print("Data saved to sql!")
except:
    print('Not able to package data to SQL!!!')
```

bbartling, 2020-06-30:
If I use this no issues... works just fine but I am a little worried its not as dependable as a CRON service I can tell the `time.sleep` drifts slightly.. But maybe its the way to go Vs CRON...? Any tips greatly appreciated...

This code below will run the .py file which is shown above...

```
import glob, os, time

try:
    while True:
        os.system(f'python3 weathBacnetCombo.py')
        print('run run run run!!! adrien run...')

        time.sleep(900)

except KeyboardInterrupt:
    pass

finally:
    print("Done deal...")
```

ChristianTremblay, 2020-07-06:
Maybe something related to the user executing the CRON job... and this user would use another Python version / site-packages

closed 2020-07-08 by bbartling