ChristianTremblay/BAC0 #204: BAC0.device and Japanese encoding

julesham, 2020-07-02:
Hi,

Thank you for this amazing library! 
We are currently testing it against some Seiko devices in Japan, and as such, the bacnet object names and descriptions are in Japanse. BAC0.device cannot create the point list, and I suspect it is because of the unknown encoding.

[image]

Is there a way to specify the encoding the use?

Kind regards,

Jules.

ChristianTremblay, 2020-07-02:
Nice ! Never thought my app would speak Japanese ! :-)
This issue will need to be addressed in bacpypes. Let me check...

ChristianTremblay, 2020-07-02:
IBM/Microsfot DBCS is not yet implemented in bacpypes. It could take some time as I can't tell if we can get access to controller using this encoding.

That said, do you think the controllers could be configured to use another encoding (like UTF-8) ?

julesham, 2020-07-02:
Hi Christian, thanks for your answer! We can't adapt the encoding of the controllers, but we could work around the issue because we have an object list with translated names. We've created the object list manually from that file, and used the English names. So for now it's not an issue anymore.

closed 2020-07-02 by julesham

JoelBender, 2020-07-02:
If there's a **pcap** file that has the read/response traffic in it so we can dig into which code page should be used (ANSI or OEM) and how the characters would map into unicode that would help. Including the mapping of a few test strings that you have done manually could be used for test cases would be excellent.

ChristianTremblay, 2020-07-02:
@julesham glad you found a workaround.
In the mean time, if you ever have the opportunity to record the pcap (using wireshark) we would have an idea of what's coming on the wire, and we may be able to support this encoding. And as my Japanese is as good as Google Translate :) We would need the translation sheet :)

That said, UTF-8 would be a better choice if possible.