ChristianTremblay/BAC0 #205: How to get vendorname

luli440, 2020-07-03:
The vendor name I scanned is an unknown vendor. How can I customize a value for it? Thank you very much
[image]

ChristianTremblay, 2020-07-06:
This looks like default bacpypes valus... not BAC0's one.
vendorName is an arguments when defining BAC0 instance.

For bacpypes, it's written in the .ini file I think

ChristianTremblay, 2020-08-03:
You can get a vendor name for free if you contact ASHREA. (http://www.bacnet.org/VendorID/)

A list can be found here.
http://www.bacnet.org/VendorID/BACnet%20Vendor%20IDs.htm

To read the vendorName, request is : 
```
bacnet.read('Address device deviceId vendorName')

closed 2020-08-03 by ChristianTremblay

luli440, 2020-08-12:
> 如果您与ASHREA联系，则可以免费获得供应商名称。（http://www.bacnet.org/VendorID/）
> 
> 列表可以在这里找到。
> http://www.bacnet.org/VendorID/BACnet%20Vendor%20IDs.htm
> 
> 要读取vendorName，请求为：
> 
> ```
> address = 2000:12 # let say it is the address
> deviceid = 5000 #let say this is the deviceID
> bacnet.read(f'{address} device {deviceId} vendorName')
> ```

Thank you very much！