ChristianTremblay/BAC0 #208: An option to define sampling rate when saving device history to a database.

kushan-26, 2020-07-08:
Current save option will save a database with resampling rate set "1s" by default. Even though polling period is not default (e.g. 10 minutes), save function will fill the same data for every second during 10 minute interval. 

It will be nice to define a sampling rate when saving a device to a database using ".save()". Also default sampling rate should be as same as polling rate because saving data every second for a long period often crash the BAC0; specially when it runs on RPi.

[image]

ChristianTremblay, 2020-07-08:
if you try develop branch, when calling `.save()` you can : 

`.save(resampling="10m")`
or
`.save(resampling=False)` # or 0 to disable resampling

kushan-26, 2020-07-08:
It did not work for me. I made sure time is synced too. Still getting data for every second and sporadically for milliseconds as well.
[image]

ChristianTremblay, 2020-07-09:
I should have not worked from memory :) "10m" = 10msec

"10Min"

https://pandas.pydata.org/pandas-docs/stable/user_guide/timeseries.html#resampling

kushan-26, 2020-07-09:
I tried "10Min", "5Min, "300s" and everything, I still get the same database export as above. Were there any changes made to scripts other than "sql.py"?

kushan-26, 2020-07-09:
I manually changed the resample rate in old "sql.py" file and this is the output I got.

I did sampling for "5Min". It appears to have done the sampling correct but not sure why the additional millisecond data is present in the file.
[image]

ChristianTremblay, 2020-07-10:
Ok I think I will have to work on that more carefully. It is not a feature I used very often and I typically just used the default resampling so I never realized this was not working correctly.

I'll dig into it and get back to you.

kushan-26, 2020-07-13:
Sounds good. Thanks you!

kushan-26, 2020-07-27:
I was able to pin point the reason why the resampling function has noisy data. The millisecond data is generated by the DO and DI points in a controller. My controllers had boolean points which indicated "active" or "inactive". These points were unable to resample by calculating the mean value. They should be handled separately to calculate the mean value.  A work around for this issue is to change the code to calculate max or min value for resampling.

"backup[point.properties.name] = point.history.resample('1s').**mean()**"

ChristianTremblay, 2020-07-28:
Normally, when pandas is available on the system, BAC0 will replace inactive and active by 0 and 1... and use .last to resample.

I suspect you don't have pandas installed, right ?
This would explain the issue as you probably fall in the else clause...

Please confirm

kushan-26, 2020-08-28:
I do have pandas installed and I am running BAC0 complete version. 

For some reason "inactive" and "active" is not getting replaced by 0 and 1. 

Just curious, do you get the expected results after resampling?

ChristianTremblay, 2020-08-30:
No... i don't
Just realized that I have bad indentations....

I'm going to rework all that. Thanks for pointing me this bug.

kushan-26, 2020-08-31:
Your welcome! Looking forward to tryout the improved script.

ChristianTremblay, 2020-09-02:
Can you have a look to develop branch ?
If running "complete", you can also have a look to the trend feature. I reworked it a lot.

kushan-26, 2020-10-19:
Sorry for the late reply. Was caught up with some other work. I did check the save function and it does seem to work correctly now. Let me do some more testing and confirm the changes. Also trend features as well.

kushan-26, 2020-10-21:
Christian- Where do I pass the 'resampling=False' argument? I tried passing it along with the save() function but gave me an error. Also what impact will it have?

ChristianTremblay, 2020-10-22:
This is still in the develop branch.... I'm really close to releasing a new version though.

```
def save(self, filename=None, resampling="1s"):
        """
        Save the point histories to sqlite3 database.  
        Save the device object properties to a pickle file so the device can be reloaded.

        Resampling : valid Pandas resampling frequency. If 0 or False, dataframe will not be resampled on save.
        """
```

kushan-26, 2020-10-23:
I came across couple of things -

1). I was checking the auto save feature and wondering if there is a way to change the resampling rate when we use auto_save option. 

2). I had "clear_history_on_save=True" when binding a device and noticed that device history doesn't get cleared upon saving the controller. This worked for the auto_save feature though.

Would these be possible to implement? Specially the 1st observation.

ChristianTremblay, 2020-10-24:
This is a reasonable request. I added this to the develop branch. Can you test ?

There is a new property named : save_resampling (defaulted to '1s')

Now, save function will call clear if required. Thanks for catching that.

I really need to work on tests....

kushan-26, 2020-10-26:
I get an error for the save function with the clear history call:

""Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/miniconda3/lib/python3.6/site-packages/BAC0/sql/sql.py", line 200, in save
    self.device.clear_histories()
AttributeError: 'RPMDeviceConnected' object has no attribute 'device'""

ChristianTremblay, 2020-10-26:
> File "", line 1, in

Ok thanks... I’ll correct that

kushan-26, 2020-10-26:
The new property "save_resampling" is making the polling stop after executing the save function for some reason. It doesn't really throw any error but simply stop the data polling. Any clue why this is happening?

ChristianTremblay, 2020-10-26:
Thanks for letting me know. Please tell if you find before me.

kushan-26, 2020-10-27:
> Thanks for letting me know. Please tell if you find before me.

Looks like the earlier typo was triggering this issue as well. I did bunch of testing and everything is working as it should be. I think all the issues have been addressed under this ticket. Thanks a lot for all your hard work!

ChristianTremblay, 2020-10-27:
What a relief... ! Was starting to think I was more breaking than making :-) 

(Probably the exception was breaking the thread)

closed 2020-10-27 by ChristianTremblay