ChristianTremblay/BAC0 #209: Write Event trigger

thamersalek, 2020-07-15:
Hi Christian,

First I would like it to thank you and (and Joel : Bacpypes for sure), for the magnificent BACnet library.
BAC0 made it too easy to implement a BACnet device with using less than 10 python statements, WOW that is really great, 
Please continue the good efforts, and I am ready to contribute to the evolution of this library, with the small experience that I do have ;)

Now, one question: I have noticed that when creating the device for the raspberry pi as a BACnet device (the one involved the LED activation : (https://github.com/JoelBender/bacpypes/issues/244).), you are creating a continuous loop that will pull for the value every some seconds, then update the led status based on active/inactive ! 
Why we need to make such loop?, isn’t there an event that got triggered when we receive a write to one of our points?
How can we have this event trigger in our code? , specifically the raspberry pi example code that you have provided in this link  (https://github.com/JoelBender/bacpypes/issues/244).

Also how to implement COV ?, which will make the code even super efficient.

Thanks

luli440, 2020-07-27:
How to use bac0 writing method? thanks

ChristianTremblay, 2020-07-27:
@thamersalek thanks for the good words. The example you refer to was only a "quick and dirty" hack serving as a proof of concept. You can do whatever you want... I never played with asyncio but this would open something cool.

Regarding COV, it's on my todo list. When I find time.

@luli440 : Please refer to the doc (https://bac0.readthedocs.io/en/latest/connect.html?highlight=write#read-and-write-using-the-bacnet-instance). If this is not sufficient, please open a new issue with your specific use case.

luli440, 2020-07-29:
BAC0 could not find the readwritescript() method
[image]

ChristianTremblay, 2020-07-29:
You seem to use an old version of the library. This docstring was an artifact. Now you should see this : 
```
 *Example*::

            import BAC0
            bacnet = BAC0.lite()
            bacnet.write('2:5 analogValue 1 presentValue 100 - 8')
```

ChristianTremblay, 2020-07-29:
ReadWriteScript is one of the first name of BAC0 :)

closed 2020-08-02 by ChristianTremblay

luli440, 2020-08-07:
OK, thank you very much