ChristianTremblay/BAC0 #212: rasp pi to determine pre cooling logic

bbartling, 2020-08-03:
Hi Christian,

Im curious if I could make a rasp pi into a BACnet device that would be online to the BAS 24/7 to determine if pre-cooling logic...

This script below will reach out to openWeatherAPI just to print the current weather conditions.. (Hoping to build this into retrieving the weatherforecast thru openWeatherAPI which 2 days hourly seems to be free..)

Would it be possible to keep this script alive where these boolean variables could be shared to the BAS?
`pre_cool_free` 
`pre_cool_mechanical` 

The idea would be to check current weather at 1AM (middle of the night) to check outdoor air humidity to determine if free cooling could be utilized or not... (Future would be to get weather forcast from openWeatherAPI)....

```
#!/usr/bin/env python3

import requests, json, time, BAC0, numpy
from datetime import datetime
import pandas as pd

bacnet = BAC0.lite()

'''
API KEYS
'''
api_key = "illEmailThisIfYouWant"
base_url = "http://api.openweathermap.org/data/2.5/weather?"
city_name = "duluth"

'''
SHARE THESE VALUES BACNET?!
'''
pre_cool_free = False
pre_cool_mechanical = False

'''
checkMonth is verfiy summertime months.
Logic is not needed in the winter!
'''
def checkMonth(month):
    val = 0
    if month in {5, 6, 7 ,8 , 9}:
        val = 1
    return val

currentMonth = datetime.now().month

try:
    complete_url = base_url + "appid=" + api_key + "&q=" + city_name + "&units=imperial"
    response = requests.get(complete_url)

    # convert json format data into
    q = response.json()

    #json keys
    y = q["main"]
    w = q["wind"]
    c = q["clouds"]

    current_temperature = y["temp"]
    current_wind_speed = w["speed"]
    current_wind_deg = w["deg"]
    current_humidiy = y["humidity"]
    current_pressure = y["pressure"]
    current_cloud = c["all"]

    # store the value of "weather"
    # key in variable z
    z = q["weather"]

    # store the value corresponding
    # to the "description" key at
    # the 0th index of z
    weather_description = z[0]["description"]

except:
    pass

print(current_temperature)

'''
if cool outside overnight in the middle of summer use free cooling to precool building.
60 °F and below will be == low humidity conditions outdoors
'''
try:
    if current_temperature < 60 and checkMonth(currentMonth) == 1:
        pre_cool_free = True

    else:
        pre_cool_free = False

except:
    pass

print(pre_cool_free)

bacnet.disconnect()
```

ChristianTremblay, 2020-08-03:
Why not ?
https://gist.github.com/ChristianTremblay/9bbd655775982730e62e538e44512771

ChristianTremblay, 2020-08-03:
It is a little outdated but the idea is there.

bbartling, 2020-08-03:
Wow cool. thanks ill have to play around with this....

bbartling, 2020-08-03:
This is pretty cool stuff thanks for the link. I knew i would get lost here pretty fast but I had to start blasting away some of the code in your Gist on the method that you were getting weather data... I definetely have something wrong in the update `def update():`

Would you have any tips? I get a `AttributeError: 'float' object has no attribute 'get_object_id'`. Im a little bit lost on how the weather data (float) gets mapped to the bacnet object, like a `get_object_id(("analogValue", 5))`

Got anymore examples to look at?

```
import requests, json, time, BAC0, numpy
from datetime import datetime
import pandas as pd

from bacpypes.basetypes import EngineeringUnits, DateTime
from bacpypes.primitivedata import CharacterString, Date, Time

from BAC0.core.devices.create_objects import (
    create_AV,
    create_MV,
    create_BV,
    create_AI,
    create_BI,
    create_AO,
    create_BO,
    create_CharStrValue,
    create_DateTimeValue,
)
from BAC0.tasks.RecurringTask import RecurringTask

'''
API KEYS
'''
api_key = "iCanEmailYouIfYouWant"
base_url = "http://api.openweathermap.org/data/2.5/weather?"
city_name = "duluth"

def start_device():
    print("Starting BACnet device")
    new_device = BAC0.lite(deviceId=10031)
    time.sleep(1)

    # Analog Values
    av = []
    current_temp = create_AV(
        oid=1, name="Current_Temp", pv=0, pv_writable=False)
    current_temp.units = EngineeringUnits("degreesFahrenheit")
    current_temp.description = CharacterString("Current Temperature in degC")
    av.append(current_temp)

    current_humidity = create_AV(
        oid=2, name="Current_Humidity", pv=0, pv_writable=False)
    current_humidity.units = EngineeringUnits("percentRelativeHumidity")
    current_humidity.description = CharacterString(
        "Current Humidity in percent relative humidity"
    )
    av.append(current_humidity)

    current_wind_deg = create_AV(
        oid=3, name="current_wind_deg", pv=0, pv_writable=False)
    current_wind_deg.units = EngineeringUnits("kilopascals")
    current_wind_deg.description = CharacterString("Current Pressure in kPa")
    av.append(current_wind_deg)

    current_wind_speed = create_AV(
        oid=4, name="Current_Wind_Speed", pv=0, pv_writable=False)
    current_wind_speed.units = EngineeringUnits("milesPerHour")
    current_wind_speed.description = CharacterString(
        "Current Wind Speed in mph"
        )
    av.append(current_wind_speed)

    current_cloud = create_AV(
        oid=5, name="current_cloud", pv=0, pv_writable=False
    )
    current_cloud.units = EngineeringUnits("percent")
    current_cloud.description = CharacterString("Current % Cloud Cover")
    av.append(current_cloud)

    # CharacterStringValues
    charstr = []
    default_pv = CharacterString("empty")
    location = create_CharStrValue(oid=1, name="Location", pv=default_pv)
    charstr.append(city_name)

    charstr.append(weather_description)
    current_conditions = create_CharStrValue(
        oid=3, name="Current_Conditions", pv=default_pv
    )

    # DateTimeValueObjects
    dtv = []
    updated = create_DateTimeValue(
        oid=1, name="Current_Conditions_Update", date=None, time=None
    )
    dtv.append(updated)

    for each in av:
        # print(each)
        new_device.this_application.add_object(each)
    for each in charstr:
        new_device.this_application.add_object(each)
    for each in dtv:
        new_device.this_application.add_object(each)
    return new_device

class getWeather:
    complete_url = base_url + "appid=" + api_key + "&q=" + city_name + "&units=imperial"

    response = requests.get(complete_url)

    # convert json format data into
    q = response.json()

    #json keys
    y = q["main"]
    w = q["wind"]
    c = q["clouds"]

    current_temp = y["temp"]
    current_wind_speed = w["speed"]
    current_wind_deg = w["deg"]
    current_humidity = y["humidity"]
    current_wind_deg = y["pressure"]
    current_cloud = c["all"]

    # store the value of "weather"
    # key in variable z
    z = q["weather"]

    # store the value corresponding
    # to the "description" key at
    # the 0th index of z
    weather_description = z[0]["description"]

r = getWeather()

def update():
    current_temp = r.current_temp.get_object_id(("analogValue", 1))
    current_wind_deg = r.current_wind_deg.get_object_id(("analogValue", 2))
    current_humidity = r.current_humidity.get_object_id(("analogValue", 3))
    current_wind_speed = r.current_wind_speed.get_object_id(("analogValue", 4))
    current_cloud = r.current_cloud.get_object_id(("analogValue", 5))
    location = city_name.get_object_id(("characterstringValue", 1))

def main():
    task_device = RecurringTask(update, delay=1800)
    task_device.start()
    while True:
        pass

if __name__ == "__main__":
    main()
```

bbartling, 2020-08-03:
In your fireplace wordpress blog #3 I can see the similar code **_Defining BACnet objects_**

bbartling, 2020-08-04:
Im looking back on some [old notes here](https://gist.github.com/ChristianTremblay/9bbd655775982730e62e538e44512771)... I think I can make something work..

If I can get this to work would you mind if I share here?... Just to see if there is a better way to write it...

bbartling, 2020-08-04:
Back to the drawing board.. This stuff is super cool btw  :^)

Any chance I can get a tip on how to link my outside air temperature & humidity to the BAC0 points I created? Im trying to get between the fire place blog post & other notes I have had on the way..

```
#!/usr/bin/env python

import BAC0, time, requests
from BAC0.core.utils.notes import note_and_log

from bacpypes.object import AnalogInputObject, BinaryInputObject, register_object_type
from bacpypes.basetypes import EngineeringUnits
from bacpypes.primitivedata import CharacterString
from BAC0.tasks.RecurringTask import RecurringTask

api_key = "6f1efece8acf334b0af1ec3538846065"
base_url = "http://api.openweathermap.org/data/2.5/weather?"

city_name = "duluth"
complete_url = base_url + "appid=" + api_key + "&q=" + city_name + "&units=imperial"
response = requests.get(complete_url)
r = response.json()

def changeValueOfMyAI():
     y = r["main"]
     w = r["wind"]
     c = r["clouds"]
     z = r["weather"]

     current_temperature = y["temp"]
     current_wind_speed = w["speed"]
     current_wind_deg = w["deg"]
     current_humidiy = y["humidity"]
     current_pressure = y["pressure"]
     current_cloud = c["all"]
     weather_description = z[0]["description"]
     print(current_temperature)

humidity_ai = AnalogInputObject(
objectIdentifier=("analogInput", 1),
objectName="HUMIDITY",
presentValue=0,
units=EngineeringUnits("percentRelativeHumidity"),
description=CharacterString("Web relative humidity outside"),
)
temperature_ai = AnalogInputObject(
objectIdentifier=("analogInput", 2),
objectName="TEMPERATURE",
presentValue=0,
units=EngineeringUnits("degreesFahrenheit"),
description=CharacterString("Web dry bulb outside temp"),
)

bacnet = BAC0.lite()
bacnet.this_application.add_object(humidity_ai)
bacnet.this_application.add_object(temperature_ai)

task = RecurringTask(changeValueOfMyAI,delay=30)
task.start()

try:
    while True:
        obj = bacnet.this_application.get_object_name('TEMPERATURE')
        print('Running and value is {}'.format(obj.presentValue))
        obj2 = bacnet.this_application.get_object_name('HUMIDITY')
        print('Running and value is {}'.format(obj.presentValue))
        time.sleep(300)

except KeyboardInterrupt:
    pass

finally:
    bacnet.disconnect()
    print("bacnet disconnected")
    print("Done deal...")
```

Im on an AX site today and I can see my laptop running BAC0 thru the JACE backnet discovery:
[image]

bbartling, 2020-08-04:
The `print(current_temperature)` works too.. For the free weatherAPI I can only make a few calls an hour. How can I update the code that the weather requests is updating every 15 minutes? Thanks for anytime you have in response :)

bbartling, 2020-08-04:
Is it always better practices to use:
```
if __name__ == "__main__":
    main()
```

Versus `while True` loop?
    `while True:
        obj = bacnet.this_application.get_object_name('TEMPERATURE')`

bbartling, 2020-08-04:
Ok! Sorry for the all posts!! Doing a bit of experimenting here... This is cool stuff to see your data come thru on the JACE :^()

Im hacking alot of code together from all I can find but I am having some issues on the `def update():` If I dont add in `while current_temp is not None:` this will error out `AttributeError: 'NoneType' object has no attribute 'presentValue'` AND i think its why only current temp comes thru on the JACE point discovery of BAC0. See screen shot below. The script seems print web temp & humidity just fine... Any ideas?

```
#!/usr/bin/env python

import BAC0, time, requests
from BAC0.core.utils.notes import note_and_log

from bacpypes.basetypes import EngineeringUnits, DateTime
from bacpypes.primitivedata import CharacterString, Date, Time, Real

from BAC0.core.devices.create_objects import (
    create_AV,
    create_MV,
    create_BV,
    create_AI,
    create_BI,
    create_AO,
    create_BO,
    create_CharStrValue,
    create_DateTimeValue,
)

from BAC0.tasks.RecurringTask import RecurringTask

import time

api_key = "6f1efece8acf334b0af1ec3538846065"
base_url = "http://api.openweathermap.org/data/2.5/weather?"

city_name = "duluth"
complete_url = base_url + "appid=" + api_key + "&q=" + city_name + "&units=imperial"
response = requests.get(complete_url)
r = response.json()

def start_device():
    print("Starting BACnet device")
    new_device = BAC0.lite()
    time.sleep(1)

    av = []
    current_humidity = create_AV(oid=0, name="Current_Humidity", pv=0, pv_writable=False)
    current_humidity.units = EngineeringUnits("percentRelativeHumidity")
    current_humidity.description = CharacterString("Current Humidity in percent relative humidity")
    av.append(current_humidity)

    current_temp = create_AV(oid=1, name="Current_Temp", pv=0, pv_writable=False)
    current_temp.units = EngineeringUnits("degreesFahrenheit")
    current_temp.description = CharacterString("Current Temperature in degF")
    av.append(current_temp)

    for each in av:
        print(each)

    new_device.this_application.add_object(each)
    return new_device

class App:
    dev = start_device()

app = App()

def update():
     y = r["main"]
     w = r["wind"]
     c = r["clouds"]
     z = r["weather"]

     web_current_temperature = y["temp"]
     web_current_wind_speed = w["speed"]
     web_current_wind_deg = w["deg"]
     web_current_humidity = y["humidity"]
     web_current_pressure = y["pressure"]
     web_current_cloud = c["all"]
     web_weather_description = z[0]["description"]
     print(web_current_temperature)
     print(web_current_humidity)

     current_humid = app.dev.this_application.get_object_name("Current_Humidity")

     current_temp = app.dev.this_application.get_object_name("Current_Temp")

     while current_temp is not None:
         current_temp.presentValue = Real(web_current_temperature)

     while current_humid is not None:
         current_humid.presentValue = Real(web_current_humidity)

def main():
     task_device = RecurringTask(update, delay=300)
     task_device.start()
     while True:
         pass

if __name__ == "__main__":
     main()
```

[image]

bbartling, 2020-08-04:
On the `current_temp.presentValue = Real(web_current_temperature)` does the `from bacpypes.primitivedata import Real` prevent the engineering units from coming thru?

bbartling, 2020-08-04:
Would it be best to get weather data on it's own task/thread? I may look into that... any tips greatly appreciated..

ChristianTremblay, 2020-08-05:
This is how I would do it

```
#!/usr/bin/env python

import BAC0, time, requests
from BAC0.core.utils.notes import note_and_log

from bacpypes.basetypes import EngineeringUnits, DateTime
from bacpypes.primitivedata import CharacterString, Date, Time, Real

from BAC0.core.devices.create_objects import (
    create_AV,
    create_MV,
    create_BV,
    create_AI,
    create_BI,
    create_AO,
    create_BO,
    create_CharStrValue,
    create_DateTimeValue,
)

from BAC0.tasks.RecurringTask import RecurringTask

import time
from meteo_parser import OpenWeather

def start_device():
    print("Starting BACnet device")
    new_device = BAC0.lite(ip='192.168.212.12/24')
    new_device._log.info('Device ID : {}'.format(new_device.Boid))
    time.sleep(1)

    av = []
    current_humidity = create_AV(
        oid=0, name="Current_Humidity", pv=0, pv_writable=False
    )
    current_humidity.description = CharacterString(
        "Current Humidity in percent relative humidity"
    )
    av.append(current_humidity)

    current_temp = create_AV(oid=1, name="Current_Temp", pv=0, pv_writable=False)
    current_temp.units = EngineeringUnits("degreesFahrenheit")
    current_temp.description = CharacterString("Current Temperature in degF")
    av.append(current_temp)

    for each in av:
        new_device.this_application.add_object(each)
    return new_device

class App:
    dev = start_device()
    weather = OpenWeather(city="duluth", units="imperial")

app = App()

def update():
    app.weather.update()
    current_humid = app.dev.this_application.get_object_name("Current_Humidity")
    current_temp = app.dev.this_application.get_object_name("Current_Temp")

    new_temp = app.weather.temp
    new_hum = app.weather.hum
    app.dev._log.info("Setting Temp to {}".format(new_temp))
    current_temp.presentValue = new_temp
    app.dev._log.info("Setting Humidity to {}".format(new_hum))
    current_humid.presentValue = new_hum

def main():
    task_device = RecurringTask(update, delay=900)
    task_device.start()
    while True:
        pass

if __name__ == "__main__":
    main()
```

For that, I like creating a specific class for the data acquisition side...
A very quick implementation would be something like : 
```
@note_and_log
class OpenWeather:

    blank = {
        "coord": {"lon": -92.11, "lat": 46.78},
        "weather": [
            {"id": 800, "main": "Unknown", "description": "Unknown", "icon": "01d"}
        ],
        "base": "stations",
        "main": {
            "temp": 0,
            "feels_like": 0,
            "temp_min": 0,
            "temp_max": 0,
            "pressure": 0,
            "humidity": 0,
        },
        "visibility": 0,
        "wind": {"speed": 0, "deg": 0},
        "clouds": {"all": 1},
        "dt": 0,
        "sys": {"type": 1, "id": 3903, "country": "US", "sunrise": 0, "sunset": 0},
        "timezone": 0,
        "id": 0,
        "name": "Unknown",
        "cod": 0,
    }

    def __init__(self, city="duluth", units="imperial"):
        self.base_url, self.api_key = self.get_api_key_and_base_url().values()
        self.city_name = city
        self.complete_url = (
            self.base_url
            + "appid="
            + self.api_key
            + "&q="
            + self.city_name
            + "&units="
            + units
        )
        self.data = None
        self.update()

    def update(self):
        backup = self.data
        try:
            response = requests.get(self.complete_url)
            self.data = response.json()
        except Exception as error:
            self._log.error("Error updating weather | {}".format(error))
            if self.data is None:
                self._log.warning("Using blank template")
                return self.blank
            self._log.warning("Using last time report")
            return backup
        return self.data

    def get_api_key_and_base_url(self):
        with open("api.key", "r") as file:
            d = json.load(file)
        return d["openweathermap"]

    @property
    def timestamp(self):
        return datetime.datetime.fromtimestamp(self.data["dt"])

    @property
    def sunrise(self):
        return datetime.datetime.fromtimestamp(self.data["sys"]["sunrise"])

    @property
    def sunset(self):
        return datetime.datetime.fromtimestamp(self.data["sys"]["sunset"])

    @property
    def temp(self):
        return self.data["main"]["temp"]

    @property
    def hum(self):
        return self.data["main"]["humidity"]
```

Then regarding the API key, a json file named api.key containing the informations that don't need to be in the script. Typically, this file would not be included in git

```
{"openweathermap": {"base_url": "http://api.openweathermap.org/data/2.5/weather?", "api_key": "secret_key"}}
```
[image]

This is opiniated so feel free to modify and change whatever you don't like.

You can also have a look to pyOWM which is a package that parse the data from Openweathermap... it may be something useful.

ChristianTremblay, 2020-08-05:
> Would it be best to get weather data on it's own task/thread? I may look into that... any tips greatly appreciated..

```
def main():
    task_device = RecurringTask(update, delay=900)
    task_device.start()
    while True:
        pass
```
The RecurringTask is done on another thread. The blocking process of updating is isolated from the BACnet object.

ChristianTremblay, 2020-08-05:
> Unsure why humidity isnt coming ...

bad indentation of `new_device.this_application.add_object(each)` 
This needs to be inside the for loop

ChristianTremblay, 2020-08-05:
Now let's turn this into a docker container and deploy it :-)

bbartling, 2020-08-05:
SUPER COOL! Im also super impressed with your software engineering skills wow I have a lot to learn 👍 

but I am having some issues...

I created the api.key json file ... And I think this is what I need to do create 2 .py files one of which is the `meteo_parser.py`

This my app.py file:
```
#!/usr/bin/env python

import BAC0, time, requests
from BAC0.core.utils.notes import note_and_log

from bacpypes.basetypes import EngineeringUnits, DateTime
from bacpypes.primitivedata import CharacterString, Date, Time, Real

from BAC0.core.devices.create_objects import (
    create_AV,
    create_MV,
    create_BV,
    create_AI,
    create_BI,
    create_AO,
    create_BO,
    create_CharStrValue,
    create_DateTimeValue,
)

from BAC0.tasks.RecurringTask import RecurringTask

import time
from meteo_parser import OpenWeather

def start_device():
    print("Starting BACnet device")
    new_device = BAC0.lite()
    new_device._log.info('Device ID : {}'.format(new_device.Boid))
    time.sleep(10)

    av = []
    current_humidity = create_AV(
        oid=0, name="Current_Humidity", pv=0, pv_writable=False
    )
    current_humidity.description = CharacterString(
        "Current Humidity in percent relative humidity"
    )
    av.append(current_humidity)

    current_temp = create_AV(oid=1, name="Current_Temp", pv=0, pv_writable=False)
    current_temp.units = EngineeringUnits("degreesFahrenheit")
    current_temp.description = CharacterString("Current Temperature in degF")
    av.append(current_temp)

    for each in av:
        new_device.this_application.add_object(each)
    return new_device

class App:
    dev = start_device()
    weather = OpenWeather(city="duluth", units="imperial")

app = App()

def update():
    app.weather.update()
    current_humid = app.dev.this_application.get_object_name("Current_Humidity")
    current_temp = app.dev.this_application.get_object_name("Current_Temp")

    new_temp = app.weather.temp
    new_hum = app.weather.hum
[... 110 lines cut ...]
2020-08-05 10:12:41,917 - INFO    | Using ip : 192.168.60.150
2020-08-05 10:12:41,919 - INFO    | Starting app...
2020-08-05 10:12:41,921 - INFO    | BAC0 started
2020-08-05 10:12:41,921 - INFO    | Registered as Simple BACnet/IP App
2020-08-05 10:12:41,924 - INFO    | Device ID : 3056396
2020-08-05 10:13:11,927 - ERROR   | Error updating weather | name 'requests' is not defined
2020-08-05 10:13:11,928 - WARNING | Using blank template
2020-08-05 10:13:11,929 - ERROR   | Error updating weather | name 'requests' is not defined
2020-08-05 10:13:11,940 - WARNING | Using blank template
Exception in thread recurring:
Traceback (most recent call last):
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\BAC0\tasks\TaskManager.py", line 59, in run
    self.process()
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\BAC0\tasks\TaskManager.py", line 65, in process
    self.task()
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\BAC0\tasks\RecurringTask.py", line 37, in task
    self.func()
  File "test5.py", line 65, in update
    new_temp = app.weather.temp
  File "C:\Users\benb\Desktop\bacTest\meteo_parser.py", line 81, in temp
    return self.data["main"]["temp"]
TypeError: 'NoneType' object is not subscriptable
```

bbartling, 2020-08-05:
The docker idea sounds super cool... Im also wanting to build this up to a point where thru the free openweather API, you can get hourly forecast weather data. 2 days worth of future hourly weather forecast which I am day dreaming about how this could get shared to the BAS thru BACnet ... Ultimately I think the BAS could use this for a determination in the middle of the night to do like a flush of the building pre-cool based on overnight outdoor dewpoint data. Thanks for the help Christian!!!!!!

bbartling, 2020-08-05:
GOT IT, yeah!!

[image]

I needed to import these packages on my meteo_parser.py file

```
from BAC0.core.utils.notes import note_and_log
import requests, json, time
from datetime import datetime
```

bbartling, 2020-08-05:
Am going to play around with this for a few days :) Thanks so much most likely have a ton more questions !

bbartling, 2020-08-05:
[image]

ChristianTremblay, 2020-08-07:
```
from ddcmath.enthalpy import enthalpy
```

and get rid of this dependency to Niagara :)

bbartling, 2020-08-09:
Cool I noticed this is another one of your creations :)

Can u do dewpoint? It seems like maintenance men can talk this language
versus explaining grains/moisture...

ChristianTremblay, 2020-08-09:
Done :) 
```
from ddcmath.psychrometric import dewpoint
```

bbartling, 2020-08-10:
Would you have any tips for sunset & sunrise times from the API? I have a simple version working (no forecast data or dewpoint calc) well with the exception this..

If I use a string the code will error out when I attempt to discover the BAC0 points... Its 
```
    current_sunrise = create_CharStrValue(
        oid=6, name="Sunrise", pv=default_pv
    )
    current_sunrise.description = CharacterString(
        "Sunrise time"
    )
```

Using datetime this will throw an exception too when starting the script to run.
```
    '''
    current_sunrise = create_DateTimeValue(
        oid=6, name="Current_Sunrise"
    )
    current_sunrise.description = CharacterString(
        "Sunrise time"
    )
```

TRACEBACK:
```
C:\Users\benb\Desktop\bacTest>py BasBotWeather_simple.py
Starting BACnet device
2020-08-10 11:09:03,169 - INFO    | Starting BAC0 version 20.02.20 (Lite)
2020-08-10 11:09:03,170 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2020-08-10 11:09:03,172 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2020-08-10 11:09:03,254 - INFO    | Using ip : 192.168.60.150
2020-08-10 11:09:03,255 - INFO    | Starting app...
2020-08-10 11:09:03,258 - INFO    | BAC0 started
2020-08-10 11:09:03,258 - INFO    | Registered as Simple BACnet/IP App
2020-08-10 11:09:03,261 - INFO    | Device ID : 3056682
2020-08-10 11:09:13,580 - INFO    | Setting Temp to 72.21
2020-08-10 11:09:13,586 - INFO    | Setting Humidity to 49
2020-08-10 11:09:13,592 - INFO    | Setting Wind Dir to 270
2020-08-10 11:09:13,597 - INFO    | Setting Wind Spd to 6.93
2020-08-10 11:09:13,602 - INFO    | Setting B Press to 1013
2020-08-10 11:09:13,608 - INFO    | Setting Cloud Cov to 1
2020-08-10 11:09:13,614 - INFO    | Setting Sunrise to 2020-08-10 06:00:33
Exception in thread recurring:
Traceback (most recent call last):
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\BAC0\tasks\TaskManager.py", line 59, in run
    self.process()
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\BAC0\tasks\TaskManager.py", line 65, in process
    self.task()
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\BAC0\tasks\RecurringTask.py", line 37, in task
    self.func()
  File "BasBotWeather_simple.py", line 210, in update
    current_sunrise.presentValue = new_sunrise
AttributeError: 'NoneType' object has no attribute 'presentValue'
```

On the JSON response from the API (on meteo_parser.py) passing this... ISO format (`time.time()` style format)
```
    @property
    def sunrise(self):
        return datetime.fromtimestamp(self.data["sys"]["sunrise"])
```

ChristianTremblay, 2020-08-10:
This will work
```
#!/usr/bin/env python

import BAC0, time, requests
from BAC0.core.utils.notes import note_and_log

from bacpypes.basetypes import EngineeringUnits, DateTime
from bacpypes.primitivedata import CharacterString, Date, Time, Real

from BAC0.core.devices.create_objects import (
    create_AV,
    create_MV,
    create_BV,
    create_AI,
    create_BI,
    create_AO,
    create_BO,
    create_CharStrValue,
    create_DateTimeValue,
)

from BAC0.tasks.RecurringTask import RecurringTask

import time
from meteo_parser import OpenWeather

from ddcmath.psychrometric import enthalpy, dewpoint

def start_device():
    print("Starting BACnet device")
    new_device = BAC0.lite(ip="192.168.212.12/24", )
    new_device._log.info("Device ID : {}".format(new_device.Boid))
    time.sleep(1)

    # AnalogValueObjects
    av = []
    current_humidity = create_AV(
        oid=0, name="Current_Humidity", pv=0, pv_writable=False
    )
    current_humidity.units = EngineeringUnits("percentRelativeHumidity")
    current_humidity.description = CharacterString(
        "Current Humidity in percent relative humidity"
    )
    av.append(current_humidity)

    current_temp = create_AV(oid=1, name="Current_Temp", pv=0, pv_writable=False)
    current_temp.units = EngineeringUnits("degreesFahrenheit")
    current_temp.description = CharacterString("Current Temperature in degF")
    av.append(current_temp)

    current_dewpoint = create_AV(oid=2, name="Current_Dewpoint", pv=0, pv_writable=False)
    current_dewpoint.units = EngineeringUnits("degreesFahrenheit")
    current_dewpoint.description = CharacterString("Current Dewpoint in degF")
    av.append(current_dewpoint)

    for each in av:
        new_device.this_application.add_object(each)

    # DateTimeValueObjects
    dtv = []
    sunrise = create_DateTimeValue(
        oid=1, name="Sunrise", date=None, time=None
    )
    dtv.append(sunrise)
    sunset = create_DateTimeValue(
        oid=2, name="Sunset", date=None, time=None
    )
    dtv.append(sunset)
    for each in dtv:
        new_device.this_application.add_object(each)
    return new_device

class App:
    dev = start_device()
    weather = OpenWeather(city="duluth", units="imperial")

app = App()

def update():
    app.weather.update()
    current_humid = app.dev.this_application.get_object_name("Current_Humidity")
    current_temp = app.dev.this_application.get_object_name("Current_Temp")
    current_dewpoint = app.dev.this_application.get_object_name("Current_Dewpoint")
    sunrise = app.dev.this_application.get_object_name("Sunrise")
    sunset = app.dev.this_application.get_object_name("Sunset")

    new_temp = app.weather.temp
    new_hum = app.weather.hum
    app.dev._log.info("Setting Temp to {}".format(new_temp))
    current_temp.presentValue = new_temp
    app.dev._log.info("Setting Humidity to {}".format(new_hum))
    current_humid.presentValue = new_hum
    new_dewpoint = dewpoint(temp=new_temp, hum=new_hum)
    app.dev._log.info("Setting Dewpoint to {}".format(new_dewpoint))
    current_dewpoint.presentValue = new_dewpoint

    sunrise.presentValue.date = Date(
        app.weather.sunrise.date().isoformat()
    )
    sunrise.presentValue.time = Time(
        app.weather.sunrise.time().isoformat()
    )

    sunset.presentValue.date = Date(
        app.weather.sunset.date().isoformat()
    )
    sunset.presentValue.time = Time(
        app.weather.sunset.time().isoformat()
    )

def main():
    task_device = RecurringTask(update, delay=900)
    task_device.start()
    while True:
        pass

if __name__ == "__main__":
    main()
```

In BAC0 if you want to read this new device, the Sunset and Sunrise will not be part of the points... but you can still read them with a `bacnet.read('address datetimeValue 1 presentValue').debug_contents()`

The Jace should see those points.

Important part is to add datetimeValue object "before" returning the `new_device`

ChristianTremblay, 2020-08-10:
[image]

bbartling, 2020-08-10:
Hey Christian,

I can get this to work... Any idea if its possible to get Last_Update_Time to show up? I think sunset & sunrise would be cool but Ill hold off for the time being... Along time ago when the weather service worked on Niagara AX we controlled parking lot lights with sunrise & sunset which it worked pretty good especially when day light hours change alot...

I know you mentioned time/Date variables maybe hard to pass up to the front end... But could it be a string variable so the BAS user would know that the data is updating or would you have any suggestions? I have even thought about creating a BACnet Boolean variable something like "Weather_Updating_Ok" any thoughts??

[image] 

Sorry alot of code here, `meoto_parser.py` also shown below too:

```
#!/usr/bin/env python

import BAC0, time, requests
from BAC0.core.utils.notes import note_and_log

from bacpypes.basetypes import EngineeringUnits, DateTime
from bacpypes.primitivedata import CharacterString, Date, Time, Real

from BAC0.core.devices.create_objects import (
    create_AV,
    create_MV,
    create_BV,
    create_AI,
    create_BI,
    create_AO,
    create_BO,
    create_CharStrValue,
    create_DateTimeValue,
)

from BAC0.tasks.RecurringTask import RecurringTask

import time
from meteo_parser import OpenWeather

def start_device():
    print("Starting BACnet device")
    new_device = BAC0.lite()
    new_device._log.info('Device ID : {}'.format(new_device.Boid))
    time.sleep(10)

    default_pv = 'blank'

    current_humidity = create_AV(
        oid=0, name="Current_Humidity", pv=0, pv_writable=False
    )
    current_humidity.units = EngineeringUnits("percent")
    current_humidity.description = CharacterString(
        "Current Humidity in percent relative humidity"
    )

    current_temp = create_AV(oid=1, name="Current_Temp", pv=0, pv_writable=False)
    current_temp.units = EngineeringUnits("degreesFahrenheit")
    current_temp.description = CharacterString("Current Temperature in degF")

    current_windspd = create_AV(
        oid=2, name="Current_Wind_Speed", pv=0, pv_writable=False
    )
    current_windspd.units = EngineeringUnits("milesPerHour")
    current_windspd.description = CharacterString(
        "Current Wind Speed"
    )

    current_winddir = create_AV(
        oid=3, name="Current_Wind_Dir", pv=0, pv_writable=False
    )
    current_winddir.units = EngineeringUnits("degreesAngular")
    current_winddir.description = CharacterString(
        "Current Wind Direction in degrees"
[... 225 lines cut ...]

    @property
    def press(self):
        return self.data["main"]["pressure"]

    @property
    def windspd(self):
        return self.data["wind"]["speed"]

    @property
    def winddir(self):
        return self.data["wind"]["deg"]

    @property
    def cloudcov(self):
        return self.data["clouds"]["all"]

    @property
    def descrip(self):
        return self.data["weather"][0]["description"]

    @property
    def city(self):
        return self.data["name"]
```

ChristianTremblay, 2020-08-10:
Let's make it a gist
Would this work ?
https://gist.github.com/ChristianTremblay/4b38a42cb212e8ef69d2c55d41c5bb8a

bbartling, 2020-08-10:
The code from the Gist is working... No time but date is coming thru just fine..

[image]

bbartling, 2020-08-10:
Im also going to work on converting this into a different API call from openweathermap... Its called "one call" where you can get current data, minute, hourly, and daily forecast too.. Its Alot more data that comes thru on the JSON response.. AND its still the free tier :)

But I think it would be cool to build some more BACnet points into this.. 

- 3HR_FUTURE_OAT
- 3HR_FUTURE_OAH
- 3HR_FUTURE_DEWPOINT (ddc math)

- 6HR_FUTURE_OAT
- 6HR_FUTURE_OAH
- 6HR_FUTURE_DEWPOINT (ddc math)

- 12HR_FUTURE_OAT
- 12HR_FUTURE_OAH
- 12HR_FUTURE_DEWPOINT (ddc math)

https://openweathermap.org/api/one-call-api

ChristianTremblay, 2020-08-10:
I made some tests here and I confirm Niagara BACnet driver is the problem for DateTime Values... if we want the time, we'll need to create a second object using TimeValueObject .... :-/ Tried different facets... nothing brings the time from the value.

A CharacterString would be possible... Any way, Niagara treats this as string.... maybe providing String Object for N4... and let DateTime for real BACnet devices.

Now make the city Writable via BACnet :)

bbartling, 2020-08-11:
Yes. Making City write-able would be super cool... It seems like the 'one call API' uses GPS coordinates... but I havent got very far with using one call yet..

Would you have any tips how I can troubleshoot this? For wind direction that comes thru as an angle. I am trying to change this to a string with a function in meteo_parser.py

```
    def getCardinalDirection(self, angle):
        directions = ['North', 'North East', 'East', 'South East', 'South', 'South West', 'West', 'North West']
        #directions = ['↑ North', '↗ North East', '→ East', '↘ South East', '↓ South', '↙ South West', '← West', '↖ North West']
        calc = directions[round(angle / 45) % 8]
        print(calc)
        return calc
```
That returns the data to main.py:
```
    @property
    def winddir(self):
        return getCardinalDirection(self.data["wind"]["deg"])
```

When I run the code I get a bacpypes error that I am doing something wrong. 
```
Traceback (most recent call last):
  File "main.py", line 130, in <module>
    class App:
  File "main.py", line 131, in App
    dev = start_device()
  File "main.py", line 59, in start_device
    oid=3, name="Current_Wind_Dir", pv=0, pv_writable=False
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\BAC0\core\devices\create_objects.py", line 131, in create_CharStrValue
    charval.presentValue = CharacterString(pv)
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\bacpypes-0.17.6-py3.7.egg\bacpypes\primitivedata.py", line 897, in __init__
    raise TypeError("invalid constructor datatype")
TypeError: invalid constructor datatype
```

On main.py I changed wind direction to a string...
```
    current_winddir = create_CharStrValue(
        oid=3, name="Current_Wind_Dir", pv=0, pv_writable=False
    )
    current_winddir.description = CharacterString(
        "Wind Direction String"
    )
```

But still throwing error :(

Here's a Gist for where I am at :)
https://gist.github.com/bbartling/a422688297f8ec9e8821ec1ecaef5bf3

ChristianTremblay, 2020-08-12:
pv=0
Try with something like : 
```
default_pv = CharacterString("empty")
current_winddir = create_CharStrValue(
        oid=3, name="Current_Wind_Dir", pv=default_pv , pv_writable=False
    )
```

bbartling, 2020-08-12:
would you have any tips for how I can incorporate this function that converts wind direction in angles to a string?

Here's the Gists:
https://gist.github.com/bbartling/a422688297f8ec9e8821ec1ecaef5bf3

On `meteo_parser.py` I am attempting this...

I created another function (or method) in the `class OpenWeather:`
```
    def getCardinalDirection(self, angle):
        directions = ['North', 'North East', 'East', 'South East', 'South', 'South West', 'West', 'North West']
        #directions = ['↑ North', '↗ North East', '→ East', '↘ South East', '↓ South', '↙ South West', '← West', '↖ North West']
        calc = directions[round(angle / 45) % 8]
        print(calc)
        return calc
```

And to call the function when returning the variable:
```
    @property
    def winddir(self):
        return getCardinalDirection(self.data["wind"]["deg"])
```

But when I run the code it errors out:
```
  File "main.py", line 157, in update
    new_winddir = app.weather.winddir
  File "C:\Users\benb\Desktop\bacTest\meteo_parser.py", line 107, in winddir
    return getCardinalDirection(self.data["wind"]["deg"])
NameError: name 'getCardinalDirection' is not defined
```

I think this is really a newbie question but would it be better to incorporate the `getCardinalDirection` function in `main.py`?

bbartling, 2020-08-12:
I think I figured this... If I just incorporate this into `meteo_parser.py`

```
    @property
    def winddir(self):
        #directions = ['North', 'North East', 'East', 'South East', 'South', 'South West', 'West', 'North West']
        directions = ['↑ North', '↗ North East', '→ East', '↘ South East', '↓ South', '↙ South West', '← West', '↖ North West']
        angle = self.data["wind"]["deg"]
        return directions[round(angle / 45) % 8]
```

But I think I need to only do the directions without the arrow symbols. Those appear to show up as question marks. Does BACnet support?

[image]

ChristianTremblay, 2020-08-12:
Even if UTF-8 is supported... I don’t think I would go on the emoji part ;-)

You know, keep it simple... do I really need an arrow in the string ? Not
really.
I will maybe animate a graphic on the value...

bbartling, 2020-08-13:
I have `dewpoint` added to meteo_parser.py  :) But not tested on a BACnet discovery of BAC0

```
    @property
    def dewpoint(self):
        return round(dewpoint(self.data["main"]["temp"], self.data["main"]["humidity"], SI=True))
```

[image]

[Updated Gists](https://gist.github.com/bbartling/a422688297f8ec9e8821ec1ecaef5bf3)

How the heck would you make the `City` a writeable BACnet point that would update the API? Sounds pretty cool...

ChristianTremblay, 2020-08-14:
```
current_location = create_CharStrValue(
        oid=8, name="Weather_Station_City", pv=default_pv, pv_writable=True
    )
    current_location.description = CharacterString(
        "Location of Weather Station"
    )
```

and pass this to update

```
    current_location = app.dev.this_application.get_object_name("Weather_Station_City")
    app.weather.update(current_location .presentValue)

    current_humid = app.dev.this_application.get_object_name("Current_Humidity")
    current_temp = app.dev.this_application.get_object_name("Current_Temp")
    current_winddir = app.dev.this_application.get_object_name("Current_Wind_Dir")
    current_windspd = app.dev.this_application.get_object_name("Current_Wind_Speed")
    current_pressure = app.dev.this_application.get_object_name("Current_Pressure")
    current_cloudcov = app.dev.this_application.get_object_name("Current_Cloud_Cover")

    last_update = app.dev.this_application.get_object_name("Last_Update")
    
    current_description = app.dev.this_application.get_object_name("Current_Weather_Description")
    current_dewpoint = app.dev.this_application.get_object_name("Current_Dewpoint")
```

In the parser, create a function that rebuild the complete url with the city coming from argument... and use it in the request.

You should be good.

ChristianTremblay, 2020-08-14:
Beware though... by talking to Joel, I understand that I should use the term "commandable" instead of pv_writable.... this will probably change in the future.

bbartling, 2020-08-18:
SO I have been playing around on `meteo_parser.py` where I pass the city name from `main.py`

Basically this checks to see if `cityWrite != self.city_name:` where I think I could try to different `requests` to openweathermap API

```
    def update(self, cityWrite):
        backup = self.data

        if cityWrite != self.city_name:
            print('cityWrite', cityWrite)
            print('self.city_name', self.city_name)

            try:
                response = requests.get(self.complete_url)
                #print(response.json())
                print(self.complete_url)
                self.data = response.json()
            except Exception as error:
                self._log.error("Error updating weather | {}".format(error))
                if self.data is None:
                    self._log.warning("Using blank template")
                    return self.blank
                self._log.warning("Using last time report")
                return backup
            return self.data

        else:
            print("City Changed!")

            try:
                response = requests.get(self.complete_url)
                #print(response.json())
                self.data = response.json()
            except Exception as error:
                self._log.error("Error updating weather | {}".format(error))
                if self.data is None:
                    self._log.warning("Using blank template")
                    return self.blank
                self._log.warning("Using last time report")
                return backup
            return self.data
```

The problem I am running into is the if statement to compare `cityWrite != self.city_name`

`printing cityWrite == CharacterString(0,X'656d707479')`
`printing self.city_name duluth`

They definitely will never equal.. Is this almost a Joel Bender question on how Bacpypes is running behind the scenes or should be running down a different path? Thanks for any tips Christian...

bbartling, 2020-08-18:
Update gists 
https://gist.github.com/bbartling/4f2afb0a49471e30f3d4f82759941055

bbartling, 2020-08-19:
Any idea if its possible to look up a character string? I think this would be something saved to memory but unsure..

bbartling, 2020-08-19:
So for this character string on main.py I can back out the words Chicago by converting to hex:

```
default_city = CharacterString("Chicago")

    current_location = create_CharStrValue(
        oid=8, name="Weather_Station_City", pv=default_city, pv_writable=True
    )
    current_location.description = CharacterString(
        "Location of Weather Station"
    )
```

If I print `current_location.presentValue` in lines on `main.py`
```
    app.weather.update(current_location.presentValue)
    print('current_location.presentValue', current_location.presentValue)
```
displays:
`current_location.presentValue CharacterString(0,X'4368696361676f')`

Where converting `4368696361676f` to hex == `Chicago'

Would you have any tips if I could slice `current_location.presentValue` to return just the hex?

Sorry to bug ya! Thanks for any tips if you have the time!

ChristianTremblay, 2020-08-19:
`variable.presentValue.value` ? will return your string.

bbartling, 2020-08-19:
Wow your amazing Christian that totally worked. No wonder why they pay you the big bucks ;)

ChristianTremblay, 2020-08-19:
Still waiting for the check.... or a beer at least

bbartling, 2020-08-19:
Lol.. Well at least getting over this hurdle the code appears to work but the writeable string goes fault when I attempt a write in workbench.

[image]

Updated Gists
https://gist.github.com/bbartling/4f2afb0a49471e30f3d4f82759941055

No rush on getting back to me this is fun side project work :)

The nuts and bolts of this I am attempting this `meteo_parser.py` which appears to be working...

```
    def update(self, cityWrite):
        backup = self.data

        if str(cityWrite) == str(self.city_name):

            try:
                response = requests.get(self.complete_url)

        else:
            print("City Changed!")

            try:
                new_url = self.alt_url + cityWrite + "&units=imperial"

```

bbartling, 2020-08-26:
Hi Christian Im going to close this for now. I have so many side projects it makes my head spin... Am going to work on a similar script as this that can also do the weather forecast. Thanks for your help. Maybe in the future I can redig into the BACnet writeable string :)

I also have some crazy ideas brewing to do a building electrical demand forecast thru BACnet too

closed 2020-08-26 by bbartling