ChristianTremblay/BAC0 #213: Default set as 'ReadableProperty'   - 'PresentValue' for objects - analogValue, MultiStateValue

snehabati, 2020-08-03:
Hi,
I am unable to write to property - 'presentValue' for objects -  analogValue, MultiStateValue.
The reason being these are set as 'ReadableProperty' in object.py file
Question here is I would like to know the reason why is that in object.py these properties are set as ReadableProperty by default.

1) class LargeAnalogValueObject(Object):
    objectType = 'largeAnalogValue'
    _object_supports_cov = True

    properties = \
        [ ReadableProperty('presentValue', Double)      <----------why is it set by default as Readable

2) class MultiStateValueObject(Object):
    objectType = 'multiStateValue'
    _object_supports_cov = True

    properties = \
        [ ReadableProperty('presentValue', Unsigned)     <----------why is it set by default as Readable

ChristianTremblay, 2020-08-03:
Unable to write to property in another device when using BAC0 as a "client" ?
Unable to write internally to local object inside BAC0 itself, when runing as a BACnet device with objects ?

snehabati, 2020-08-03:
@ChristianTremblay I am using bac0 as client and trying to write to my device which is just an object with properties defined as a dictionary...

ChristianTremblay, 2020-08-05:
I don't understand "properties defined as dictionary"

If you are writhing to a real BACnet device on the network, BAC0 will not use bacpypes properties. It will just create requests to write to them,
You may receive a writeAccessDenied from the device as not all AV, MV, BV are made writable by the different manufacturers.

snehabati, 2020-08-05:
I do not have a real device, just using the below config to simulate it as a device. Using BACnet IP, my below device config data is another PC1(having IP) and the client from which I want to write is in PC2(having another IP). 
From PC2 trying to write like this : 
backnet = BAC0.connect(ip=Client_IP, port=Client_port)
backnet.write(pc1_IP analogValue 0 presentValue 22.1)
backnet.write(pc1_IP multiStateValue 0 presentValue 15)
Error i get for both writes are same : NoResponseFromController: APDU Abort Reason : writeAccessDenied

dev_object_data={'Key': 8534,
                 'apduTimeout' : 3000,
                 'applicationSoftwareVersion' : '0.1.0',
                 'databaseRevision' : 0,
                 'description' : 'BACnet Smart Temperature Sensor Simulator (B-SS) , AC ''SB,2020',
                 'firmwareRevision' : '0.1.0',
                 'maxApduLengthAccepted' : 1476,
                 'modelName' : 'Room_FC_2020',
                 'numberOfApduRetries': 3,
                 'objectIdentifier': ('device', 8534),
                 'objectName': 'BACnet Smart Temperature Sensor Simulator (B-SS) ',
                 'protocolRevision': 14,
                 'protocolVersion': 1,
                 'segmentationSupported': 3,
                 'systemStatus': 0,
                 'vendorIdentifier': 614,
                 'vendorName': 'Sneha Bati, HM, licence, 2020'
                 }

obj_data={"AnalogOutputObject_0": {"Key": 0,
                                  "description": "BACnet Smart Temperature Sensor, ""B-SS.Indoor",
                                  "eventState": 4,
                                  "objectIdentifier": ("analogOutput", 0),
                                  "objectName": "BACnet Smart Temperature Sensor, ""B-SS.Indoor",
                                  "outOfService": True,
                                  "presentValue": 18.00,
                                  "reliability": 0,
                                  "statusFlags": [0]
                                  },
								  "AnalogValueObject_0": {"Key": 0,
                                  "description": "Motor Shaft Power",
                                  "eventState": 0,
                                  "objectIdentifier": ("analogValue", 0),
                                  "objectName": "Load (power)",
                                  "outOfService": True,
                                  "presentValue": 21,
                                  "reliability": 0,
                                  "statusFlags": [0],
                                  "units": 98
                                  },
								  "MultiStateValueObject_0": {"Key": 0,
                                  "description": "Controler State Command",
                                  "eventState": 0,
                                  "objectIdentifier": ("multiStateValue", 0),
                                  "objectName": "Rotor Speed",
                                  "outOfService": True,
                                  "presentValue": 2,
                                  "relinquishDefault": 2,
                                  "statusFlags": [0]},

snehabati, 2020-08-06:
> Unable to write to property in another device when using BAC0 as a "client" ?
> Unable to write internally to local object inside BAC0 itself, when runing as a BACnet device with objects ?

Yes the 1st point above is the issue I am facing, - Unable to write to property in another device when using BAC0 as a "client" ? but its not a real device. json file with objects and properties

ChristianTremblay, 2020-08-06:
Not sure I get how you create your fake device/points. Dicts from json ? 

Do you add them to another BAC0 instance application ?

If so, and you need them to be commandable, there is work to do so they get created with a priorityArray, a relinquishDefault and a writable presentValue. 

In its actual form, bacpypes allows *CmdObject in bacpypes.local.objects that you can use. I recently made a PR to add the possibility to use decorators to define commandable points but it is not yet accepted. 

Apologize if I'm slow understanding your use case.

snehabati, 2020-08-07:
Yes your understanding is right, To create a fake device we add the dicts from json to a BAC0 instance application, would you be able to help me with the needed changes I need to do so that I make the objects commandable for now ?

ChristianTremblay, 2020-08-07:
For now, you need to import (For example)
```
from bacpypes.local.object import BinaryOutputCmdObject
```

Those classes are standard classes to which are added mixins
ex.
```
class BinaryOutputCmdObject(Commandable(BinaryPV), MinOnOff, BinaryOutputObject):
    pass
```

My proposal to bacpypes would be something like that.

Proposal branch is here [https://github.com/JoelBender/bacpypes/tree/decorator_for_objects] 

```
properties = {
    "outOfService": False,
    "relinquishDefault": 0,
    "units": "degreesCelsius",
    "highLimit": 98,
}

@bacnet_properties(properties)
@commandable()
def av(instance, objectName, presentValue, description):
    OBJECT_TYPE = AnalogValueObject
    return create(OBJECT_TYPE, instance, objectName, presentValue, description)
```

You still need to create function to create the classes you need but I feel this would ease the process.
Note here I added highLimit... but as highLimit should generate an event (unreliable) if presentValue is higher than highLimit, we could work on something like : 

```
@bacnet_properties(properties)
@commandable()
@add_feature(event_limit, highLimit, lowLimit)
def av(instance, objectName, presentValue, description):
    OBJECT_TYPE = AnalogValueObject
    return create(OBJECT_TYPE, instance, objectName, presentValue, description)
```

snehabati, 2020-08-07:
ok got it, thanks much for the details.

closed 2020-08-13 by ChristianTremblay