ChristianTremblay/BAC0 #214: In BAC0-create_objects.py only few objects are imported from bacpypes objects.py

snehabati, 2020-08-06:
Hi Christian,
I am using BAC0 library in my client to read and write the object properties of the device.
I am already able to write and read to objects listed in BAC0->core>devices>create_objects.py as below.
from bacpypes.object import (
    MultiStateValueObject,
    AnalogValueObject,
    BinaryValueObject,
    AnalogInputObject,
    BinaryInputObject,
    AnalogOutputObject,
    BinaryOutputObject,
    CharacterStringValueObject,
    DateTimeValueObject,
    Property,
    register_object_type,
)
My next goal is to write to the other objects of a device which are not present in create_objects.py for example i want to read/write to  the Calendar Object, Program and Schedule objects. 
How will i be able to achieve this? Is there a update available from BAC0 to include the other objects from bacpyes or do I need to write on own. If I have to write on my own what are the things I should modify? Can you please help.

Thanks,
Sneha

ChristianTremblay, 2020-08-07:
I'm actually making proposal to bacpypes so the local device object creation becomes easier. The create_xyz functions will probably be deprecated. 

My goal is to provide some decorators that will help the configuration aspect of some features or the addition of properties. 

Having to create all the different functions to implement all BACnet objects is not optimal... 

I'd like to have access to easy way of declaring an object with a MinOnOff feature, have an event using highLimit or lowLimit for example. Actually, it is complicated. Same with the "commandable" aspect. A work in progress.

snehabati, 2020-08-07:
Sure, Thanks for the information. Would be able to share the code with me ?

ChristianTremblay, 2020-08-07:
See https://github.com/ChristianTremblay/BAC0/issues/213 for code

closed 2020-08-07 by ChristianTremblay