ChristianTremblay/BAC0 #216: Errorno 24 : Too many files open

San42, 2020-08-14:
I am using BAC0 in a codeto read a bunch of bacnet points.  I am reading those points every 1 min. Essentially I am using 3 lines of code involving BAC0:
``` 
while True:
 bac=BAC0.lite("192.168.xx.xx")
 value=bac.read(read_string)
 bac.disconnect()
```
It runs for sometime fine(on an average 2-3 days) and then stops with the following exception:
```
Aug 14 03:47:59 raspberrypi python3[720]:     raise InitializationError("Gros probleme : {}".format(error))
Aug 14 03:47:59 raspberrypi python3[720]: BAC0.core.io.IOExceptions.InitializationError: Gros probleme : Error starting app: [Errno 24] Too many open files
```
I have tried increasing the files limit via ulimit , both the soft and the hard limits, with no success. Any help would be highly appreciated! Thank you!

ChristianTremblay, 2020-08-17:
Due to the nature of BACnet, BAC0 is a device on the network. Constantly creating it, then disconnecting it goes against the idea of a device, which would constantly disappear from the network.
My recommandation is to start the instance once. Then run all your requests until you are sure the thing is not needed anymore.

```
bac=BAC0.lite("192.168.xx.xx")

while True:
    value=bac.read(read_string)
    # anything you do with value

bac.disconnect()

```

San42, 2020-08-19:
Thanks for the reply. The snippet of code I provided was an oversimplified version involving multiple scripts and functions. It would be a little difficult to make the changes as you have suggested. I will try it nonetheless and report if the same issue persists.

closed 2020-09-16 by ChristianTremblay