ChristianTremblay/BAC0 #217: Why can't ports be duplicated

luli440, 2020-08-17:
[image]

ChristianTremblay, 2020-08-17:
Because BAC0 is a BACnet device on the network and must respond to other device if required. So as long as the instance is running, it captures the socket and listen for any request that come com from the network.

If you need more than one device per IP, you can use another port (47809 for example) or create another IP for the same network interface card.

luli440, 2020-08-19:
Thank you very mush，There is one more question to ask
Is it possible to read different COM ports circularly and forward the same IP and port？

closed 2020-11-04 by ChristianTremblay