ChristianTremblay/BAC0 #218: main.py perform a read request

bbartling, 2020-08-28:
Hey Christian,

Im experimenting around with a similar script that performs a `request` like meteo_parser.py

On my main.py file, im getting hung up on trying to perform a BACnet read request in the `update` function.. Maybe this is an issue of threading?

```
  File "main2.py", line 84, in update
    temp = new_device.read('110400:11 analogInput 1 presentValue')
NameError: name 'new_device' is not defined
```

Even with attempting to share the a global to the update function. 
```
def update():
    global new_device
```

Im using a function to 'build' the BACnet device same as the weather script:

```
#!/usr/bin/env python

import BAC0, time, requests
from BAC0.core.utils.notes import note_and_log

from bacpypes.basetypes import EngineeringUnits, DateTime
from bacpypes.primitivedata import CharacterString, Date, Time, Real

from BAC0.core.devices.create_objects import (
    create_AV,
    create_MV,
    create_BV,
    create_AI,
    create_BI,
    create_AO,
    create_BO,
    create_CharStrValue,
    create_DateTimeValue,
)

from BAC0.tasks.RecurringTask import RecurringTask

import time
from send_data import Brain

def start_device():
    print("Starting BACnet device")
    #new_device = BAC0.lite(ip='192.168.67.2/16', deviceId='123423')#falls school
    new_device = BAC0.lite(ip='192.168.254.253/16', deviceId='123423')#st thomas
    new_device = BAC0.lite()
    new_device._log.info('Device ID : {}'.format(new_device.Boid))
    time.sleep(10)

    default_pv = CharacterString("empty")

    demand_responce = create_BV(
        oid=0, name="Demand_Responce", pv=0, pv_writable=False
    )
    demand_responce.description = CharacterString(
        "Boolean demand respone event"
    )

    last_update = create_DateTimeValue(
        oid=1, name="Last_Update"
    )
    last_update.description = CharacterString(
        "Last update timestamp"
    )

    temp_data = create_AV(
        oid=2, name="Temp_Data", pv=0, pv_writable=True
    )
    temp_data.units = EngineeringUnits("degreesFahrenheit")
    temp_data.description = CharacterString(
        "Writeable point to send to cloud"
    )

    new_device.this_application.add_object(demand_responce)
    new_device.this_application.add_object(last_update)
    new_device.this_application.add_object(temp_data)

    return new_device

class App:
    dev = start_device()
    brain = Brain()

app = App()

def update():
    global new_device

    temp_data = app.dev.this_application.get_object_name("Temp_Data")
    demand_responce = app.dev.this_application.get_object_name("Demand_Responce")
    last_update = app.dev.this_application.get_object_name("Last_Update")

    temp = new_device.read('110400:11 analogInput 1 presentValue')
    app.dev._log.info("Bacnet read request of temp is {}".format(temp))
    app.brain.update(temp)
    app.dev._log.info("Attempting to send this value to the brain!")

    new_demand_responce = app.brain.demand_responce
    new_update_date = app.brain.update

    last_update.presentValue.date = Date(
    app.brain.timestamp.date().isoformat()
    )

    app.dev._log.info("Setting demand_responce to {}".format(new_demand_responce))
    demand_responce.presentValue = new_demand_responce

def main():
    task_device = RecurringTask(update, delay=900)
    task_device.start()
    while True:
        pass

if __name__ == "__main__":
    main()

```

Any tips to try greatly appreciated...

ChristianTremblay, 2020-08-28:
Here you must use

app.dev

The variable app is the class App that intanciate the new_device and make
it available with the dev property.

bbartling, 2020-08-28:
Ah thanks

`temp = app.dev.read('110400:11 analogInput 1 presentValue')`

This would be device instance 110400 on mstp network 11 right?

bbartling, 2020-08-28:
Got it thanks for the help !

closed 2020-08-28 by bbartling

ChristianTremblay, 2020-08-28:
> Ah thanks
> 
> `temp = app.dev.read('110400:11 analogInput 1 presentValue')`
> 
> This would be device instance 110400 on mstp network 11 right?

Reverse
network:mac