ChristianTremblay/BAC0 #219: problem with polling even when segmentation_supported=False

ionelpetrus, 2020-08-31:
Hi Christian,
I am getting the error below, even when I setup _segmentation_supported=False_
Thoughts? How can I setup polling on this device if it does not support segmentation?
Thank you in advance for all your help.
network_2_analog_input_module = BAC0.device('2:3', 12345, bacnet, segmentation_supported=False)
2020-08-30 21:17:39,851 - INFO    | Changing device state to DeviceDisconnected'>
2020-08-30 21:17:40,119 - INFO    | Changing device state to RPDeviceConnected'>
2020-08-30 21:17:40,554 - INFO    | Device 12345:[Analog_Input] found... building points list
2020-08-30 21:18:14,501 - INFO    | Ready!
2020-08-30 21:18:14,502 - INFO    | Device defined for normal polling with a delay of 10sec
2020-08-30 21:18:14,504 - INFO    | Polling started, values read every 10 seconds
2020-08-30 21:18:15,993 - ERROR   | Something is wrong with polling...stopping. Try setting off segmentation

ChristianTremblay, 2020-08-31:
It is not always segmentation the problem and some implementation of devices in the field may be hard to pin point.
I would try : 

```
network_2_analog_input_module = BAC0.device('2:3', 12345, bacnet, segmentation_supported=False, poll=0)
# then I would choose specific point to poll
network_2_analog_input_module['input1'].poll(10)
```

ionelpetrus, 2020-09-03:
Thank you! this was helpful! Much appreciated.

closed 2020-09-16 by ChristianTremblay