ChristianTremblay/BAC0 #220: Error handling for "Unknown Property" in Read.py -leads to valid read result

hehenkel9, 2020-09-14:
For conformity testing i tried to read  all properties 0...511 and encountered strange things with some properties.
I got valid values instead of error.

after modifying Read.py like this, the problems were gone

        if iocb.ioError:  # unsuccessful: error/reject/abort
            apdu = iocb.ioError
            reason = find_reason(apdu)
            if reason == "segmentationNotSupported":
                value = self._split_the_read_request(args, arr_index)
                return value
            else:
                if reason == "unknownProperty":
                    #if "description" in args:
                    #    return "Not Implemented"
                    #elif "inactiveText" in args:
                    #    return "Off"
                    #elif "activeText" in args:
                    #    return "On"
                    #else:
                    raise UnknownPropertyError("Unknown property {}".format(args))
                elif reason == "unknownObject":
                    self._log.warning("Unknown object {}".format(args))
                    raise UnknownObjectError("Unknown object {}".format(args))
                else:
                    # Other error... consider NoResponseFromController (65)
                    # even if the real reason is another one
                    raise NoResponseFromController(
                        "APDU Abort Reason : {}".format(reason)
                    )

ChristianTremblay, 2020-09-14:
Thanks for letting me know. You are right I should not return value in that case.

ChristianTremblay, 2020-09-14:
Now that I think about it, I remember why it's done this way.
Some controller do not implement a description. BAC0 wants a description to build its version of point. I was intercepting this error a generating local values for those properties. Same with activeText and inactiveText.
So if I read a description on a point for which the property is non-existent, I'll get "Not Implemented"

What is your use case ? I understand you would need the exception to be raised

This would have a side effect of breaking the creation of `BAC0.device` when one of those properties are not implemented by the device.

I guess a flag could be implemented to prevent this for some "niche" application

hehenkel9, 2020-09-15:
The current use case is BACnet functional & compliance testing . For compliance test, you have to check that all properties are readable / writable as stated in the EPICS. Therefore I implemented a detailed test and encountered that e.g. analog inputs returned a value for active text as well as inactive text instead of "write access denied" when trying to write to these properties.

hehenkel9, 2020-09-15:
missed the right button

ChristianTremblay, 2020-09-15:
I see. This is an interesting use case. 

I never tried to write to a property not implemented. 

I will have to look at this. 
Also, I'm open to PR if you have an idea of implementation. 

Or if you can write tests ;-)

closed 2021-02-13 by github-actions[bot]