ChristianTremblay/BAC0 #223: Write property multiple only sends request for last element in args list

RachelErin, 2020-10-19:
build_wpm_request resets was[] with each iteration and when the request is created, only sends the last item in the list. In addition, if you move was[] reset to outside the loop, multiple requests for writing to the same object create new object tags for each property request instead of creating a property tag under the existing object. I have included code below of build_wpm_request that I modified to fix both of these issues.

```
def build_wpm_request(self, args, vendor_id=0, addr=None):
        if not addr:
            raise ValueError("Please provide addr")

        self.log_subtitle("Creating Write Multiple Request")
        self._log.debug(
            "{:<20} {:<20} {:<20} {:<20}".format(
                "indx", "priority", "datatype", "value"
            )
        )

        was = []
        for each in args:
            property_values = []
            if isinstance(each, str):
                obj_type, obj_inst, prop_id, value, priority, indx = self._parse_wp_args(
                    each
                )
            elif isinstance(each, tuple):
                # Supported but not really the best choice as the parser will
                # catch some edge cases for you
                obj_type, obj_inst, prop_id, value, priority, indx = each
            else:
                raise ValueError("Wrong encoding of request")
            value = self._validate_value_vs_datatype(
                obj_type, prop_id, indx, vendor_id, value
            )

            existingObject = next((obj for obj in was if obj.objectIdentifier == (obj_type, obj_inst)), None)

            if existingObject == None:
                property_values.append(
                    PropertyValue(
                        propertyIdentifier=prop_id,
                        propertyArrayIndex=indx,
                        value=value,
                        priority=priority,
                    )
                )

                was.append(
                    WriteAccessSpecification(
                        objectIdentifier=(obj_type, obj_inst),
                        listOfProperties=property_values,
                    )
                )
            else:
                existingObject.listOfProperties.append(PropertyValue(
                        propertyIdentifier=prop_id,
                        propertyArrayIndex=indx,
                        value=value,
                        priority=priority,
                    ))

            datatype = get_datatype(obj_type, prop_id, vendor_id=vendor_id)

            self._log.debug(
                "{!r:<20} {!r:<20} {!r:<20} {!r:<20}".format(
                    indx, priority, datatype, value
                )
            )

        # build a request
        request = WritePropertyMultipleRequest(listOfWriteAccessSpecs=was)
        request.pduDestination = Address(addr)

        self._log.debug("{:<20} {}".format("REQUEST", request))
        return request
```

ChristianTremblay, 2020-10-19:
Thank you very much for taking the time to fix my errors !

I will review that ASAP and this will merge with develop.

This is not a feature I use so chances are that I would never have seen the
flaw.

ChristianTremblay, 2020-10-20:
Merged https://github.com/ChristianTremblay/BAC0/pull/224

closed 2020-10-22 by ChristianTremblay