ChristianTremblay/BAC0 #225: Understanding breaking change from 19.9.9 to 20.2.20

iamliamc, 2020-10-22:
Hey Chrstian/Maintainers... 

I had some code that was happily writing Bacnet weeklySchedule fields until I updated to 20.2.20: 

My little bit of custom wrapping turned: 

```
[{'daySchedule': [{'time': datetime.time(3, 0), 'value': 1}, {'time': datetime.time(19, 0), 'value': 0}]}, {'daySchedule': [{'time': datetime.time(4, 0), 'value': 1}, {'time': datetime.time(19, 0), 'value': 0}]}, {'daySchedule': [{'time': datetime.time(4, 0), 'value': 1}, {'time': datetime.time(19, 0), 'value': 0}]}, {'daySchedule': [{'time': datetime.time(4, 0), 'value': 1}, {'time': datetime.time(19, 0), 'value': 0}]}, {'daySchedule': [{'time': datetime.time(4, 0), 'value': 1}, {'time': datetime.time(19, 0), 'value': 0}]}, {'daySchedule': []}, {'daySchedule': []}]
```
into 

```
>>> my_schedule.dict_contents()
[{'daySchedule': [{'time': (3, 0, 0, 0), 'value': <bacpypes.primitivedata.Enumerated object at 0x108f0a470>}, {'time': (19, 0, 0, 0), 'value': <bacpypes.primitivedata.Enumerated object at 0x108f0a4e0>}]}, {'daySchedule': [{'time': (4, 0, 0, 0), 'value': <bacpypes.primitivedata.Enumerated object at 0x108f0a588>}, {'time': (19, 0, 0, 0), 'value': <bacpypes.primitivedata.Enumerated object at 0x108f0a5f8>}]}, {'daySchedule': [{'time': (4, 0, 0, 0), 'value': <bacpypes.primitivedata.Enumerated object at 0x108f0a6a0>}, {'time': (19, 0, 0, 0), 'value': <bacpypes.primitivedata.Enumerated object at 0x108f0a710>}]}, {'daySchedule': [{'time': (4, 0, 0, 0), 'value': <bacpypes.primitivedata.Enumerated object at 0x108f0a7b8>}, {'time': (19, 0, 0, 0), 'value': <bacpypes.primitivedata.Enumerated object at 0x108f0a828>}]}, {'daySchedule': [{'time': (4, 0, 0, 0), 'value': <bacpypes.primitivedata.Enumerated object at 0x108f0a8d0>}, {'time': (19, 0, 0, 0), 'value': <bacpypes.primitivedata.Enumerated object at 0x108f0a940>}]}, {'daySchedule': []}, {'daySchedule': []}]
>>> type(my_schedule)
<class 'bacpypes.constructeddata.ArrayOf.<locals>.ArrayOf'>
>>> my_schedule.__name__
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'ArrayOfDailySchedule' object has no attribute '__name__'
>>> my_schedule.subtype
<class 'bacpypes.basetypes.DailySchedule'>
```

Here is the custom wrapping... that utilizes bacpypes.basetypes and bacypypes.constructedata v0.18.3
```
    def dict_to_weeklyschedule(self, val):
        return ArrayOf(DailySchedule)([ self.dict_to_dailyschedule(x) for x in val])

    def dict_to_dailyschedule(self, val):
        return DailySchedule(
            daySchedule=[self.dict_to_timevalue(x) for x in val['daySchedule']]
        )

    def dict_to_timevalue(self, val):
        return TimeValue(
            time=self.to_time_tuple(val['time']),
            value=self.to_enumerated(val['value'])
        )

    def to_time_tuple(self, val):
        try:
            if isinstance(val, dt.time):
                return Time(hour=val.hour, minute=val.minute, second=val.second, hundredth=val.microsecond).value
            elif isinstance(val, str):
                return self.to_time_tuple(dt.datetime.strptime(val, '%H:%M:%S').time())
            else: 
                raise Exception("Unsupported time format encountered!")
        except Exception as e:
            raise Exception("Error during parsing time format to bacnet!")
```

When I updated to 20.2.20 now I get this error:
https://github.com/ChristianTremblay/BAC0/blob/f8232653928e78916ea1d40cc31184847f73c589/BAC0/core/io/Write.py#L169
```
Traceback (most recent call last):
  File "/home/debian/bacnet_read_publisher/src/app/schedules/action_executor.py", line 47, in run
    executed_at = await self.execute_action()
  File "/home/debian/bacnet_read_publisher/src/app/schedules/action_executor.py", line 168, in execute_action
    self.action_value
  File "/home/debian/bacnet_read_publisher/src/bacnet_interface/bacnet_controller.py", line 133, in write
    await self.loop.run_in_executor(None, self.bacnet_instance.write, request)
  File "/usr/lib/python3.5/asyncio/futures.py", line 380, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 304, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
  File "/usr/lib/python3.5/concurrent/futures/thread.py", line 55, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/debian/.local/lib/python3.5/site-packages/BAC0/core/io/Write.py", line 88, in write
    iocb = IOCB(self.build_wp_request(args, vendor_id=vendor_id))
  File "/home/debian/.local/lib/python3.5/site-packages/BAC0/core/io/Write.py", line 198, in build_wp_request
    obj_type, prop_id, indx, vendor_id, value
  File "/home/debian/.local/lib/python3.5/site-packages/BAC0/core/io/Write.py", line 178, in _validate_value_vs_datatype
    "invalid result datatype, expecting {}".format((datatype.__name__,))
TypeError: invalid result datatype, expecting ('ArrayOfDailySchedule',)
```

If I roll back to 19.9.9 everything works fine... what am I missing here?

ChristianTremblay, 2020-10-26:
Interesting. I will have a look. I never tried to write schedules and I never experienced any issue of this nature by changing version.

ChristianTremblay, 2020-10-26:
Can you tell if bacpypes version changed ?

ChristianTremblay, 2020-10-27:
Ok, I see. In version 19.9.9 there were no special function to validate the datatype. The function added since do not support this specific datatype. Let me check if I can fix it quick.

ChristianTremblay, 2020-10-27:
Can you share more of your code ? I miss the part where you write to the schedule... do you use arrays to write one day at a time ? What do you send over the wire ?

iamliamc, 2020-10-27:
Hey Christian, 

Yeah let's see if I can provide that context for you without having to unwind the whole stack over here... 

In 19.9.9 we `write` the whole: Array: `<class 'bacpypes.constructeddata.ArrayOf.<locals>.ArrayOf'>` of `<class 'bacpypes.basetypes.DailySchedule'>` at once... 

So we pass in the unique identifying information of the bacnet register, 
the "field" in this case 'weeklySchedule' 
and the ArrayOf ~ DailySchedules

So i've broken out the actual control flow cause I think it obfuscates what we are actually working on here: 

This is how we instantiate BAC0
```
self.bacnet_instance = BAC0.lite(self.bacnet_ip)
```

In this case the stream looks like: 
`{'identifier': 'our-in-house-identifier', 'units': '', 'network': '2:0x010000000000', 'mac': '1', 'inUse': '', 'id': '18', 'type': 'binaryInput'}`

```
        request = bacpypes_request_builder().build_write_request(
            bacnet_stream,
            field,
            value
        )
```

The validate_parameters here is where our write values get marshaled into bacpypes objects... 

```
    def build_write_request(self, bacnet_stream, field, value):
        parsed_value = self.validate_parameters(bacnet_stream, field, value)

        parameters = bac0_wrapper(
            self.build_address(bacnet_stream),
            bacnet_stream['type'],
            bacnet_stream['id'],
            field,
            parsed_value
        )

        return parameters
```

Then we call: 
`self.bacnet_instance.write(request)`

iamliamc, 2020-10-27:
I think it's right in here: 
https://github.com/ChristianTremblay/BAC0/blob/f8232653928e78916ea1d40cc31184847f73c589/BAC0/core/io/Write.py#L169

If the check of the value being written is a bacpypes ArrayOf(DailySchedule):
The subtype will be DailySchedule != ArrayOf...

ChristianTremblay, 2020-10-27:
Yes it's around there. Issue is that I'm not sure how to get around that as I don't yet understand how schedules are created. Also, the write function is a generic function. Its goal was to take a list of arguments (mostly strings) and turn that into a request. 

There are some limits to that when things get complicated (schedules are a good example as they are arrays of constructed data...)

I think I would prefer having a specific function to build a ScheduleWrite request.

Right now I'm trying to understand/make work a schedule write on one of my device... just to see how this work. When done, I should be able to fix that. Thanks for your help.

iamliamc, 2020-10-27:
Yeah that's starting to sound familiar now I think the reason the request get's wrapped in that `bac0_wrapper` class is so that it mimics a "string"

ChristianTremblay, 2020-10-28:
ok I got something that I will publish on the develop branch tonight.

As said, I think we should not use bacnet.write. It is not well suited for complicated stuff.

I've thought for some time about the way I would like to see weeklySchedules. Your dict format inspired the final version.

If you want to read a weeklySchedule on the network, a standard read still work... but there is a supplemental function now called `bacnet.read_weeklySchedule`

This function will return a dict with relevant information about the schedule : 

```
{'object_references': [('multiStateValue', 59)],
 'references_names': ['OCC-SCHEDULE'],
 'states': {'Occupied': 1, 'UnOccupied': 2, 'Standby': 3, 'Not Set': 4},
 'week': {'monday': [('01:00', 'UnOccupied'), ('17:00', 'Occupied')],
  'tuesday': [('02:00', 'UnOccupied'), ('17:00', 'Occupied')],
  'wednesday': [('03:00', 'UnOccupied'), ('17:00', 'Occupied')],
  'thursday': [('04:00', 'UnOccupied'), ('17:00', 'Occupied')],
  'friday': [('05:00', 'UnOccupied'), ('17:00', 'Occupied')],
  'saturday': [('06:00', 'UnOccupied'), ('17:00', 'Occupied')],
  'sunday': [('07:00', 'UnOccupied'), ('17:00', 'Occupied')]}}
```

To write to a schedule, you can reuse that format, or provide a similar dict with at least, "week" and "states".
Then you can use `bacnet.write.weeklySchedule(address, schedule_instance, schedule_dict)`
For write requests, only week and states are used.
In your case, you would need to change your dict to something like : 

```
schedule_example = {
    "states": {"UnOccupied": 0, "Occupied": 1},
    "week": {
        "monday": [("3:00", "Occupied"), ("19:00", "UnOccupied")],
        "tuesday": [("4:00", "Occupied"), ("19:00", "UnOccupied")],
        "wednesday": [("4:00", "Occupied"), ("19:00", "UnOccupied")],
        "thursday": [("4:00", "Occupied"), ("19:00", "UnOccupied")],
        "friday": [("4:00", "Occupied"), ("19:00", "UnOccupied")],
        "saturday": [],
        "sunday": [],
    },
}
```

I think you will see things more clearly with that format. It is human readable.

In next few iterations, this feature will be added to the `BAC0.devices` and schedules will be added to points list.

iamliamc, 2020-10-28:
Hey Christian really appreciate your effort i'll check this out in more detail! 

Clearly i'm a bit narrow minded w/r to how i'm using this project and it's overall architecture, 

> Also, the write function is a generic function. Its goal was to take a list of arguments (mostly strings) and turn that into a request.
>
> There are some limits to that when things get complicated [...]

I am happy to eventually update my project to investigate the current bacnet write it's trying to process, and if it's of bacnet field 'weeklySchedule' to then utilize a different BAC0 write method... is this going to become an anti-pattern if you write more support for other "complex bacnet fields" (i.e. a different write method for lots of different fields)?

ChristianTremblay, 2020-10-28:
BAC0 is a personal side project I'm building since a few years now. I must admit that I'm paying today the price of what I didn't know then. The first implementations were pretty naive... the latest ones are better but they will probably suffer from this same flaw compared to the next versions....

The `.write` function is actually working great for the majority of the transactions commonly needed (AI, AO, AV, MI, MO, MV, BI, BO, BV...) It works also very well for plenty of other objects and properties. I don't think I will rework it (at least not in the next future). But when things get complicated, I guess I could have make it work... but honestly, building a bacpypes request and send it to the stack makes way more sense.

Regarding anti-pattern... my lack of computer science terminology prevents me from answering correctly. But some functions have already been built using something else than `bacnet.write` (Timesync, Reinitialize, dcc..). So yes, in the future, maybe I will have to... But as soon as I can use `bacnet.read` or `bacnet.write` I use them. I hate reinventing the wheel. One of the next challenge will be dealing with events... Who knows what I will find !

Also, I know BAC0 would benefit from a refactoring/architecturing with some clearer view from someone good at those kind of things... I'm not. So I build it the best way I can. With the time I have.  

My ultimate goal : a few lines of code and you interact with BACnet stuff... and actually BAC0 is pretty on spot I think.

That said, thanks for your interest in the project, feel free to provide feedback on this, hope you'll find this easier.

iamliamc, 2020-10-28:
Hope I didn't come off as ungrateful hah. I am just trying to understand why things are the way they are, where they are going and how I can accomplish my goals while hopefully contributing to the great work you share with us all!

ChristianTremblay, 2020-10-28:
Not at all my friend. I just thought it was required to add a little
context/history.

The result would not be the same if I was working full time on it.

I’ll be grateful for any contribution.

ChristianTremblay, 2020-10-29:
Leaving this here : https://bac0.readthedocs.io/en/develop/schedules.html

I'll close for now. Thanks again

closed 2020-10-29 by ChristianTremblay