ChristianTremblay/BAC0 #227: how to use readRange

Alive2212, 2020-11-08:
Hi 
I used from your package and it's very powerful but it's very very slow because i have ~ 90,000 point in my building
so i want to use readRange method but it not work for me.
please give me a true example from it.
thanks

ChristianTremblay, 2020-11-08:
My understanding of readRange is like reading an array of data inside a property. Like trendLogs where I use readRange to read all data of a trend logs but 10records at a time. Not sure how that could help you with your point numbers... 

Do you mean readMultiple ?

That said, for any network with a lot of points, all tools will suffer from speed issues. Especially if you deal with MSTP devices. You have to make choices and use different strategies to mitigate the speed issues. 

I would not use BAC0.device (or at least, not without using a custom object list). In fact, it is good practice to select the points you really need to monitor. There are a lot of points that are not used on a typical network or device. 

The new version of BAC0 allows COV registration. That can help by defining some very important points. 

Polling time should be more than the default of 10sec... for that number of points you could not make it anyway. With Niagara station of 2k-5k points we usually see a polling time of 1 min. And not all points are read as the driver will not refresh a point if there is no trend/alarm or point is not actually registered by a view. 

It is really a matter of what do you need, what is your goal ?

Depending of you goal... you could even think dividing the load over different processes of BAC0 that could monitor different segment of a network filling a database for example... again... it really depends on your goal.

Alive2212, 2020-11-08:
Thanks Christian
I benefited a lot from your explanation.
And about the goal I have to say I want to read many point in one time for example in following query just read `presentValue` from AI:0:
`11:2 analogInput 0 presentValue`
main question is there any way to get `presentValue` from AI:0-128 (all analog inputs)?

ChristianTremblay, 2020-11-08:
BACnet.read is a request for only one property at a time. It will take forever for multiple points. 

You need to use readMultiple

I have some mechanisms to build requests for readMultiple inside a device... let me check if I can provide something. 

On the other hand, have you tried defining a device with a custom object list (analogInput 1-128)

This would create a device with only those points polled every x seconds (adjustable) using readMultiple so very fast. You would also have access to in-memory history for each point. 

```
dev = BAC0.device("11:2", deviceID, BACnet, poll=30, custom_object_list=yourlist)

dev.points
```

ChristianTremblay, 2020-11-09:
It's been a long while now I wanted to rework the readMultiple function to make it simpler. I took the opportunity of your question to prepare something. 

If you want to try, it's in the develop branch. But I think you will like the simplicity.

The actual method to create read property multiple requests is working but lacks some clarity when building it and when we receive the results, they are just a simple list and it's easy to get lost.

Now there will be another way to create RPM requests using a dict.

Preleminary documentation can be found here : https://bac0.readthedocs.io/en/develop/read.html#read-multiple

The basic idea is to create a dict with "what you need to know"

```
_rpm = {'address': '303:9',
        'objects': {
            'analogInput:1094': ['objectName', 'presentValue', 'statusFlags', 'units','description'],
            'analogValue:4410': ['objectName', 'presentValue', 'statusFlags', 'units', 'description']
            }
        }
```

Then you send it

```
bacnet.readMultiple('303:9', request_dict=_rpm)
```

And you play with the results
```
# result of request
{
    ('analogInput', 1094): [
        ('objectName', 'DA-VP'),
        ('presentValue', 4.233697891235352),
        ('statusFlags', [0, 0, 0, 0]),
        ('units', 'pascals'),
        ('description', 'Discharge Air Velocity Pressure')
        ],
    ('analogValue', 4410): [
        ('objectName', 'SAFLOW-ABSEFFORT'),
        ('presentValue', 0.005016503389924765),
        ('statusFlags', [0, 0, 1, 0]),
        ('units', 'percent'),
        ('description', '')
        ]
}
```

in your case, you can build the request like this : 
```
req = {}
req['address'] = '11:2'
req['objects'] = {}

for each in range(0,129):
    req['objects']['analogInput:{}'.format(each)] = ['presentValue']
req
```
```
bacnet.readMultiple('303:9', request_dict=req)
```

Beware : you may need to chunk the request if the result gets too big for your device and its memory+network limitations.

I let that to you, hoping this will help you get some speed from you workflow.

Alive2212, 2020-11-10:
Very very thanks my friend
You are amazing ;)
But unfortunately I just install new release but it seems something went wrong ;(
The error is:
`
AttributeError: module 'BAC0' has no attribute 'connect'
`
or 
`
AttributeError: module 'BAC0' has no attribute 'lite'
`
and I see following link, but not resolved
[https://github.com/ChristianTremblay/BAC0/issues/52](https://github.com/ChristianTremblay/BAC0/issues/52)

I try:
`tmp = BAC0.connect()`
and this
`tmp = BAC0.connect(bokeh_server=False)`
and this
`tmp = BAC0.lite()`

and all ways of installation has this error, my installation ways are:
1. pip install https://github.com/ChristianTremblay/BAC0.git
2. copy zip file to library forlder
3. git clone and python install setup.py

ChristianTremblay, 2020-11-10:
You may need to `pip uninstall BAC0` until nothing is left. Then reinstall.

Alive2212, 2020-11-12:
Hi Christian
Exactly True.
Thanks.
I try to get multiple value of points like following:
`
    _rpm = {'address': '11:8',
            'objects': {
                'analogValue:1': ['presentValue'],
                'analogValue:2': ['presentValue'],
                'analogValue:3': ['presentValue'],
                'analogValue:4': ['presentValue'],
                'analogValue:5': ['presentValue'],
                'analogValue:6': ['presentValue'],
                'analogValue:7': ['presentValue'],
                'analogValue:8': ['presentValue'],
            }
            }
    result = app.bacnet.readMultiple('11:8', request_dict=_rpm)
`
but it seems something went wrong and error is:
 `
BAC0.core.io.IOExceptions.SegmentationNotSupported
`
When I try 1-7 it is ok and get data.
But if I try 1-8  or 0-7 or more points in on request, show this error :|

ChristianTremblay, 2020-11-12:
This is what I was saying when I refered to "chunk" your request.
You will need to make multiple requests with 7 points each to respect the physical limit of the device, which do not support segmentation.

Segmentation would allow the device to send all the information but using multiple network packets, that would be rebuilt by BAC0.

Alive2212, 2020-11-16:
Hi Christian.
This discussion is very very helpful for me and after store value of points in with 7 item packets,
Reading of data speed up 6 time. ;)
Many thanks.

closed 2020-11-16 by Alive2212