ChristianTremblay/BAC0 #234: bacnet_properties throws SegmentationNotSupported error

KingbiteM, 2020-12-03:
Hi all,
I am trying to make a simple program in python that can send certain values to properties and points to a device on BACNET MS/TP. I have a bacnet gateway to connect the device with my network. For now I am just trying to see my device, and manually write to a point or property, once that works I will upgrade the code to do this on its own.
`import BAC0
IPadres='192.168.92.60'
bacnet=BAC0.connect(IPadres)
my_old_device=BAC0.device('2001:1', 4, bacnet, segmentation_supported=False, poll=0)`

When I run my_old_device.points, I get all the points of the device, so this works.
However when I run my_old_device.bacnet_properties I get following error:

`2020-12-03 07:20:23,393 - WARNING | APDU Abort Reject Reason : segmentationNotSupported
Traceback (most recent call last):

  File "C:\Users\maarten.cannoodt\Anaconda3\lib\site-packages\BAC0\core\devices\Device.py", line 727, in update_bacnet_properties
    res = self.properties.network.readMultiple(

  File "C:\Users\maarten.cannoodt\Anaconda3\lib\site-packages\BAC0\core\io\Read.py", line 356, in readMultiple
    raise SegmentationNotSupported()

SegmentationNotSupported

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "<ipython-input-14-8837d45db6aa>", line 1, in <module>
    my_old_device.bacnet_properties

  File "C:\Users\maarten.cannoodt\Anaconda3\lib\site-packages\BAC0\core\devices\Device.py", line 748, in bacnet_properties
    return self._bacnet_properties(update=True)

  File "C:\Users\maarten.cannoodt\Anaconda3\lib\site-packages\BAC0\core\devices\Device.py", line 743, in _bacnet_properties
    self.update_bacnet_properties()

  File "C:\Users\maarten.cannoodt\Anaconda3\lib\site-packages\BAC0\core\devices\Device.py", line 739, in update_bacnet_properties
    raise Exception("Problem reading : {} | {}".format(self.properties.name, e))

Exception: Problem reading : test                             | 
`

I have been looking in the resolved issues here, as far as I could see the solution was either updating BAC0 or linked to another problem.
I am currently running BAC0 version 20.02.20
Any help will be greatly appreciated!

ChristianTremblay, 2020-12-07:
Hi, bacnet properties is the equivalent of : 

`bacnet.readMultiple('2001:1 device 4 all')`

The repsonse will contain a lot of data and if the device cannot use segmentation, you could first try to segment the request yourself. If I'm correct, reading the first array index of the response will give you the lenght of the message... then you can iterate.

But before doing this. Maybe you don't need this to make your script work.

What are the objects and properties on which you would like to write ?

KingbiteM, 2020-12-07:
Hi,

From the properties I would like to read (and write) the description, location and name of the device (for now).
From the objects I would like to read some analoge and multistate inputs and read/write analoge values and multistate output.
This is the documentation of the device I am testing my code with:

[belimo 6weg BACNET.pdf](https://github.com/ChristianTremblay/BAC0/files/5654637/belimo.6weg.BACNET.pdf)

If I understand correctly I should use read_property and write_property instead of bacnet_properties?
In the documentation under 'supported BIBBs' is 'Data Sharing - ReadPropertyMultiple-B'. I assume that this is the same as the readMultiple from BAC0? Any idea why it does not support this?

Thank you!

ChristianTremblay, 2020-12-07:
Segmentation needs memory...small devices sometimes don't have enough.... To handle the majority of cases without having to write an overcomplicated code, I consider that if a device do not support segmentation, I won't use readMultiple "in the context of a BAC0.device".

You could probably make some tests using raw `bacnet.readMultiple` commands and see what comes out.

You can first try if it's possible to write to the properties.
`bacnet.read('2001:4 analogValue 1 description')`
`bacnet.write('2001:4 analogValue 1 description test')`

Sometimes, description (and others) is (are)  readonly... 

BAC0 "device" is a convenient way to interact with devices but in some cases, it is not needed... simple bacnet.read or bacnet.write will be sufficient for the use case.

KingbiteM, 2020-12-08:
This does indeed work for analog values, multistatevalues,..., and it helped me a lot further!
Thank you for your time!

closed 2020-12-08 by KingbiteM