ChristianTremblay/BAC0 #235: bacnet.cov doesn't sent out subscribe frames

JerryXinhui, 2020-12-09:
Hello, recently I upgrade BAC0 to 20.11.21 and try the new cov feature. But I encounter  a problem.
**When I write my code as following, I can see BAC0 send out subscribe frames in the wireshark and my device reply  "value-out-of-range".**
```
import BAC0
bacnet = BAC0.connect(localIP)
bacnet.cov(deviceIP, ("analogValue", 0))
```
[image]

**But if I write my code as this, BAC0 don't send out subscribe(checked by wireshark) any more. Even though, I change back my code to upon, still won't send out subscribe.**

```
import BAC0
def my_callback(elements):
    print(elements)
deviceIP = "192.168.251.119"
localIP = "192.168.251.202"
bacnet = BAC0.connect(localIP)
bacnet.cov(deviceIP, ("analogValue", 0), True, 300,my_callback)
```

My lib version is as followings:

- BAC0==20.11.21
- bacpypes==0.18.3
- bokeh==2.2.3
- Django==3.0.3
- numpy==1.15.4
- openpyxl==2.6.2
- packaging==19.0
- pandas==0.24.2
- Pillow==8.0.1

ChristianTremblay, 2020-12-10:
Does this object exist in the controller ? (AV:0) ?

JerryXinhui, 2020-12-10:
Yes, I'm sure this object exist in my controller.  Otherwise, when I write code like this "bacnet.cov(deviceIP, ("analogValue", 0))" , I won't get the response "value-out-of-range".   Because my controller does't support unlimited lifetime, so I have to write my code to define the lifetime: "bacnet.cov(deviceIP, ("analogValue", 0), True, 300"

ChristianTremblay, 2020-12-10:
Here is what I found

> Christian Tremblay @ChristianTremblay 15:19
> @JoelBender any idea why a device would respond with value_out_of_range when subscribing to a COV ?
> Joel Bender @JoelBender 15:21
> What kind of object is it?
> Ah - wait - maybe I found it - the lifetime or the maximum notification delay is outside of the range supported by the device.
> Christian Tremblay @ChristianTremblay 15:26
> that would make sense....thanks !
> _

Maybe you can try another lifetime value ?

evalles4, 2020-12-11:
I had None in lifetime, changed to 10 and solve the same subscription failed!

JerryXinhui, 2020-12-15:
@ChristianTremblay  Thank you for your investigation. I tried write my code like this, no subscribe frames sent out in the wireshark.
```
import BAC0
def my_callback(elements):
    print(elements)

deviceIP = "192.168.251.100"
localIP = "192.168.251.202"
bacnet = BAC0.connect(localIP)
bacnet.cov(deviceIP, ("analogValue", 0), True, 30, my_callback)
```
[image]

JerryXinhui, 2020-12-15:
The whole wireshark is as followings, no subscribe was captured
[image]

ChristianTremblay, 2020-12-18:
I see nothing bad..... I would be more explicit with the subnet as a good practice (localIP = "192.168.251.202/24").
I would use BAC0.lite... because, well, you probably don't need the web server.
Maybe if you try with the develop branch ?

ChristianTremblay, 2021-02-02:
@JerryXinhui 
There was a bug reported in Issue #247 which is fixed in the develop branch. Can you have a look and tell me if it's fixing something on your side ?

JerryXinhui, 2021-03-23:
@ChristianTremblay  Sorry for the late reply. I tried the develop branch, yes ,  It's works well, only happened in the master branch. Thank you!

AlbertFX91, 2021-06-28:
Hi! It's planned this bugfix to be released in a new BAC0 version soon? Thanks! ❤️

TobiasOberle, 2021-12-03:
Hey @ChristianTremblay,
how is it going with the next release? Will it fixing this issue? I don't want to stress out... do you know approximately when the new release is rolled out?

ChristianTremblay, 2021-12-03:
You are right.... I should just go with it and roll a new version.

ChristianTremblay, 2021-12-03:
@TobiasOberle @AlbertFX91 
Done : 21.12.03
Time to 
```
pip install BAC0 --upgrade
```

TobiasOberle, 2021-12-07:
Thank you a lot @ChristianTremblay  for your work. I tried it but still no subscribe frames. What I do:
b = BAC0.connect()
b.cov('15220:1', 'analogOutput:1', confirmed=False, lifetime=3600)

ChristianTremblay, 2021-12-17:
@TobiasOberle 

I made 2 tests here with one of my devices

```
def my_callback(elements):
    ...:     print("Hi, I'm a callback")
    ...: 
    ...: #deviceIP = "192.168.251.100"
    ...: #localIP = "192.168.251.202"
    ...: #bacnet = BAC0.connect(localIP)
    ...: bacnet.cov('304:4', ("analogInput", 10053), True, 30, my_callback)

In [17]: 2021-12-16 20:41:26,335 - INFO    | Subscription success
2021-12-16 20:41:31,032 - INFO    | Received COV Notification for {'source': <RemoteStation 304:4>, 'object_changed': ('analogInput', 10053), 'properties': {'presentValue': 19.078420639038086, 'statusFlags': [0, 0, 0, 0]}}
Hi, I'm a callback
In [17]: 
```

```
In [21]: def my_callback(elements):
    ...:     print("Hi, I'm a callback")
    ...: 
    ...: #deviceIP = "192.168.251.100"
    ...: #localIP = "192.168.251.202"
    ...: #bacnet = BAC0.connect(localIP)
    ...: bacnet.cov('304:4', ("analogInput", 10053), True, 30, my_callback)

In [22]: 2021-12-16 20:47:36,453 - INFO    | Subscription success
In [22]: def my_callback(elements):
    ...:     print("Hi, I'm a callback")
Hi, I'm a callback
In [22]: gar['GARAGE_ZN-T'] = 21

In [23]: 2021-12-16 20:47:45,978 - INFO    | Received COV Notification for {'source': <RemoteStation 304:4>, 'object_changed': ('analogInput', 10053), 'properties': {'presentValue': 19.081329345703125, 'statusFlags': [0, 0, 0, 1]}}
Hi, I'm a callback
In [23]: 2021-12-16 20:47:50,984 - INFO    | Received COV Notification for {'source': <RemoteStation 304:4>, 'object_changed': ('analogInput', 10053), 'properties': {'presentValue': 21.0, 'statusFlags': [0, 0, 0, 1]}}
Hi, I'm a callback
In [23]: gar['GARAGE_ZN-T'] = 'auto'

In [24]: 2021-12-16 20:48:01,001 - INFO    | Received COV Notification for {'source': <RemoteStation 304:4>, 'object_changed': ('analogInput', 10053), 'properties': {'presentValue': 19.082578659057617, 'statusFlags': [0, 0, 0, 0]}}
Hi, I'm a callback
In [24]: bacnet.cancel_cov('304:4', ("analogInput", 10053))

In [25]: 2021-12-16 20:48:18,969 - INFO    | Subscription success
In [25]: gar['GARAGE_ZN-T'] = 21

In [26]: gar['GARAGE_ZN-T'] = 'auto'
```

Both worked... do you have more details about your issue ?

JerryXinhui, 2021-12-28:
@ChristianTremblay   Are you sure you don't use branch for mistake?  I upgrade the BAC0 to 21.12.3 via pip install BAC0 --upgrade,  use the same code , but can't subscribe successfully, checked the wireshark, no subscribeCOV frames find.

import BAC0
def my_callback(elements):
    print(elements)

localIP = "192.168.251.10"
bacnet = BAC0.connect(localIP)
bacnet.cov('1:5', ("analogValue", 0), True, 30, my_callback)

**I have no ideal any other information I can offer.  Maybe the dependencies?**
BAC0                           21.12.3
bacpypes                     0.18.5
bokeh                          2.4.2
numpy                          1.15.4
Pillow                         8.4.0

**all the log is as followings:**
2021-12-28 14:16:56,810 - INFO    | Starting BAC0 version 21.12.03 (Complete)
2021-12-28 14:16:56,811 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2021-12-28 14:16:56,811 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2021-12-28 14:16:56,811 - INFO    | Starting TaskManager
2021-12-28 14:16:56,812 - INFO    | Using ip : 192.168.251.10
2021-12-28 14:16:56,814 - INFO    | Starting app...
2021-12-28 14:16:56,815 - INFO    | BAC0 started
2021-12-28 14:16:56,815 - INFO    | Registered as Simple BACnet/IP App
2021-12-28 14:16:56,835 - INFO    | Update Local COV Task started
2021-12-28 14:16:58,042 - INFO    | Server started : http://192.168.251.10:8111
 * Serving Flask app "BAC0.web.FlaskServer" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
2021-12-28 14:16:59,085 - INFO    | 192.168.251.111 network number is 6
2021-12-28 14:16:59,100 - INFO    | 192.168.251.111 network number is 6
 * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2021-12-28 14:17:00,147 - INFO    | 192.168.251.111 router to [1]
2021-12-28 14:17:00,162 - INFO    | 192.168.251.111 router to [1]
2021-12-28 14:17:02,091 - INFO    | Addr : 192.168.251.111
2021-12-28 14:17:02,154 - INFO    | 192.168.251.111 routing table
2021-12-28 14:17:02,154 - INFO    |     6 1 bytearray(b'')
2021-12-28 14:17:02,155 - INFO    |     10 1 bytearray(b'')
2021-12-28 14:17:02,155 - INFO    |     10 1 bytearray(b'')
2021-12-28 14:17:02,155 - INFO    |     55657 1 bytearray(b'')
2021-12-28 14:17:02,155 - INFO    |     1 2 bytearray(b'')
2021-12-28 14:17:02,169 - INFO    | 192.168.251.111 routing table
2021-12-28 14:17:02,169 - INFO    |     6 1 bytearray(b'')
2021-12-28 14:17:02,169 - INFO    |     10 1 bytearray(b'')
2021-12-28 14:17:02,169 - INFO    |     10 1 bytearray(b'')
2021-12-28 14:17:02,169 - INFO    |     55657 1 bytearray(b'')
2021-12-28 14:17:02,169 - INFO    |     1 2 bytearray(b'')
2021-12-28 14:17:04,119 - INFO    | Found those networks : {1, 6}
2021-12-28 14:17:04,119 - INFO    | Discovering network 1
2021-12-28 14:17:07,155 - INFO    | Discovering network 6

ChristianTremblay, 2021-12-28:
Can you try 
```
bacnet=BAC0.lite() 
```

Instead of connect ?

JerryXinhui, 2021-12-29:
@ChristianTremblay  not quite sure I get you real point. But I tried like this, still not work. No error and also no subscribe.
Test1:
bacnet = BAC0.lite()
bacnet.cov('1:5', ("analogInput", 12), True, 30, my_callback)

Test2:
bacnet = BAC0.lite("192.168.251.10/24")
bacnet.cov('1:5', ("analogInput", 12), True, 30, my_callback)

ChristianTremblay, 2021-12-29:
Connect is subclass of lite ... was hoping there was no missing part in connect that could explain your issue. 

I keep thinking something in your setup prevents global broadcast to reach your Python instance.

TobiasOberle, 2022-01-11:
Same for me as @JerryXinhui said. Tried this:

_import BAC0
def my_callback(elements):
    print(elements)
    print('heeeeello')
bacnet = BAC0.lite("10.22.254.3")
bacnet.cov('15220:1', ("analogOutput", 1), True, 30, my_callback)

with this output:

2022-01-11 10:12:38,690 - INFO    | Starting BAC0 version 21.12.03 (Lite)
2022-01-11 10:12:38,690 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-01-11 10:12:38,690 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-01-11 10:12:38,691 - INFO    | Starting TaskManager
2022-01-11 10:12:38,694 - INFO    | Using ip : 10.22.254.3
2022-01-11 10:12:39,082 - INFO    | Starting app...
2022-01-11 10:12:39,083 - INFO    | BAC0 started
2022-01-11 10:12:39,084 - INFO    | Registered as Simple BACnet/IP App
2022-01-11 10:12:39,089 - INFO    | Update Local COV Task started

The callback seems to be not even called

ChristianTremblay, 2022-01-12:
this is weird... @TobiasOberle the 30 means 30 seconds then stop my subscription, is that what you want ?
But you should at least see some kind of answer from the device...

## Device that do not support COV
```
In [7]: bacnet.cov('303:8',('analogValue',21), True, 30, my_callback)

In [8]: 2022-01-11 21:56:40,698 - ERROR   | Subscription failed. <bacpypes.apdu.RejectPDU(None,206) instance at 0x1f7af130790>
In [8]: bacnet.cov('303:8',('analogValue',21), False, 30, my_callback)

In [9]: 2022-01-11 21:56:49,088 - ERROR   | Subscription failed. <bacpypes.apdu.RejectPDU(None,207) instance at 0x1f7b0add940>
In [9]: 
```

## Working example
```
In [12]: bacnet.cov('303:9',('analogValue',249073), True, 30, my_callback)

In [13]: 2022-01-11 21:58:38,820 - INFO    | Subscription success
2022-01-11 21:58:41,440 - INFO    | Received COV Notification for {'source': <RemoteStation 303:9>, 'object_changed': ('analogValue', 249073), 'properties': {'presentValue': 20.0, 'statusFlags': [0, 0, 0, 0]}}
{'source': <RemoteStation 303:9>, 'object_changed': ('analogValue', 249073), 'properties': {'presentValue': 20.0, 'statusFlags': [0, 0, 0, 0]}}
heeeeello
In [13]: bacnet.cancel_cov('303:9',('analogValue',249073))

In [14]: 2022-01-11 21:59:18,554 - INFO    | Subscription success
In [14]: bacnet.cov('303:9',('analogValue',249073), True, 30, my_callback)

In [15]: 2022-01-11 21:59:35,679 - INFO    | Subscription success
2022-01-11 21:59:36,409 - INFO    | Received COV Notification for {'source': <RemoteStation 303:9>, 'object_changed': ('analogValue', 249073), 'properties': {'presentValue': 20.0, 'statusFlags': [0, 0, 0, 0]}}
{'source': <RemoteStation 303:9>, 'object_changed': ('analogValue', 249073), 'properties': {'presentValue': 20.0, 'statusFlags': [0, 0, 0, 0]}}
heeeeello
In [15]: 2022-01-11 21:59:46,395 - INFO    | Received COV Notification for {'source': <RemoteStation 303:9>, 'object_changed': ('analogValue', 249073), 'properties': {'presentValue': 22.0, 'statusFlags': [0, 0, 0, 0]}}
{'source': <RemoteStation 303:9>, 'object_changed': ('analogValue', 249073), 'properties': {'presentValue': 22.0, 'statusFlags': [0, 0, 0, 0]}}
heeeeello
In [15]: bacnet.cov('303:9',('analogValue',249073), True, 0, my_callback)

In [16]: 2022-01-11 22:01:14,487 - INFO    | Subscription success
2022-01-11 22:01:16,389 - INFO    | Received COV Notification for {'source': <RemoteStation 303:9>, 'object_changed': ('analogValue', 249073), 'properties': {'presentValue': 24.5, 'statusFlags': [0, 0, 0, 0]}}
{'source': <RemoteStation 303:9>, 'object_changed': ('analogValue', 249073), 'properties': {'presentValue': 24.5, 'statusFlags': [0, 0, 0, 0]}}
heeeeello
In [16]: 2022-01-11 22:01:41,391 - INFO    | Received COV Notification for {'source': <RemoteStation 303:9>, 'object_changed': ('analogValue', 249073), 'properties': {'presentValue': 23.0, 'statusFlags': [0, 0, 0, 0]}}
{'source': <RemoteStation 303:9>, 'object_changed': ('analogValue', 249073), 'properties': {'presentValue': 23.0, 'statusFlags': [0, 0, 0, 0]}}
heeeeello
In [16]: 2022-01-11 22:02:06,425 - INFO    | Received COV Notification for {'source': <RemoteStation 303:9>, 'object_changed': ('analogValue', 249073), 'properties': {'presentValue': 22.0, 'statusFlags': [0, 0, 0, 0]}}
{'source': <RemoteStation 303:9>, 'object_changed': ('analogValue', 249073), 'properties': {'presentValue': 22.0, 'statusFlags': [0, 0, 0, 0]}}
heeeeello
In [16]: bacnet.cancel_cov('303:9',('analogValue',249073))

In [17]: 2022-01-11 22:02:15,292 - INFO    | Subscription success

```

In the working example, I changed the delay to 0 for "infinite" subscription so I could make some changes and see them...

Can you tell me more details about the devices you are trying to talk to using COV ?

JerryXinhui, 2022-05-07:
Try with the latest code, works fine, many thanks for your great support@ChristianTremblay

closed 2022-05-07 by JerryXinhui

JaneWang92, 2022-06-22:
@ChristianTremblay, when I use bacnet.cov('1:5', ("analogValue", 0), True, 30, my_callback) to subscribe a confirmed COV, it can subscribe success, but when the notification send out, bacnet client does not response the notification, what should I do? thanks!

JerryXinhui, 2022-06-23:
@ChristianTremblay   until now, we can use bacnet.cov to subscribe cov for an object .  What if I want to subscribe cov for an object property?

JaneWang92, 2022-06-24:
@ChristianTremblay I can use bacnet.cov (py file) to subscribe cov for an object and cov notification can be sent out, but when I import this method to robotframework to test it, seems that it cannot subscribe normally, no wireshark frame sent out when I run robot file, I am not sure whether it is cov method issue or robot framework issue, is there any advise? thanks!
cov ${deviceip} ${object_id} confirmed=True lifetime=9600 callback=my_callback