ChristianTremblay/BAC0 #236: Can't Find Points of Cylon Controller

Alive2212, 2020-12-12:
Hi 
I have a test environment to discover _Cylon_ controller and it's points.
_Cylon_ device was detected in _Yabe_ but in BAC0 not funded.
And _Yabe_ can not discover any point of this controller.

ChristianTremblay, 2020-12-13:
Wireshark is your friend there. You will be able to see what is going on.
You can even compare what YABE is doing vs BAC0. 

Looking of objects in device : 

`bacnet.read('address device deviceInstance objectList')`

If it comes out empty... you are facing a controller with nothing exposed.

Alive2212, 2020-12-13:
Excuse me It's my fault, 
Yabe also can't find it. 
But Niagara have a special driver for Cylon Controller with Bacnet protocol.

ChristianTremblay, 2020-12-17:
Never worked with that...but this : https://docs.cylon.com/public/SWDS0003 let me think port 4950 is probably a way to exposer the controllers to the Web for a client to connect, without using the BACnet protocol itself on the internet (which is unsecure).

I would not be surprised you are trying to connect remotely using the public IP on this system... but doing so you are not connecting using BACnet but the "whatever web share 4950 port Java protocol" they created to get access to their controller from outside the subnet.

closed 2021-02-23 by github-actions[bot]