ChristianTremblay/BAC0 #237: BAC0 & Flask API?

bbartling, 2020-12-13:
Hi Christian!

Out of curiosity can a BAC0 lite app ever run along side/support a restful Flask API? My idea would be to make HTTP requests to the BAC0/Flask_restful URL where the HTTP requests would define what BACnet reads/writes to make on the HVAC system.. Hopefully that makes sense.

I am trying to combine the code of the flask_restful hello world example ([Gist](https://gist.github.com/bbartling/816a0ab970c35c6a4a16f9afc4843d38)) with some BAC0 lite app code but Python throws errors out, would you have any insight if this this possible or not?

```
C:\Users\bbartling\OneDrive - Slipstream\Desktop\TesterFlask>py -3.7 main.py
Starting BACnet device
2020-12-13 07:54:12,709 - INFO    | Starting BAC0 version 20.02.20 (Lite)
2020-12-13 07:54:12,710 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2020-12-13 07:54:12,710 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2020-12-13 07:54:12,742 - INFO    | Using ip : 192.168.0.103
Traceback (most recent call last):
  File "main.py", line 47, in <module>
    class App:
  File "main.py", line 48, in App
    dev = start_device()
  File "main.py", line 39, in start_device
    new_device = BAC0.lite()
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python37\lib\site-packages\BAC0\scripts\Lite.py", line 129, in __init__
    **params
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python37\lib\site-packages\BAC0\scripts\Base.py", line 117, in __init__
    localIPAddr
BAC0.core.io.IOExceptions.InitializationError: IP Address provided (192.168.0.103) invalid. Check if another software is using port 47808 on this network interface. If so, you can define multiple IP per interface. Or specify another IP using BAC0.lite(ip='IP/mask')
```

How does the full BAC0 app work to accommodate the Flask/Bokeha side?

Thanks!
Ben

NubeDev, 2020-12-13:
Here is a simple example. 
https://github.com/NubeIO/rubix-bacnet-server/blob/master/master/master.py

bbartling, 2020-12-14:
Wow cool thanks!

bbartling, 2020-12-14:
Would you have any examples on how to use the API? Like an example with
http requests to make a BACNET read or write? Thanks!

NubeDev, 2020-12-14:
Its all in the code https://github.com/NubeIO/rubix-bacnet-server/blob/master/master/master.py#L57, 
This was just some test code I did. if it dosnt work let me know and ill test it and do some postman docs

HTTP POST
`/read`

body
```
device_id = 123
object_id =1
object_type = binaryOutput
prop = 85
```

HTTP POST
`/write`

body
```
device_id = 123
object_id =1
object_type = binaryOutput
value = active
```

bbartling, 2020-12-14:
Awesome thank you so much

bbartling, 2020-12-16:
@NubeDev 

I have the repo git cloned on a rasp pi, running `master.py` :)

From another computer on the LAN I am attempting to do an http POST request at the rasp pi, but I cant get around the:
```
{
    "description": "Read request was missing a required parameter. Required: [device_id, object_id, object_type]",
    "status_code": 500
}
```

Post man screen shot below, apologize as I am not an expert with this stuff. Should I be using the 'body' or 'params' of the http request for the `[device_id, object_id, object_type]`?

[image]

This is the MSTP device on the LAN that I was hoping to do an http request the rasp pi to read the temperature sensor on analog input 2.

[image]

Any tips greatly appreciated!!! On how I should be using Postman!
Thanks,
Ben

NubeDev, 2020-12-16:
Hey mate,
As per the func `do_read` it wants in the `body` `prop`  like `85` for BACnet `pv`
` prop = request.get_json().get('prop')`

```
    @app.route('/read', methods=['POST'])
    def do_read():
        try:
            device_id = request.get_json().get('device_id')
            object_id = request.get_json().get('object_id')
            object_type = request.get_json().get('object_type')
            prop = request.get_json().get('prop')
        except:

```

closed 2020-12-17 by bbartling