ChristianTremblay/BAC0 #239: network connection times out

ionelpetrus, 2020-12-17:
Hi, 
Our connection to the devices appears to be timing out after about 10-12 hours or so of continuous connection to the bacnet network; it is a recurring issue. While we are connected, we have no issue at all with accessing devices, points etc.
Any help is greatly appreciated.
Thank you.
Ionel

ChristianTremblay, 2020-12-17:
Please provide more details

ionelpetrus, 2020-12-21:
Hi Christian, 
We connect to one bacnet/IP network that has two routers, each with its own MS/TP network downstream.
We mostly just read points; started to collect data and all was good. Left the connection active overnight; in the morning we realized that BAC0 stopped communicating with all devices on the net. Tried the same thing the second night, got the same result. Checked linux settings and so far we didn't see anything out of the ordinary.
Don't really know if it is an issue with BAC0 or a linux issue; keep digging to see of there is some sort of linux net setting that creates this problem. 
Hope this helps.
Thank you again for all your work.
Ionel

ChristianTremblay, 2020-12-21:
Any error ?

How do you start your app ? Simple script using terminal ? If so, do you use something like tmux that will allow you to "not loose" the terminal connection if connecting using SSH....

ionelpetrus, 2021-01-05:
we were using jupyter notebook; switched to using tmux and all is good.
Thank you for all your help.

closed 2021-01-11 by ChristianTremblay