ChristianTremblay/BAC0 #240: BACnet write, relinquish priority 8

bbartling, 2020-12-18:
Hello,

Would anyone know how to relinquish a command written from BAC0? I can definitely write to a BACnet MSTP device (script below) but curious about how would I release a command?

Would releasing commands only be for actual hardware output (`binaryOutput analogOutput`) and not for `analogValue`?

Any tips greatly appreciated!

Is the code below writing at priority 8 in BACnet?

```
import BAC0, time, random

bacnet = BAC0.lite()

address = '12345:2'
object_type = 'analogValue'
object_instance = '302'
value = str(random.randint(66, 122))#random number

# create write statement and run
write_vals = f'{address} {object_type} {object_instance} presentValue {value}'
print("Excecuting write_vals statement:", write_vals)

write_result = bacnet.write(write_vals)
print('BACnet written to MSTP device')
print('Wait a few seconds....')
time.sleep(10)

read_vals = f'{address} {object_type} {object_instance} presentValue'
check = bacnet.read(read_vals)
print(check)

bacnet.disconnect()
print('BACnet disconnected')
```

ChristianTremblay, 2020-12-19:
```
point.write("null", priority=1)
```

ChristianTremblay, 2020-12-19:
by default, BAC0 write at priority 16

bbartling, 2020-12-20:
Hi Christian,

Thanks so much for taking the time to respond. Am i doing anything goofy in the code below?

For example, if I incorporated a `null` into the second write statement but when I read the analog value again it comes back as `nan` in `read_vals2`

I was thinking if I wrote to analogValue and then released it would just resort back to the default value. But Maybe I am getting screwed in my thinking as the reads/writes are direct to an MSTP device and but through a supervisory controller like a JACE where when the point is released it just goes back to a default value. Would I need to incorporate a default value for a device if I am not using some sort of a supervisory controller?

```
import BAC0, time, random

bacnet = BAC0.lite()

address = '12345:2'
object_type = 'analogValue'
object_instance = '302'
value = str(random.randint(66, 122))#random number

# create write statement and run
write_vals = f'{address} {object_type} {object_instance} presentValue {value}'
print("Excecuting write_vals statement:", write_vals)

write_result = bacnet.write(write_vals)
print('BACnet written to MSTP device')
print('Wait a few seconds....')
time.sleep(10)

read_vals = f'{address} {object_type} {object_instance} presentValue'
check = bacnet.read(read_vals)
print(check)

# create write release statement and run
write_rel = f'{address} {object_type} {object_instance} presentValue null'
print("Excecuting write_vals statement:", write_rel)

write_result = bacnet.write(write_rel)
print('BACnet written to MSTP device')
print('Wait a few seconds....')
time.sleep(10)

read_vals2 = f'{address} {object_type} {object_instance} presentValue'
check2 = bacnet.read(read_vals2)
print(check2)

bacnet.disconnect()
```

ChristianTremblay, 2021-01-18:
Pretty sure you can get rid of `sleep` as the IOCB should make you wait between write and read. If not....reduce it... 10sec is a very long delay.

Do you have an issue with your code ?

bbartling, 2021-01-19:
Hi Christian,

(2 Questions)

I was just curious if there was a method to release points when written too. The code above for example I was experimenting with writing a `random value` to a point on an MSTP device, then writing `null` to release to auto, and then reading the point again it comes back as `nan`. The only thing I can compare it too is workbench, write `null` to release to auto. This must not be the same for BAC0 :)

I think because my device the point I am experimenting with is not pulled into the supervisory controller which could give this some default value I would need to define myself? Hopefully this makes sense, I would just have to define like a `relinquish_default` in Python for an integer or float that instead of `nan` when ready to release?

(Question 2)
If your BAC0 script is only manipulating setpoints (versus IO) would this be anything different? For example in Workbench you can takeaway the `override/auto` priority 8's feature for the building operator and only give them a `set.` Would you ever do anything like that with BAC0 if just manipulating setpoints? OR would the process be the same exact as question 1?

Thanks!

ChristianTremblay, 2021-01-27:
@bbartling found a solution or I made you wait too long ? :)

bbartling, 2021-01-27:
Hi Christian,

Well its no rush its more of a side project :) Hoping to use more in the future I dont want to bug you too much ya know. I also wasnt quite sure how relevant the question was either but I would keep it open if its something worth working on :) Absolutely no rush at all and thanks for the response!

Ben

ChristianTremblay, 2021-03-31:
> Hi Christian,
>
> (2 Questions)
>
> I was just curious if there was a method to release points when written too. The code above for example I was experimenting with writing a `random value` to a point on [...]

If I take JCI controllers as reference. Inside the controller itself, points are BACnet object which are written to to a certain priority. Take an hardware output. Let say the software inside the controller write to the present value directly. If you, in workbench write to the point @ 10 (or even @ 16) you will override the internal sequence. Releasing this write (send `null` to the priority you wrote, will release this. This is the exact same behaviour than a manual operator write @ 8 (AKA an override).

The only thing is that you must keep in mind that the priorities of the (let say) NumericWritable in Niagara are not exactly the same priorities than the BACnet object. Niagara "mimic" this feature. So when you make overrides on the display of a controller... Niagara won't see it. Because it should sync the priorityArray property which is another property (a different read than the presentValue... which would double the traffic for each points).

```
# in BAC0
dev['output'] = 10
dev['output'] = 'auto' #<- release
```

[ref] : https://bac0.readthedocs.io/en/latest/controller.html?highlight=write#releasing-an-input-simulation-or-output-override

> I think because my device the point I am experimenting with is not pulled into the supervisory controller which could give this some default value I would need to define myself? Hopefully this makes [...]

Yes, that would work. The priority array of the point will be traversed and if all is null... the default value wins.

> (Question 2)
> If your BAC0 script is only manipulating setpoints (versus IO) would this be anything different? For example in Workbench you can takeaway the `override/auto` priority 8's feature for [...]

There is no difference. Nothing prevents overriding setpoints...even if it's bad practice (personally, I "hide" the overrides slots).

> 
> Thanks!

bbartling, 2021-03-31:
Wow cool thanks Christian.

closed 2021-04-07 by bbartling