ChristianTremblay/BAC0 #241: Reading all objects and their properties

mosterme1003, 2021-01-06:
Hello! I just want to ask if there's a way to easily read all objects of a bacnet device and all the available properties of each object. Currently I read the object list of the device object and iterate through the object list. With the given objects I try to read every property that could be available like in this code example for analog objects.

```
propertyListAnalog = ["objectName", "presentValue", "description", "deviceType", "statusFlags", "eventState", 
                      "reliability", "outOfService", "updateInterval", "units", "minPresValue", "maxPresValue", 
                      "resolution", "covIncrement", "timeDelay", "notificationClass", "highLimit", "lowLimit", 
                      "deadband", "limitEnable", "eventEnable"]

bacnet = BAC0.connect(ip=ipLocal)
objectList= bacnet.read('%s device %s objectList' %(ipDevice, idDevice))
objectPropertyList = {}

for obj in objectList:
     objectType = obj[0]
     objectInstance = obj[1]
     
     if objectType == "analogInput" or objectType == "analogOutput" or objectType == "analogValue":
          for prop in propertyListAnalog:
               try:
                    value = bacnet.read('%s %s %s %s' % (deviceIp, objectType, objectInstance, prop))
                    objectPropertyList [objectType + str(objectInstance)][prop] = value
               except BAC0.core.io.IOExceptions.UnknownPropertyError:
                    continue
```

Now this is kinda slow. It takes roughly 5 minutes for about 300 bacnet objects. Is there a way to read all properties faster? I think it would be faster if I would use the readMultiple function on each object but I'll have to know which properties are available first, right? Is there a way to get this information? I know that there is a bacnet property called propertyList but close to none of the bacnet objects I saw in my network had this property filled. I'll appreciate any help!

mosterme1003, 2021-01-06:
Well, I think I found a way that works for me. I'm defining the device and read out all points and its properties. Seems to work fine like in this example code. Still thank you for your work!

```
import BAC0
bacnet = BAC0.connect(ip='192.168.1.3/24')
device = BAC0.device('192.168.1.2', 2100101, bacnet)
point = device.points[0]
properties = point.bacnet_properties
```

mosterme1003, 2021-01-12:
I really should have read the docs here.. You also can use `bacnet.readMultiple('address object object_instance all')`

mosterme1003, 2021-01-26:
So after some attempts I just don't find a way that works for my case. I want to discover all bacnet objects and all their properties. First way:  `point = device.points[0]` `properties = point.bacnet_properties` 
I realized that a defined device just has analog, binary and multistate objects in its points list. Is there a way to get the other objects too?
Second way: `bacnet.readMultiple('address object object_instance all')`
When I have an object list with all identifiers I can read all properties from each object like this. But this function returns a list with all property values and I don't know which value is belonging to which property.

Is there a way to get a list of **all** objects with all their properties? Something like the first way but with the other object types.

ChristianTremblay, 2021-03-31:
> So after some attempts I just don't find a way that works for my case. I want to discover all bacnet objects and all their properties. First way: `point = device.points[0]` `properties = [...]

The goal of the device was to have a "Python" representation of the BACnet device. I concentrated my efforts on typical points on which I usually make actions. This is why not all BACnet objects are part of the device itself. Some objects are very complex, made from structured arrays... It's not easy.

This is why the read and write functions are not always working with all objects (ex. schedules). 

But nothing prevents you to use network read/write to deal with weird objects. When I find something I like for complex objects, I create a specific class and the device can then interact with the object. Schedules are probably the best example.

https://bac0.readthedocs.io/en/latest/schedules.html?highlight=schedule

So the answer is : yes it's possible to read any bacnet point and write to them... but you will need to work harder for complex objects and figure out how to build a write request with the information you need to send it to the stack.

https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/functions/Schedule.py

> Second way: `bacnet.readMultiple('address object object_instance all')`
> When I have an object list with all identifiers I can read all properties from each object like this. But this function [...]

```
bacnet.readMultiple('%s %s %s %s' % (address, 'analogInput', 1, 'all',prop_id_required=True)
```

closed 2021-04-07 by github-actions[bot]

nixcodes, 2024-06-13:
Hey! I was wondering if there is any way I can get all the object_instance values for a particular device.

wonch002, 2024-08-21:
> Hey! I was wondering if there is any way I can get all the object_instance values for a particular device.

@nixcodes I believe you're looking for a way to find all objects for a device. You can do this by using bacnet.read to query the objectList property. Here’s an example:

```python
import BAC0

address = "<your device address>"
device_instance = "<your device instance>"
object_type = "device"
property_name = "objectList"

args = f"{address} {object_type} {device_instance} {property_name}"

bacnet = BAC0.lite()
object_list = bacnet.read(args)

for obj_type, obj_instance in object_list:
    print(f"Object Type: {obj_type}, Instance: {obj_instance}")
```

I hope this helps!