ChristianTremblay/BAC0 #242: Scheduler Object Commanding is not supported over BAC0

aniruddhkhalatkar1, 2021-01-08:
Bacnet Type 17:Scheduler Type Object  which is standard Bacnet Object is not supported for commanding value. 
Scheduler Object can be classified into three types:
1. Analog
2. Binary
3. Multistate

aniston, 2021-01-14:
Hi @aniruddhkhalatkar1 , a BACnet scheduler Obj. typ 17 would set its _List of Object Property References_ where the list of objects (objects must be from the same device) need their Present Value updated (PV) using a given Priority Array.

So, all AV(02), BV(05) or MV(19) Obj. Prop. sould be able to have their PV set as long as the correct PA is given. Are you saying that it cannot do this ?

ChristianTremblay, 2021-01-18:
I don't understand. The current behaviour of BAC0 is : 
take a Python representation of a schedule and write it to the schedule object.
https://bac0.readthedocs.io/en/latest/schedules.html?highlight=schedule

or read it to get a python representation (a dict...)

```
schedule_example_multistate = {
    "states": {"Occupied": 1, "UnOccupied": 2, "Standby": 3, "Not Set": 4},
    "week": {
        "monday": [("1:00", "Occupied"), ("17:00", "UnOccupied")],
        "tuesday": [("2:00", "Occupied"), ("17:00", "UnOccupied")],
        "wednesday": [("3:00", "Occupied"), ("17:00", "UnOccupied")],
        "thursday": [("4:00", "Occupied"), ("17:00", "UnOccupied")],
        "friday": [("5:00", "Occupied"), ("17:00", "UnOccupied")],
        "saturday": [("6:00", "Occupied"), ("17:00", "UnOccupied")],
        "sunday": [("7:00", "Occupied"), ("17:00", "UnOccupied")],
    },
}
schedule_example_binary = {
    "states": {"inactive": 0, "active": 1},
    "week": {
        "monday": [("1:00", "active"), ("16:00", "inactive")],
        "tuesday": [("2:00", "active"), ("16:00", "inactive")],
        "wednesday": [("3:00", "active"), ("16:00", "inactive")],
        "thursday": [("4:00", "active"), ("16:00", "inactive")],
        "friday": [("5:00", "active"), ("16:00", "inactive")],
        "saturday": [("6:00", "active"), ("16:00", "inactive")],
        "sunday": [("7:00", "active"), ("16:00", "inactive")],
    },
}
schedule_example_analog = {
    "states": "analog",
    "week": {
        "monday": [("1:00", 22), ("18:00", 19)],
        "tuesday": [("2:00", 22), ("18:00", 19)],
        "wednesday": [("3:00", 22), ("18:00", 19)],
        "thursday": [("4:00", 22), ("18:00", 19)],
        "friday": [("5:00", 22), ("18:00", 19)],
        "saturday": [("6:00", 22), ("18:00", 19)],
        "sunday": [("7:00", 22), ("18:00", 19)],
    },
}
```

aniruddhkhalatkar1, 2021-01-20:
Hi Christian,
I am facing this issue regarding writing presentValue , since Scheduler have different types, the write operation of presentValue is giving error

ChristianTremblay, 2021-01-20:
I don't think I've ever wrote to a Schedule PV... 
Why do you want to write to the present value instead of modifying the schedule itself that will generate the correct present value ?

aniruddhkhalatkar1, 2021-01-20:
I have a requirement for testing a hvac program in which I need to change the presentValue of Scheduler 0(off) to 1(on), which would enable scheduler operation in hvac program

ChristianTremblay, 2021-01-20:
[ref: https://bac0.readthedocs.io/en/latest/schedules.html?highlight=schedule]

start by reading the schedule using :
```
sched = bacnet.read_weeklySchedule(address, instance)
```
modify the `sched` dict received so the present value change for the actual time of day and write it back to the controller
```
bacnet.write_weeklySchedule(address, instance, schedule=sched )
```

ChristianTremblay, 2021-02-02:
@aniruddhkhalatkar1 does this work for you ?

closed 2021-04-12 by github-actions[bot]