ChristianTremblay/BAC0 #244: pip3 install BAC0 fails

georgegoglodze, 2021-01-25:
hello, please be advised the pip3 install fails but it works if I provide older version of BAC0

```
pip3 install BAC0
Processing /Users/zgoglo716/Library/Caches/pip/wheels/58/8e/71/554a56d3182b3c6c9eca912b48581f46e1fa64d11c15d36e30/BAC0-20.11.21-py3-none-any.whl
Requirement already satisfied: bacpypes in /usr/local/lib/python3.9/site-packages (from BAC0) (0.18.3)
Requirement already satisfied: colorama in /usr/local/lib/python3.9/site-packages (from BAC0) (0.4.4)
Collecting netifaces
  Using cached netifaces-0.10.9.tar.gz (28 kB)
Building wheels for collected packages: netifaces
  Building wheel for netifaces (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/local/opt/python@3.9/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/qm/9jnwq5m542q6qwwdpwm3dh300000gp/T/pip-install-6onp053e/netifaces/setup.py'"'"'; __file__='"'"'/private/var/folders/qm/9jnwq5m542q6qwwdpwm3dh300000gp/T/pip-install-6onp053e/netifaces/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/qm/9jnwq5m542q6qwwdpwm3dh300000gp/T/pip-wheel-j99r7gp4
       cwd: /private/var/folders/qm/9jnwq5m542q6qwwdpwm3dh300000gp/T/pip-install-6onp053e/netifaces/
  Complete output (16 lines):
  running bdist_wheel
  running build
  running build_ext
  checking for getifaddrs...not found.
  checking for getnameinfo...not found.
  checking for socket IOCTLs...not found.
  checking for optional header files...none found.
  checking whether struct sockaddr has a length field...no.
  checking which sockaddr_xxx structs are defined...none!
  checking for routing socket support...no.
  checking for sysctl(CTL_NET...) support...no.
  checking for netlink support...no.
  building 'netifaces' extension
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -I/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -DNETIFACES_VERSION=0.10.9 -I/usr/local/include -I/usr/local/opt/openssl@1.1/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python@3.9/3.9.0_1/Frameworks/Python.framework/Versions/3.9/include/python3.9 -c netifaces.c -o build/temp.macosx-10.14-x86_64-3.9/netifaces.o
  xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
  error: command '/usr/bin/clang' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for netifaces
  Running setup.py clean for netifaces
Failed to build netifaces
Installing collected packages: netifaces, BAC0
    Running setup.py install for netifaces ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/local/opt/python@3.9/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/qm/9jnwq5m542q6qwwdpwm3dh300000gp/T/pip-install-6onp053e/netifaces/setup.py'"'"'; __file__='"'"'/private/var/folders/qm/9jnwq5m542q6qwwdpwm3dh300000gp/T/pip-install-6onp053e/netifaces/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/qm/9jnwq5m542q6qwwdpwm3dh300000gp/T/pip-record-op6qbbia/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.9/netifaces
         cwd: /private/var/folders/qm/9jnwq5m542q6qwwdpwm3dh300000gp/T/pip-install-6onp053e/netifaces/
    Complete output (16 lines):
    running install
    running build
    running build_ext
    checking for getifaddrs...not found.
    checking for getnameinfo...not found.
    checking for socket IOCTLs...not found.
    checking for optional header files...none found.
    checking whether struct sockaddr has a length field...no.
    checking which sockaddr_xxx structs are defined...none!
    checking for routing socket support...no.
    checking for sysctl(CTL_NET...) support...no.
    checking for netlink support...no.
    building 'netifaces' extension
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -I/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -DNETIFACES_VERSION=0.10.9 -I/usr/local/include -I/usr/local/opt/openssl@1.1/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python@3.9/3.9.0_1/Frameworks/Python.framework/Versions/3.9/include/python3.9 -c netifaces.c -o build/temp.macosx-10.14-x86_64-3.9/netifaces.o
    xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
    error: command '/usr/bin/clang' failed with exit code 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/local/opt/python@3.9/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/qm/9jnwq5m542q6qwwdpwm3dh300000gp/T/pip-install-6onp053e/netifaces/setup.py'"'"'; __file__='"'"'/private/var/folders/qm/9jnwq5m542q6qwwdpwm3dh300000gp/T/pip-install-6onp053e/netifaces/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/qm/9jnwq5m542q6qwwdpwm3dh300000gp/T/pip-record-op6qbbia/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.9/netifaces Check the logs for full command output.

```

ChristianTremblay, 2021-01-26:
Hummm.. sorry about that. I should not have put netifaces in the requirements. My bad. I'll try to fix this ASAP.

ChristianTremblay, 2021-01-28:
Problem is that netifaces provide wheels for python <3.7
I modified the code so I will try the import only when it's needed (when trying to find the subnet mask). 
User will decide if he wants to install it but it's not mandatory. User can also provide subnet when declaring its address.

ChristianTremblay, 2021-02-02:
@georgegoglodze  Can you please confirm that the new Pypi version works for you ?

georgegoglodze, 2021-02-02:
Hello @ChristianTremblay, it it works!
Thanks!

closed 2021-02-02 by georgegoglodze