ChristianTremblay/BAC0 #245: Auto Save Function Error

kushan-26, 2021-01-25:
I have auto save function enable for a particular device but the history does not auto save after user defined polling cycles. However, history does save when save function manually executed for this device. I had other devices on the network performed as expected with auto save function. Please see the error message below: 
[image]

ChristianTremblay, 2021-01-26:
This is a super mega giga big error message :) Which is not really supposed to appear until thing goes really bad.... Can you tell me more about the use case and when / why / how it happened.

kushan-26, 2021-01-29:
Even I got scared after seeing the error message. But the only issue is auto save not working. I used following function inputs to bind the device. 
"AHU_RF= BAC0.device('203:2', 88902, bacnet, object_list = AHU_RF_List,
                            poll=300, auto_save=20,save_resampling='120s',
                            clear_history_on_save=True, segmentation_supported=False)"

This device does not support segmentation. So I called each point for polling after device is connected. However, after 20 polling cycles, history does not save to a sql database. Instead I get the subjected ERROR message. I checked point histories and they are logging values. So I used this function "AHU_RF.save(filename='AHU_RF',resampling='120s')" to save to database which executed without any error. All the history entries were present in the .db file

Moreover, I have other similar devices on the same network which did not give me this error, weird! 

I'm not sure what it means by "unable to open database file" but I could run some tests if you can point me to a direction.

ChristianTremblay, 2021-01-29:
I wonder if it's not the segmentation support false that leads to that....

kushan-26, 2021-01-29:
I doubt that because I had few other devices configured the exact same way and did not have this issue.  Looks like the name of the registered device is causing this issue. 

Save function works when I give a filename to save. Gives an error when asked to save with registered device name.
[image]

This is the registered device name. I think its breaking the save function.
[image]

ChristianTremblay, 2021-01-29:
Would be safer to use something else than the name in this case. I usually name my controller type-network-Mac which is always good... but real life isn't that easy. 

What would you suggest ?

Escaping characters ?
Using deviceId ?

kushan-26, 2021-01-29:
I have seen devices with the same name that live in same or two separate MS/TP networks. You cannot make that mistake with deviceId because its important for connectivity unlike the naming convention. In that case changing to deviceId would be safer. And of course you wont find any weird characters that can break your program either.

ChristianTremblay, 2021-02-02:
@kushan-26 can you check if the develop branch fixes your bugs ?

kushan-26, 2021-02-04:
> @kushan-26 can you check if the develop branch fixes your bugs ?

@ChristianTremblay Sorry for being late to check this out. Had few other things in my plate. I'll test this as soon as I can and let you know.

kushan-26, 2021-02-09:
@ChristianTremblay this issue is fixed by saving to database with deviceID instead of device name. Everything is working as expected.

A small personal suggestion, it would be neat if "Device_{}" is used instead of "dev_{}" in new script change. Not a big deal though. 

Thank you!

closed 2021-02-09 by ChristianTremblay