ChristianTremblay/BAC0 #246: Write description texts with spaces to BACnet objects

mosterme1003, 2021-01-26:
When I write a text with spaces to the description property I get an error because the spaces indicate more than one argument for the property value. Is there a way to fix this?

```
bacnet = BAC0.connect(ip=ipLocal)
bacnet.write('192.168.178.10 analogInput 1 description test text')
```
```
File "E:\Python\lib\site-packages\BAC0\core\io\Write.py", line 88, in write
    iocb = IOCB(self.build_wp_request(args, vendor_id=vendor_id))
  File "E:\Python\lib\site-packages\BAC0\core\io\Write.py", line 200, in build_wp_request
    obj_type, obj_inst, prop_id, value, priority, indx = self._parse_wp_args(args)
  File "E:\Python\lib\site-packages\BAC0\core\io\Write.py", line 131, in _parse_wp_args
    indx = int(args[4])
ValueError: invalid literal for int() with base 10: 'text'
```

ChristianTremblay, 2021-01-26:
that is a limit of the way this write function have been made. 
To be able to write to description, I would have to extract the value from the string and pass it as an arg.... 
In brief, rewrite the write function... I knew it would be necessary one day

I don't have the time right now.... but PR are welcome !

In the meantime, you could get it working building a requet and sending it to the stack

```
from bacpypes.object import get_datatype
from bacpypes.iocb import IOCB
from bacpypes.core import deferred
from bacpypes.apdu import (WritePropertyRequest)
from BAC0.core.io.Write.WriteProperty
from bacpypes.primitivedata import CharacterString
from bacpypes.constructeddata import Any
from bacpypes.pdu import Address

def build_write_request(addr='192.168.178.10', obj_type='analogInput', obj_inst=1, prop_id='description', value='my text'):
        request = WritePropertyRequest(
            objectIdentifier=(obj_type, obj_inst), propertyIdentifier=prop_id
        )
        request.pduDestination = Address(addr)

        _value = Any()
        _value.cast_in(CharacterString(value))
        request.propertyValue = _value 

        return request

def write_value(self, request, vendor_id=0, timeout=10):   
    try:
        iocb = IOCB(request)
        iocb.set_timeout(timeout)
        # pass to the BACnet stack
        deferred(bacnet.this_application.request_io, iocb)

    except WritePropertyException as error:
        # construction error
        print(("exception: {!r}".format(error))

    iocb.wait()  # Wait for BACnet response

    if iocb.ioResponse:  # successful response
        apdu = iocb.ioResponse

        if not isinstance(apdu, SimpleAckPDU):  # expect an ACK
            print("Not an ack, see debug for more infos.")
            return

    if iocb.ioError:  # unsuccessful: error/reject/abort
        apdu = iocb.ioError
        reason = find_reason(apdu)
        raise NoResponseFromController("APDU Abort Reason : {}".format(reason))

```

This is a draft version, you will need to work on it, but basically, it is all you need (or almost all you need)

Disclaimer : untested :)

gtamboise, 2021-02-10:
Marginally better, "it works on my computer" and against my PLC - using `update_description`:

```python
from bacpypes.iocb import IOCB
from bacpypes.core import deferred
from bacpypes.apdu import (
    WritePropertyRequest,
    SimpleAckPDU
)
from bacpypes.primitivedata import CharacterString
from bacpypes.constructeddata import Any
from bacpypes.pdu import Address
from BAC0.core.io.IOExceptions import (
    NoResponseFromController,
    WritePropertyException
)
from BAC0.core.io.Read import find_reason

def update_description(bacnet, device, obj_type, obj_inst, value):
    """ Updates the description for the bacnet object that belongs to the device
    device, in the application bacnet. The object has a type obj_type, its
    identifier BACNet is obj_inst and we want the new description to be value. """
    addr = device.properties.address.split(":")[0]
    request = build_write_request(addr=addr,
                                  obj_type=obj_type,
                                  obj_inst=obj_inst,
                                  prop_id='description',
                                  value=value)
    write_value(bacnet, request)

def build_write_request(addr='192.168.178.10', obj_type='analogInput',
                        obj_inst=1, prop_id='description', value='my text'):
    request = WritePropertyRequest(
        objectIdentifier=(obj_type, obj_inst), propertyIdentifier=prop_id
    )
    request.pduDestination = Address(addr)

    _value = Any()
    _value.cast_in(CharacterString(value))
    request.propertyValue = _value

    return request

def write_value(bacnet, request, vendor_id=0, timeout=10):
    try:
        iocb = IOCB(request)
        iocb.set_timeout(timeout)
        # pass to the BACnet stack
        deferred(bacnet.this_application.request_io, iocb)

        iocb.wait()  # Wait for BACnet response

        if iocb.ioResponse:  # successful response
            apdu = iocb.ioResponse

            if not isinstance(apdu, SimpleAckPDU):  # expect an ACK
                print("Not an ack, see debug for more infos.")
                return

        if iocb.ioError:  # unsuccessful: error/reject/abort
            apdu = iocb.ioError
            reason = find_reason(apdu)
            raise NoResponseFromController(
                "APDU Abort Reason : {}".format(reason))

    except WritePropertyException as error:
        # construction error
        print(("exception: {!r}".format(error)))
```

ChristianTremblay, 2021-02-10:
nice, what would you think if I'd put that somewhere in the stack ? 
I think it would fit under BAC0/core/functions
and could be part of any "point" which could lead to `dev['point'].update_description('My super long description with any $/% character in it')`

until I invest efforts in a new way of creating write requests....

gtamboise, 2021-02-11:
I am not super familiar with the stack, feel free to put that code wherever it makes sense for you.
Notice as well that I moved the `except` around because `iocb` gets created inside the `try`. How the code behaves if a `WritePropertyException` gets thrown has not been tested.

ChristianTremblay, 2021-02-11:
On va y aller comme ça et on corrigera au besoin. Merci pour la contribution

ChristianTremblay, 2021-02-12:
Reopening because someone should document this new feature.... Anyone ? :)

gtamboise, 2021-02-12:
Sure, I will be happy to help. You are expecting a pull request against one or more file(s) in `BAC0/doc/source/`, correct?

ChristianTremblay, 2021-02-12:
Correct. Against the develop branch. 

Thanks Guillaume

closed 2021-02-17 by ChristianTremblay