ChristianTremblay/BAC0 #247: Local COV

EARNSdev1, 2021-01-28:
Hi,
in a smal project I try to provide external datapoints to BACnet using BAC0.
I used YABE on an other machine to subscribe to my new datapoints (AV) and noticed COV related errors spawned by bacpypes. The reason are (None, (), {}) tuples in deferredFns in bacpypes.core. I guess you create them by accidently calling send_cov_notifications instead of just passing it.
https://github.com/ChristianTremblay/BAC0/blob/3d3509343de35f0721a468cd2048593f34de3a95/BAC0/tasks/UpdateCOV.py#L54

Passing the function itself seems to work.

ChristianTremblay, 2021-01-28:
hummm... it would make sense.... not executing the function right now.... wait for it, it's deferred...
Good catch my friend

EARNSdev1, 2021-01-28:
happy to help

closed 2021-02-02 by ChristianTremblay