ChristianTremblay/BAC0 #249: SQLite db continuously save the data?

bbartling, 2021-02-01:
Hello,

Would anyone know a way that the SQLite db get continuously appended when a script runs? Maybe I am missing something but it appears to only add data to the SQLite db (as well as create the db) when I disconnect from the controller.

I was interested in seeing if the db can be shared by another application to get the live polled data :)

```
import BAC0
import time

#build custom points list
my_obj_list = [('file', 1),
('analogInput', 2)]#tempUoTwoBalco

bacnet = BAC0.lite()

#print(bacnet.whois())
#print(bacnet.devices)

controller = BAC0.device('12345:2', 201201, bacnet,object_list=my_obj_list)

try:
    while True:
        print(controller['tempUoTwoBalco'].value)
        time.sleep(60)

except KeyboardInterrupt:
    pass

print("disconneting BACnet")
bacnet.disconnect()
```

bbartling, 2021-02-02:
Would anyone have any insight if this is funny at all to continuously save data with Pandas to an SQLite db with polling?

In BACnet polling is always better right? Vs performing BACnet reads every n-minutes to get data? Is polling easier on BACnet network traffic? Thanks...

```
import BAC0
import time
from datetime import datetime
import pandas as pd
from sqlalchemy import create_engine

'''
build custom points list
my device doesnt support segmentation
errors out without my_obj_list
'''

my_obj_list = [('file', 1),
('analogInput', 2)]#tempUoTwoBalco

bacnet = BAC0.lite()
#print(bacnet.whois())
#print(bacnet.devices)

#define BAC0 device to poll
controller = BAC0.device('12345:2', 201201, bacnet,object_list=my_obj_list)

#for SQLite automatically create db & append data
master_data = pd.DataFrame()

try:
    while True:
        sensor_value = controller['tempUoTwoBalco'].lastValue
        print(f'@ time {datetime.now()} sensor_value is {sensor_value}')

        engine = create_engine('sqlite:///save_data.db')
        sqlite_connection = engine.connect()
        sqlite_table = "all_data"
        master_data.to_sql(sqlite_table, sqlite_connection, if_exists='append')
        sqlite_connection.close()

        print(">>>>>>>>>>>Data saved to Sqlite<<<<<<<<<<<<<")
        time.sleep(60)

except Exception as error:
    print(error)

except KeyboardInterrupt:
    pass

print("disconneting BACnet")
bacnet.disconnect()

```

ChristianTremblay, 2021-02-02:
This is the `auto_save` parameter when you define a device.
Issue #245 reported a bug and develop version should cover the fix.

Regarding polling, not sure I understand your question. I used the term polling because when a device is created, it is possible to generate big requests in the form of readMultiple requests so I could get access to updated presentValues for a bunch of points at the same time.  
Polling is in fact in opposition to COV. 

- Polling : hey, how are you ? ..... hey, how are you ? hey, how are you ? ..... hey, how are you ? ..... hey, how are you ? .....
- COV : tell me if you ever feel bad, I'll be there for you

The save to SQLite in BAC0 just transfer the in-memory data in the form of Pandas Series to the db. When required. By default, when you disconnect (which is not super practical if your script is meant to run all the time...)

To help with long-term running script, device have those parameters : 

-  auto_save=False,
-  save_resampling="1s",
-  clear_history_on_save=False,
-  history_size=None,

So you can limit the history size to a fixed number of records (and preserve memory)
You can clear the in-memory history when you save to disk
Change the resampling for the data saved on the db (if you need 15-min data.... no need to over populate the DB)

NubeDev, 2021-02-02:
@bbartling 
I would also consider the `db` getting locked due to your code making a new connection to the db if you wanna use your code in production, Issue being that `BAC0` could be writing to the DB aswell.

https://stackoverflow.com/questions/1005206/does-sqlite-lock-the-database-file-on-reads

ChristianTremblay, 2021-02-02:
The basic idea behind the SQLite usage was not to have a full-proof db feature... it was a simple way to save the data from short periods while using the script. 

That said, I chose SQLite because it was opening the door to an eventual support of more robust architecture (like a real SQL server). 

But to reach this point, there would be a lot of work... and I don't have the time / knowledge / experience to go down this rabbit hole. 

The people I know that use BAC0 for very long period of time often use their own solution to connect to a dB of their choice using something else in parallel of BAC0.

bbartling, 2021-02-03:
Thanks guys I am going to close this issue for now, got a lot to learn :)

@NubeDev do you have any sql scripts like this where BAC0 can be ran for long periods of time to continuously append data to an sql db?

closed 2021-02-03 by bbartling

NubeDev, 2021-02-03:
So script's arent really made for running for long periods of time. 
if you want a quick and easy i only have this https://github.com/NubeDev/BAC0-MQTT/blob/main/bacnet_mqtt_gateway.py
Then from there options are endless as there are many MQTT apps and bridges and or node-red you could use.