ChristianTremblay/BAC0 #254: Use BAC0 to setup local network on pc for development?

DrawMen, 2021-02-11:
Hi there,
For my thesis I need to communicate with certain devices on the BACnet network in my school. My problem is that I dont have acces to this network right now. Is it possible to use BAC0 to mock the network on my pc at home? From what I have read it seems that you have to be connected to a BACnet network and can't host your own?

PS This is my first time with python and BAC0, so I'm a noob ;)

NubeDev, 2021-02-11:
Ya can use bacnet-stack to run a simple demo server.

Downlaod it onto a PI and unzip it and build it
```
wget https://github.com/bacnet-stack/bacnet-stack/archive/bacnet-stack-1.0.0.zip
make clean all
cd bin
./bacserve
```

DrawMen, 2021-02-11:
Is there any way to do that on a windows machine? I got a PI to use, but the delivery is delayed.

NubeDev, 2021-02-11:
To start you need 2 machines 1 for server and one for master.

Sorry i havent used windows in years

DrawMen, 2021-02-11:
Ok Thanks for the answer, I thought that was the case. I gues I also need a switch between my PI and pc?

ChristianTremblay, 2021-02-12:
# Setup 2 IPs on your network interface
In Windows, you can define 2 IPs for the same interface.
So configure 2 different static IPs on your network card.
[image]
Here I chose this
[image]

# Setup 2 BAC0 instances
Then use 2 command prompt and launch ipython on both. Declare two instances of BAC0 using each one of the IP defined. Do not forget the subnet mask
[image]

Voilà, you now have 2 BACnet devices that can interact together
[image]

# Add objects to one of the device
You can also define objects and pretend your something...
```
from BAC0.core.devices.local.models import analog_input

_new_object = analog_input(
    instance=1,
    name="Current_Temp",
    description="Current Temperature in degC",
    presentValue=19.2,
    properties={"units": "degreesCelsius"},
    )

_new_object.add_objects_to_application(bacnet)
# in this context, bacnet variable is the name of the instance of BAC0 running... which is a BACnet device itself
```

Register the other object... so it learn the object
[image]

## Make things change
Using a recurring task is an easy way to update the value of a value... here I use random but it could be anything. GPIO value, CPU load, etc...
[image]

Have fun !

ChristianTremblay, 2021-02-12:
Should probably copy that in the doc....

DrawMen, 2021-02-12:
Wow I didn't know that, thanks for the help!

closed 2021-02-13 by DrawMen