ChristianTremblay/BAC0 #258: How to write calendar entries to dateList?

mosterme1003, 2021-02-17:
Hey! I would like to add entries to a calendar's property `dateList`. Now I have a list with multiple `bacpypes.basetypes.CalendarEntry`. How can I write it to  `dateList`? I tried to figure out how you did similar things in the `Schedule.py` but I don't really get how this whole request building and sending processes work, especially when I'm sending a list or an array. Can you help me? Is there an easy way for me to write dates to a calendar?

mosterme1003, 2021-02-17:
I think I got it to work for me. Is this a clean way to do it?

```
from bacpypes.core import deferred
from bacpypes.pdu import Address
from BAC0.core.io.Read import find_reason
from BAC0.core.io.IOExceptions import NoResponseFromController, WritePropertyException
from bacpypes.apdu import WritePropertyRequest, SimpleAckPDU
from bacpypes.iocb import IOCB
from bacpypes.constructeddata import ArrayOf, Any
from bacpypes.basetypes import CalendarEntry
import BAC0

class Calendar:
    DateList = ArrayOf(CalendarEntry)

    def build_write_request(self, deviceIp, calendarInstance, dateList):
        request = WritePropertyRequest(
            objectIdentifier=("calendar", calendarInstance),
            propertyIdentifier="dateList"
        )
        request.pduDestination = Address(deviceIp)

        _value = Any()
        _value.cast_in(dateList)
        request.propertyValue = _value

        return request

    def write_value(self, bacnet, request, vendor_id=0, timeout=10):
        try:
            iocb = IOCB(request)
            iocb.set_timeout(timeout)
            # pass to the BACnet stack
            deferred(bacnet.this_application.request_io, iocb)

            iocb.wait()  # Wait for BACnet response

            if iocb.ioResponse:  # successful response
                apdu = iocb.ioResponse

                if not isinstance(apdu, SimpleAckPDU):  # expect an ACK
                    print("Not an ack, see debug for more infos.")
                    return

            if iocb.ioError:  # unsuccessful: error/reject/abort
                apdu = iocb.ioError
                reason = find_reason(apdu)
                raise NoResponseFromController(
                    "APDU Abort Reason : {}".format(reason))

        except WritePropertyException as error:
            # construction error
            print(("exception: {!r}".format(error)))

    def write_calendar(self, localIp, deviceIp, calendarInstance, dateList):
        bacnet = BAC0.connect(ip=localIp)
        request = self.build_write_request(deviceIp=deviceIp, calendarInstance=calendarInstance, dateList=dateList)
        self.write_value(bacnet, request)

if __name__ == '__main__':
    dt1 = (121, 2, 19, 255)
    dt2 = (121, 2, 20, 255)
    listEntries = [CalendarEntry(date=dt1), CalendarEntry(date=dt2)]
    dateList = Calendar.DateList(listEntries)

    Cal = Calendar()
    Cal.write_calendar(localIp="192.168.178.2", deviceIp="192.168.178.3", calendarInstance=10104, dateList=dateList)

```

ChristianTremblay, 2021-02-17:
Now that the basic mechanism works, I'd like to think about the way BAC0 should represent the calendar, preferably with something like a dict, like schedules. So it's easy to read or write.

I'm curious also to see if we could use the `calendar` python package(https://github.com/python/cpython/blob/3.9/Lib/calendar.py)

Would be cool to have access to a calendar view...

mosterme1003, 2021-03-01:
I didn't look into `calendar.py` yet but I thought about a way to represent the calendar object's datelist as a dict. I also wrote the `write_calendar_dateList()` and `read_calendar_dateList()` functions that could be implemented in BAC0. Currently supported are (non-)recurring dates and date ranges but  not yet the weeknday option of the calendar entry. What do you think?
```python
from bacpypes.core import deferred
from bacpypes.pdu import Address
from BAC0.core.io.Read import find_reason
from BAC0.core.io.IOExceptions import NoResponseFromController, WritePropertyException
from bacpypes.apdu import WritePropertyRequest, SimpleAckPDU
from bacpypes.iocb import IOCB
from bacpypes.constructeddata import ArrayOf, Any
from bacpypes.basetypes import CalendarEntry, DateRange

import BAC0
import datetime

class Calendar:
    """
    Everything you need to write dates and date ranges to a calendar object.
    """

    DateList = ArrayOf(CalendarEntry)
    datelist_example = {
        "dates": [
            {
                "date": "2021/3/14",
                "recurring": False,
            },
            {
                "date": "2021/3/10",
                "recurring": True,
            },
        ],
        "dateRanges": [
            {
                "startDate": "2021/3/16",
                "endDate": "2021/3/21",
            },
            {
                "startDate": "2021/3/5",
                "endDate": "2021/3/7",
            },
        ]
    }

    def build_write_request(self, deviceIp, calendar_instance, dateList):
        request = WritePropertyRequest(
            objectIdentifier=("calendar", calendar_instance),
            propertyIdentifier="dateList"
        )
        request.pduDestination = Address(deviceIp)

        _value = Any()
        _value.cast_in(dateList)
        request.propertyValue = _value

        return request

    def write_value(self, bacnet, request, vendor_id=0, timeout=10):
        try:
            iocb = IOCB(request)
            iocb.set_timeout(timeout)
            # pass to the BACnet stack
            deferred(bacnet.this_application.request_io, iocb)

            iocb.wait()  # Wait for BACnet response

            if iocb.ioResponse:  # successful response
                apdu = iocb.ioResponse

                if not isinstance(apdu, SimpleAckPDU):  # expect an ACK
                    print("Not an ack, see debug for more infos.")
[... 75 lines cut ...]
                    weekday = 255
                else:
                    weekday = datetime.date(year, month, day).weekday() + 1
                    if weekday > 7: weekday = 1
                _date = (year - 1900, month, day, weekday)
                entries.append(CalendarEntry(date=_date))

        if "dateRanges" in dateList_dict.keys():
            for date_range_entry in dateList_dict["dateRanges"]:
                year, month, day = (int(x) for x in date_range_entry["startDate"].split('/'))
                weekday = datetime.date(year, month, day).weekday() + 1
                if weekday > 7: weekday = 1
                start_date = (year - 1900, month, day, weekday)

                year, month, day = (int(x) for x in date_range_entry["endDate"].split('/'))
                weekday = datetime.date(year, month, day).weekday() + 1
                if weekday > 7: weekday = 1
                end_date = (year - 1900, month, day, weekday)

                date_range = DateRange(startDate=start_date, endDate=end_date)
                entries.append(CalendarEntry(dateRange=date_range))

        dateList = self.DateList(entries)
        return dateList
```

ChristianTremblay, 2021-05-07:
I've been on other things lately. Sorry. I'll find time to look at this.

ChristianTremblay, 2021-05-29:
I'm actually adding this almost as-is
Thanks

closed 2021-08-05 by github-actions[bot]