ChristianTremblay/BAC0 #259: connection issue with raspberry

ArthurEmcon, 2021-02-24:
hello,

I am having an issue when attempting to connect my raspberry PI to BACnet. I can successfully connect my desktop to the network, but not the RPI.

The picture that follows show how each device is seen by the workstation. The desktop has its description pointing to this github webpage, however the RPI shows "NR NULL".

[image]

Despite both being seen by the workstation, the difference between them is my concern, has two aspects:

a) The RPI is seen by the workstation much later than the desktop, after the BAC0.connect() command is entered;

b) The desktop can successfully execute the .read() command, and returns present values of objects from other devices, while the RPI doesn't.

Thanks in advance, and sorry if this issue may not be exactly related to the programming.

ChristianTremblay, 2021-02-25:
Be sure that the gateway is setup correctly in the Pi.
Be sure to specify the subnet mask when you create your BACnet instance.

Use `BAC0.lite(ip='192.168.1.2/24')`
Connect is full of stuff you probably don't need.

Be sure no firewall block port UDP 47808

ArthurEmcon, 2021-02-27:
Hi Christian, thanks for the response

I think the problem was a blocking of port UDP 47808, but it was not because of a firewall, but because another process called "svn" was using it. I detected it with the command "sudo fuser 47808/udp", and eliminated it.

The next week I will have the opportunity to test it again, and to report here if the issue was closed.

ArthurEmcon, 2021-02-27:
Meanwhile, another issue is that I can't follow the documentation example named "How to define a device and interact with points". The picture shows the snipped of the example

[image]

When I enter the command "mycontroller = BAC0.device('3:4', 5504, bacnet)", the controller gets disconnected status, and doesn't appear in the list of registered devices with "bacnet.registered_devices".

ChristianTremblay, 2021-03-26:
Because you need a real controller at this specific address on your network.
Maybe I should have said 
# Try to connect to a controller... instead of "define"

ArthurEmcon, 2021-03-26:
ah ok, understood. Thanks for the responses!

ArthurEmcon, 2021-03-26:
ops, just reopened this thread to state that the solution of the following comment of mine worked! So this issue can be closed

> Hi Christian, thanks for the response
>
> I think the problem was a blocking of port UDP 47808, but it was not because of a firewall, but because another process called "svn" was using it. I detected [...]

closed 2021-03-26 by ArthurEmcon