ChristianTremblay/BAC0 #261: Exception on calling .disconnect() twice

Safrone, 2021-02-26:
```
In [11]: bacnet = BAC0.connect()
In [12]: bacnet.disconnect()

In [13]: bacnet.disconnect()
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-13-4c4805f0a1d6> in <module>
----> 1 bacnet.disconnect()

~/Env/data_handler/lib/python3.5/site-packages/BAC0/scripts/Lite.py in disconnect(self)
    414         for each in self.registered_devices:
    415             each.disconnect()
--> 416         super().disconnect()
    417
    418     def __repr__(self):

~/Env/data_handler/lib/python3.5/site-packages/BAC0/scripts/Base.py in disconnect(self)
    269         self.t.join()
    270         self._started = False
--> 271         Base._used_ips.remove(self.localIPAddr)
    272         self._log.info("BACnet stopped")
    273

KeyError: <Address 192.168.7.110>
```

probably want to replace `remove` with `discard`

ChristianTremblay, 2021-03-26:
I didn't know about discard ! Thanks

ChristianTremblay, 2021-05-28:
integrated to develop branch

closed 2021-05-28 by ChristianTremblay