ChristianTremblay/BAC0 #263: Pb with create a controller

KostBMF, 2021-03-05:
Hello

I'v got a probleme when I'm trying to create a controller

I write this : mycontroller = BAC0.device('12:1', 4194303, bacnet)

at the execution, i'v got this messages 

2021-03-05 10:55:47,659 - INFO    | Device 4194303:[iEM3165_0019175015] found... building points list
2021-03-05 10:55:48,714 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2021-03-05 10:55:49,115 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2021-03-05 10:55:49,504 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2021-03-05 10:55:49,897 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2021-03-05 10:55:50,296 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2021-03-05 10:55:50,680 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2021-03-05 10:56:12,922 - WARNING | The inactiveText property is not implemented in the device. Using a default value of Off for internal needs.
2021-03-05 10:56:13,052 - WARNING | The activeText property is not implemented in the device. Using a default value of On for internal needs.
2021-03-05 10:56:13,457 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2021-03-05 10:56:13,595 - WARNING | The inactiveText property is not implemented in the device. Using a default value of Off for internal needs.
2021-03-05 10:56:16,001 - INFO    | Ready!
2021-03-05 10:56:16,001 - INFO    | Device defined for normal polling with a delay of 10sec
2021-03-05 10:56:16,001 - INFO    | Polling started, values read every 10 seconds
connected [for ReadProperty]

KostBMF, 2021-03-18:
Then, when I'm trying a  mycontroller.points,  nothing happened.

ChristianTremblay, 2021-03-22:
What is the output of 
```
bacnet.read('12:1 device 4194303 objectList')
```

closed 2021-07-13 by github-actions[bot]