ChristianTremblay/BAC0 #265: Cannot run on two different interfaces at the same time

Safrone, 2021-03-18:
```
> bacnet1 = BAC0.connect(ip="192.168.1.111/24")
2021-03-18 19:56:44,125 - INFO    | Starting BAC0 version 21.02.25 (Lite)
2021-03-18 19:56:44,125 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2021-03-18 19:56:44,125 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2021-03-18 19:56:44,125 - INFO    | Starting TaskManager
2021-03-18 19:56:44,125 - INFO    | Using ip : 192.168.1.111
2021-03-18 19:56:44,126 - INFO    | Starting app...
2021-03-18 19:56:44,127 - INFO    | BAC0 started
2021-03-18 19:56:44,127 - INFO    | Registered as Simple BACnet/IP App
2021-03-18 19:56:44,129 - INFO    | Update Local COV Task started
> bacnet2 = BAC0.connect(ip='192.168.250.1/24')
2021-03-18 19:56:46,525 - INFO    | Starting BAC0 version 21.02.25 (Lite)
2021-03-18 19:56:46,525 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2021-03-18 19:56:46,525 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2021-03-18 19:56:46,525 - INFO    | Using ip : 192.168.250.1
2021-03-18 19:56:46,528 - INFO    | Starting app...
2021-03-18 19:56:46,528 - INFO    | BAC0 started
2021-03-18 19:56:46,528 - INFO    | Registered as Simple BACnet/IP App
2021-03-18 19:56:46,530 - INFO    | Update Local COV Task started
an error has occurred: no adapters
Traceback (most recent call last):
  File "/home/user/data_handler/venv/lib/python3.6/site-packages/bacpypes/core.py", line 165, in run
    fn(*args, **kwargs)
  File "/home/user/data_handler/venv/lib/python3.6/site-packages/bacpypes/service/device.py", line 32, in startup
    self.i_am()
  File "/home/user/data_handler/venv/lib/python3.6/site-packages/bacpypes/service/device.py", line 133, in i_am
    self.request(iAm)
  File "/home/user/data_handler/venv/lib/python3.6/site-packages/BAC0/core/app/ScriptApplication.py", line 246, in request
    super(BAC0Application, self).request(apdu)
  File "/home/user/data_handler/venv/lib/python3.6/site-packages/bacpypes/app.py", line 485, in request
    super(ApplicationIOController, self).request(apdu)
  File "/home/user/data_handler/venv/lib/python3.6/site-packages/bacpypes/app.py", line 364, in request
    super(Application, self).request(apdu)
  File "/home/user/data_handler/venv/lib/python3.6/site-packages/bacpypes/comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "/home/user/data_handler/venv/lib/python3.6/site-packages/bacpypes/appservice.py", line 1508, in sap_indication
    self.request(xpdu)
  File "/home/user/data_handler/venv/lib/python3.6/site-packages/bacpypes/comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "/home/user/data_handler/venv/lib/python3.6/site-packages/bacpypes/appservice.py", line 1329, in sap_indication
    self.request(apdu)
  File "/home/user/data_handler/venv/lib/python3.6/site-packages/bacpypes/comm.py", line 282, in request
    self.clientPeer.indication(*args, **kwargs)
  File "/home/user/data_handler/venv/lib/python3.6/site-packages/bacpypes/netservice.py", line 334, in indication
    raise ConfigurationError("no adapters")
bacpypes.errors.ConfigurationError: no adapters
```

ChristianTremblay, 2021-03-20:
Can't reproduc here : 
```
In [1]: import BAC0

In [2]: bacnet = BAC0.lite()
2021-03-20 11:37:24,999 - INFO    | Starting BAC0 version 21.02.25 (Lite)
2021-03-20 11:37:24,999 - INFO    | Use BAC0.log_level to adjust verbosity of the app.      
2021-03-20 11:37:24,999 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2021-03-20 11:37:24,999 - INFO    | Starting TaskManager
2021-03-20 11:37:25,192 - INFO    | Using ip : 192.168.211.54
2021-03-20 11:37:25,193 - INFO    | Starting app...
2021-03-20 11:37:25,194 - INFO    | BAC0 started
2021-03-20 11:37:25,194 - INFO    | Registered as Simple BACnet/IP App
2021-03-20 11:37:25,220 - INFO    | Update Local COV Task started

In [3]: bacnet2 = BAC0.lite(ip='192.168.212.12/24')
2021-03-20 11:37:43,430 - INFO    | Starting BAC0 version 21.02.25 (Lite)
2021-03-20 11:37:43,430 - INFO    | Use BAC0.log_level to adjust verbosity of the app.      
2021-03-20 11:37:43,431 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2021-03-20 11:37:43,431 - INFO    | Using ip : 192.168.212.12
2021-03-20 11:37:43,433 - INFO    | Starting app...
2021-03-20 11:37:43,434 - INFO    | BAC0 started
2021-03-20 11:37:43,434 - INFO    | Registered as Simple BACnet/IP App
2021-03-20 11:37:43,464 - INFO    | Update Local COV Task started

In [4]: bacnet2.discover()
2021-03-20 11:38:06,853 - INFO    | Found those networks : set()
2021-03-20 11:38:06,854 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
Out[4]: []

In [5]: bacnet.discover()
2021-03-20 11:38:13,127 - INFO    | 192.168.211.197 network number is 1
2021-03-20 11:38:13,143 - WARNING | 192.168.211.3 Rejected message to network (reason : It is an unknown network layer message)
2021-03-20 11:38:13,158 - INFO    | 192.168.210.253 network number is 1
2021-03-20 11:38:14,610 - INFO    | 192.168.211.3 router to [303]
2021-03-20 11:38:14,626 - INFO    | 192.168.210.253 router to [2]
2021-03-20 11:38:16,555 - INFO    | Addr : 192.168.210.253
2021-03-20 11:38:16,624 - INFO    | 192.168.210.253 routing table
2021-03-20 11:38:16,625 - INFO    |     2 1 bytearray(b'')
2021-03-20 11:38:16,625 - INFO    |     1 2 bytearray(b'')
2021-03-20 11:38:18,582 - INFO    | Found those networks : {1, 2, 303}
2021-03-20 11:38:18,583 - INFO    | Discovering network 1
2021-03-20 11:38:21,625 - INFO    | Discovering network 2
2021-03-20 11:38:24,775 - INFO    | Discovering network 303
Out[5]: 
[('192.168.211.197', 1098300),
 ('192.168.211.3', 81113),    
 ('192.168.210.253', 210253),
 ('2:5', 5),
 ('303:5', 560004),
 ('303:7', 5007),
 ('303:8', 72008),
 ('303:9', 5009),
 ('303:10', 60010),
 ('303:12', 5012),
 ('303:13', 5213),
 ('303:33', 5004),
 ('303:115', 80000)]

In [6]: 
```

Safrone, 2021-04-22:
@ChristianTremblay hmm, I was testing recently and it was still happening, then I updated bacpypes 0.18.3 -> 0.18.4 and it started working...

if you can test with 0.18.3 to confirm if I'm just crazy I would appreciate knowing if it was just me after all

otherwise seems like whatever the issue was is fixed so alls well right?
Thanks!

ChristianTremblay, 2021-05-07:
Pretty sure your not crazy.....and we'll let 0.18.3 in the past. :-)

closed 2021-05-07 by ChristianTremblay