ChristianTremblay/BAC0 #266: How to create a BACnet Analog Value witch BAC0 in Python

Obaq-web, 2021-03-19:
To start off I have to say that I am a beginner when it comes to Python and the BAC0 library.

I want to create a Bacnet Analog Value with my python script that is supposed to be sent to another (physical) Bacnet-device. For now I have only "created" a Bacnet-device with the bacnet = BAC0.lite() command. It's not much but it works.

I spend a lot of time browsing the documentation but I can not find the right way to let my script send an Analog Value.

Can anyone help?

Background: I have a device that is able to send an analog value (0 ... 10 V) to an MQTT-broker very easiliy. Now I want to let a Raspi recieve that analog value and "transform" it into a Bacnet Analog Value. This Bacnet Analog Value will be sent to a DDC in order to controll the power of a pump. To do this I need the correct command / code that "creates" the Bacnet Analog Value in the python script.

Obaq-web, 2021-03-20:
I managed to creat an analog value with these commands:

```
temp = create_AV(oid=12, name="temp", pv=22, pv_writable=True)
temp.units = EngineeringUnits.enumerations["degreesCelsius"]
temp.description = "AVG Temp"
hvac.this_application.add_object(temp)
```

I hope this may help someone.

ChristianTremblay, 2021-03-22:
Hi @Obaq-web 

This is something I promised myself to do.... documenting how to create local objects and BAC0 evolved recently to use a way I found easier.

Those are some extract from a weather script I use.

```
from BAC0.core.devices.local.models import (
    analog_input,
    datetime_value,
    character_string,
)

print("Starting BACnet device")
new_device = BAC0.lite(deviceId=10031, port=47809)
time.sleep(1)

# Analog Values
_new_objects = analog_input(
    instance=1,
    name="Current_Temp",
    description="Current Temperature in degC",
    presentValue=0,
    properties={"units": "degreesCelsius"},
)
_new_objects = analog_input(
    instance=2,
    name="Current_Pressure",
    description="Current Pressure in kPa",
    presentValue=0,
    properties={"units": "kilopascals"},
)
# Character Strings
_new_objects = character_string(
    instance=1,
    name="City",
    description="City code for data",
    presentValue="on-24",
    is_commandable=True,
)
# DateTime Values
_new_objects = datetime_value(instance=1, name="Last_Update")

# This line will add all objects to the device
_new_objects.add_objects_to_application(new_device)
```

Now if you need in your script to modify the presentValue...
```
new_device["Current_Temp"].presentValue = 20
new_device["Current_Temp"].statusFlags = [0, 0, 0, 0]
new_device['City'].presentValue = = CharacterString('myTown')
```

There is a new object in this logic which is a factory that build the local objects using the parameters provided. I tried to make the most complete object possible (make sure there is a statusFlags for example, or a description, etc.). Also tried to simplify making the objects commandable (adding all that was necesseray like the priorityArray, etc...

Hope you'll find that useful

Obaq-web, 2021-03-25:
This is very helpful, thanks. Im combination with an MQTT-module the BAC0-module offers a solution to a technical problem I had concerning data transfer from a sensor to a pump. Thank you very much for the great work!

closed 2021-05-07 by ChristianTremblay