ChristianTremblay/BAC0 #269: How to write complex datatypes such as Object_Property_Reference of the Event Enrollment Object.

Moravien, 2021-03-25:
## Introduction:

Hello,

First, I would like to personally thank you so much for putting BAC0 together 👍 🥇 !!  I really enjoy the simple setup of this tool to help me do some basic BACnet Testing.  It has saved me a great deal of time and effort.

## Problem:

I have been trying to figure out how to write a complex attribute such as **Object_Property_Reference** of the **Event Enrollment Object Type**.

From my limited understanding of BAC0, I currently do not see a simple way to write this or other complex types using the current implementation of the write function `bacnet.write()`  in `Write.py`.  From my understanding it looks like all of the parameters for this function to work need to be passed in as a single string separated by spaces for the different arguments.

## Reference:

For reference this property is structured as the following:

```
BACnetDeviceObjectPropertyReference ::= SEQUENCE {
      object-identifier     [0]     BACnetObjectIdentifier,
      property-identifier   [1]     BACnetPropertyIdentifier,
      property-array-index  [2]     Unsigned OPTIONAL, -- used only with array datatype
                                                       -- if omitted with an array then
                                                       -- the entire array is referenced
      device-identifier     [3]     BACnetObjectIdentifier  OPTIONAL
}
```

## Questions:

My questions are as follows:

1. Can complex datatypes be written using the current version of BAC0?
2. If the answer above is **yes** can you please provide me with a simple example of this I can follow?
3. If the answer above is **no** is there a roadmap or plan to support this type of feature in the future?

## Closing Remarks:

In closing, thank you again for supporting and putting this awesome tool together!  I am really looking forward to your response.  Thanking you in advance!

Best Regards,

Moravien

ChristianTremblay, 2021-03-26:
`write` is not made for that you are right.
For complex data structure, a new class is needed.

You could refer to 

https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/functions/Schedule.py

for an example.

I usually develop feature when I need them and I must admit that I never use events... But I gladly take PR and I can help if required.

Moravien, 2021-03-26:
Thank you for the quick response and providing the example!  I will take a look at this and study it up a bit to see how this might be achieved.  If I get something working I can put it up for your review to help get this useful feature in thanks again! 👍 

Moravien

ChristianTremblay, 2021-03-31:
Let me know, it would be nice to have something for events already made.

closed 2021-09-12 by github-actions[bot]