ChristianTremblay/BAC0 #270: ASHRAE Trim respond logic helper

bbartling, 2021-04-10:
Hey Christian,

Any chance I could get a tip for how to butter up this code below? I am hoping to create a "dummy" BACnet device that could be used to test AHU duct pressure trim respond logic. Like JACE supervisory level ladder logic...

I know there is probably a much better way to write this but when I run the code I can only get `DPR1` to accept the `randrange` value from `recurring task`... The writeable AV for duct pressure setpoint appears to work fine when tested from a BACnet sniffing tool.

```
import BAC0
import random
import time
from BAC0.core.devices.create_objects import create_AO,create_AV
from BAC0.tasks.RecurringTask import RecurringTask 
from bacpypes.primitivedata import Real
from BAC0.core.devices.create_objects import create_AI

def changeValueOfMyAO():
     obj = bacnet.this_application.get_object_name('DPR1-O')
     value = obj.ReadProperty('presentValue').value
     value = random.randrange(5, 95, 1)
     obj.presentValue = Real(value)

bacnet = BAC0.lite()
bacnet.this_application.add_object(create_AO(oid=0, name='DPR1-O', pv=Real(25)))
bacnet.this_application.add_object(create_AO(oid=1, name='DPR2-O', pv=Real(35)))
bacnet.this_application.add_object(create_AO(oid=2, name='DPR3-O', pv=Real(45)))
bacnet.this_application.add_object(create_AO(oid=3, name='DPR4-O', pv=Real(55)))
bacnet.this_application.add_object(create_AO(oid=4, name='DPR5-O', pv=Real(65)))
bacnet.this_application.add_object(create_AV(oid=5, name='DAP-SP',  pv=1, pv_writable=True))

def main():
    task = RecurringTask(changeValueOfMyAO,delay=5)
    task.start()
    while True:
        dpr1 = bacnet.this_application.get_object_name('DPR1-O')
        print('Damper 1 value is {}'.format(dpr1.presentValue))
        dpr2 = bacnet.this_application.get_object_name('DPR2-O')
        print('Damper 2 value is {}'.format(dpr2.presentValue))
        dpr3 = bacnet.this_application.get_object_name('DPR3-O')
        print('Damper 3 value is {}'.format(dpr3.presentValue))
        dpr4 = bacnet.this_application.get_object_name('DPR4-O')
        print('Damper 4 value is {}'.format(dpr4.presentValue))
        dpr5 = bacnet.this_application.get_object_name('DPR5-O')
        print('Damper 5 value is {}'.format(dpr5.presentValue))

        duct_pres_sp = bacnet.this_application.get_object_name('DAP-SP')
        print('Duct Pressure Setpoint is {}'.format(duct_pres_sp.presentValue))
        time.sleep(10)

if __name__ == "__main__":
    main()
```

Ive also been noticing this for a while too :)

```
********************************************************************************
create_xx functions are deprecated and will disappear from a future release
BAC0.core.device.local.object using the ObjectFactory will be the new way to define objects
Refer to the doc for details
********************************************************************************
********************************************************************************
create_xx functions are deprecated and will disappear from a future release
BAC0.core.device.local.object using the ObjectFactory will be the new way to define objects
Refer to the doc for details
********************************************************************************
```

bbartling, 2021-04-11:
I think I maybe good to go on this one until `ObjectFactory` requirement :)

closed 2021-04-11 by bbartling

ChristianTremblay, 2021-04-12:
@bbartling 
This is an extract from conftext.py in the test suite.... I'm pretty sure you will see the idea.
```
from BAC0.core.devices.local.models import (
    analog_input,
    analog_output,
    analog_value,
    binary_input,
    binary_output,
    binary_value,
    multistate_input,
    multistate_output,
    multistate_value,
    date_value,
    datetime_value,
    temperature_input,
    temperature_value,
    humidity_input,
    humidity_value,
    character_string,
    make_state_text
)

_new_objects = analog_input(
        name="ZN-T",
        properties={"units": "degreesCelsius"},
        description="Zone Temperature",
        presentValue=21,
    )

    states = make_state_text(["Normal", "Alarm", "Super Emergency"])
    _new_objects = multistate_value(
        description="An Alarm Value",
        properties={"stateText": states},
        name="BIG-ALARM",
        is_commandable=True,
    )

_new_objects.add_objects_to_application(device)

```

bbartling, 2021-04-15:
For what its worth, this appears to be working for an AV or AO:

```
import BAC0
import random
import time

from BAC0.tasks.RecurringTask import RecurringTask 
from bacpypes.primitivedata import Real

from BAC0.core.devices.local.models import (
    analog_output,
    analog_value
    )

def changeValueOfMyAO1():
     obj = bacnet.this_application.get_object_name('DPR1-O')
     value = obj.ReadProperty('presentValue').value
     value = random.randrange(5, 95, 1)
     obj.presentValue = Real(value)

     

bacnet = BAC0.lite()

_new_objects = analog_output(
        name="DPR1-O",
        properties={"units": "percent"},
        description="VAV Box 1 Damper Output",
        presentValue=50,is_commandable=False
    )

_new_objects = analog_value(
        name="DAP-SP",
        properties={"units": "inchesOfWater"},
        description="AHU Duct Pressure Setpoint",
        presentValue=1,is_commandable=True
    )

_new_objects.add_objects_to_application(bacnet)

def main():
    task1 = RecurringTask(changeValueOfMyAO1,delay=3)
    task1.start()

    while True:
        dpr1 = bacnet.this_application.get_object_name('DPR1-O')
        bacnet._log.info('VAV damper 1 position is {}'.format(dpr1.presentValue))

        duct_pres_sp = bacnet.this_application.get_object_name('DAP-SP')
        bacnet._log.info('Duct Pressure Setpoint is {}'.format(duct_pres_sp.presentValue))
        time.sleep(5)

if __name__ == "__main__":
    main()
```