ChristianTremblay/BAC0 #271: APDU Timeout

KostBMF, 2021-04-12:
Hello
I have a Schneider device connected to a BAS Router 
When the router is connected to my freebox router, I can communicate with my device via read requests, whois, etcc..
I run my scripts from a home automation solution, and everything is fine

However, when I'm at work, when I want to use the same scripts, I get APDU Timeout errors

2021-04-12 14:14:58][DEBUG]: ###################################
2021-04-12 14:14:58][DEBUG]: # Read property
[2021-04-12 14:14:58][DEBUG]: ###################################
[2021-04-12 14:14:58][DEBUG]: ['12:1', 'analogInput', '18', 'presentValue']
[2021-04-12 14:14:58][DEBUG]: ###################################

2021-04-12 14:15:08][ERROR] : Send command to demon error : APDU Abort Reason : Timeout

The router is connected to a switch at my work
This is the only difference I can notice
If I use my script from PyCharm for example, the communications go through. But as soon as I switch to the home automation solution at work, I only get timeout errors

Can someone help me?

Translated with www.DeepL.com/Translator (free version)

ChristianTremblay, 2021-04-18:
It's a network issue probably.
When you are not in the same subnet than the BACnet device, BAC0 won't see it. When you are away, on another subnet, you can register as a foreign device by targetting the BACnet router 

```
bacnet = BAC0.lite(ip='IPAddressOfYourPCOntheVPNNIC', 'bbmdAddress='IPofRemoteRouter:47808', bbmdTTL=900) 
```

KostBMF, 2021-04-19:
Thanks for your reply.
But i'm still in the same network. So, i don't understand where is the problem.

ChristianTremblay, 2021-04-19:
Can you detail how your network is done ?
Both the BAC0 script and the BACnet router (as you seem to connect to a MSTP device 12:1), must be on the same subnet.
If not, BBMD must be used.

closed 2021-04-26 by KostBMF