ChristianTremblay/BAC0 #272: DevicePoll Looping

aviola7x7, 2021-04-16:
I've found an instance where the TaskManager.py appears to be getting stuck on a particular device when running DevicePoll. The standard process of connecting to the BACnet network, running discovery, and defining controllers goes without issue. The problem only appears once the polling interval (900s in this case but I've tried other values) occurs. As TaskManager.py goes through the items under `_tasks` in order, each finishes successfully before moving to the next. However, when this particular device is reached, the tasks never reaches completion and instead appears to loop indefinitely. The devices which appear after this one in `_tasks` never get polled.

Where the previous items under `_tasks` have their `count` value at one and then the next one begins, this device's `count` value increase indefinitely. The `average_latency` value continues to update and only increases. The `last_time` value also updates accordingly. However, `next_time`, `execution_time`, and `average_execution_delay` never change with `execution_time` and `average_execution_delay` remaining zero. 

I've attempted to trace the process that the tasks are completed and _think_ I've narrowed it down. In `TaskManager.py`, under `def execute(self)` I see that first the `count` is increased then the `average_latency` is recalculated. There is a section of `if`, `elif`, and `else` after which the `average_execution_delay` and `execution_time` are calculated. Because these two variables are never updated but the ones prior are, I'm assuming the issue lies somewhere in the if/elif/else section that is causes it to loop back to the beginning of the task and restart indefinitely. 

I've included some information on the device itself:

Device Type: Trane SC+
`len()` of `points`: 507
`segmentation_supported: True`
poll delay is 900s

At first I considered if it was an issue with the number of points but the device immediately preceding this one completes successfully despite having 2,253 points. (I also attempted removing this larger device to see if there was some issue it was causing with devices following but it made no difference).  

I'm going to try to better understand what's happening in the below section:

if self.fn and self.args is not None:
            self.fn(self.args)
        elif self.fn:
            self.fn()
        else:
            if self._kwargs is not None:
                self.task(self._kwargs)
            else:
                self.task()

That's my current best guess of where the problem lies. It's particularly off because this is the only device I've seen that has had this problem. I'll report back with anything I find.

Thank you!

ChristianTremblay, 2021-05-07:
Any update on this ?

aviola7x7, 2021-07-27:
Sorry for taking so obscenely long but I think I've finally identified the issue and put in a quick fix to bypass it for the time being. Ultimately, I was able to go through the list of points that were being updated via the poll task and find the first one which wasn't being updated. I found that instead of there being a value for the point, there was an error code string in its place. Below is what was there:

`enumValue:'Traceback (most recent call last):\n  File "/home/pi/.local/lib/python3.7/site-packages/BAC0/core/devices/Points.py", line 913, in enumValue\n    if ":" in self.lastValue:\nTypeError: argument of type \'int\' is not iterable\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n  File "/home/pi/.vscode/extensions/ms-python.python-2021.5.926500501/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_resolver.py", line 193, in _get_py_dictionary\n    attr = getattr(var, name)\n  File "/home/pi/.local/lib/python3.7/site-packages/BAC0/core/devices/Points.py", line 918, in enumValue\n    return self.get_state(self.lastValue)\n  File "/home/pi/.local/lib/python3.7/site-packages/BAC0/core/devices/Points.py", line 903, in get_state\n    return self.properties.units_state[v - 1]\nIndexError: list index out of range\n'` 

In Points.py, I wasn't able to determine what was actually causing the out of range error but I was able to tweak the try/except statement from this:

`def get_state(self, v):
        try:
            return self.properties.units_state[v - 1]
        except TypeError:
            return "n/a"`

to this:

`def get_state(self, v):
        try:
            return self.properties.units_state[v - 1]
        except:
            return "n/a"`

I know using a catch all `except` isn't exactly a good fix but it did allow the polling task to work properly and all of the histories to update properly.

ChristianTremblay, 2021-07-27:
Or we could try 
```
try:
    [bla]
except (TypeError, IndexError):
    return "n/a"
```

closed 2021-10-03 by github-actions[bot]