ChristianTremblay/BAC0 #273: APDU Abort Reason 'unrecognizedService' when requesting CreateObject service

mosterme1003, 2021-04-28:
Hey! I want to create objects on an existing device in the network. With your request template in mind I tried to figure out a way to do it. I have the following code:

```python
from bacpypes.core import deferred
from bacpypes.pdu import Address
from BAC0.core.io.Read import find_reason
from BAC0.core.io.IOExceptions import NoResponseFromController, WritePropertyException
from bacpypes.apdu import SimpleAckPDU, CreateObjectRequestObjectSpecifier, CreateObjectRequest
from bacpypes.iocb import IOCB

import BAC0

my_ip = "192.168.19.176"
bacnet = BAC0.connect(ip=my_ip)
deviceIp= '192.168.19.146'

createObjectRequestObjectSpecifier = CreateObjectRequestObjectSpecifier(
            objectType="analogInput",
            objectIdentifier=("analogInput", int(123))
        )
request = CreateObjectRequest(
            objectSpecifier=createObjectRequestObjectSpecifier,
        )

request.pduDestination = Address(deviceIp)

try:
    iocb = IOCB(request)
    iocb.set_timeout(10)
    deferred(bacnet.this_application.request_io, iocb)
    iocb.wait()
    if iocb.ioResponse:
        apdu = iocb.ioResponse
        if not isinstance(apdu, SimpleAckPDU):
            print("Not an ack, see debug for more infos.")
    if iocb.ioError:
        apdu = iocb.ioError
        reason = find_reason(apdu)
        raise NoResponseFromController(
            "APDU Abort Reason : {}".format(reason))
except WritePropertyException as error:
    print(("exception: {!r}".format(error)))
```

And I get the following error. Do you know why this request is unrecognized? There is no problem with other requests like ReadPropertyRequest or WritePropertyRequest. Did I do something wrong? Thank you very much!

```
Traceback (most recent call last):
  File "E:/main.py", line 44, in <module>
    "APDU Abort Reason : {}".format(reason))
BAC0.core.io.IOExceptions.NoResponseFromController: APDU Abort Reason : unrecognizedService
```

ChristianTremblay, 2021-05-07:
The device is informing you that this service is not implemented.
Not really surprised... in fact, I never saw any BACnet device on the field on which we could create object dynamically.

mosterme1003, 2021-05-17:
Well I think I overlooked the fact that the CreateObject service is for creating objects on the serving device and not remotely. I created a BACnet object on a device in the field using another software and wireshark recorded a package with the UnconfirmedPrivateTransfer service. From what I understand I think this dynamic object creation is based on proprietary services.

mosterme1003, 2021-09-14:
I actually don't know if the device I was testing before was one of the few that doesn't support this service or if I had a mistake in my code. Joel Bender helped me with this in https://github.com/JoelBender/bacpypes/issues/426 and it seems to be relatively simple to create an object on a device in the field. I tested it now on some devices from different vendors and most of them would let me create an object.

ChristianTremblay, 2021-09-14:
Good news even if I'm surprised... which is a good thing as it points that I'm too enclosed in the "this is the way it is since always" kinda thing. 

What is your use case ? 

What are the controller you are using ?

Once the object is created in the remote object, does it live there un connected to the sequence ? How does it work ?

Thanks for the info

mosterme1003, 2021-09-14:
My use case is that I have many controllers in a network that are controlling different facilities on different properties. For now I just wanted to be able to automatically create and configure `trendLog` objects on these devices. Manually it would take me much longer especially because we need to define the `objectName` and `description` of the `trendLog` in a certain way. 

The controllers I am using are building automation DDCs from different vendors like SysCom, Siemens, Kieback&Peter, Sauter, Johnson Controls, etc..

I mean it's not really a remote object. The way I understand it is that I just trigger the `CreateObject` service on the device (remotely) as long as this service is implemented. Most of our controllers actually happen to support it though. After the device created the object, you can find its identifier in the device's `objectList`. So I think it will be there permanently.

Please correct me if I'm wrong with something. I hope this information helps you.

ChristianTremblay, 2021-09-14:
I never thought adding TrendLogs on the fly....this is genious ! I want that ! :-)

mosterme1003, 2021-09-28:
Thank you! I'm sure that this will save us a lot of time in the future 👍 

As we're talking about `trendLogs`.. is there a way to interact with `trendLogMultiple` objects? I want to automatically create one and merge the previously created `trendLogs` into it but it says that it's an unsupported object type when I try to read a property. Does this mean that this object type is not supported by BAC0 and/or Bacpypes?

```py
bacnet = BAC0.connect(ip=localIp)
mtDescription = bacnet.read("192.168.19.80 trendLogMultiple 106 description")
```

Output:
```
Traceback (most recent call last):
  File "E:/.../testing/testing.py", line 21, in <module>
    mtDescription = bacnet.read("192.168.19.80 trendLogMultiple 106 description")
  File "E:\Python\lib\site-packages\BAC0\core\io\Read.py", line 217, in read
    "APDU Abort Reason : {}".format(reason)
BAC0.core.io.IOExceptions.NoResponseFromController: APDU Abort Reason : unsupportedObjectType
```

ChristianTremblay, 2021-12-03:
keeping it here

closed 2022-02-09 by github-actions[bot]