ChristianTremblay/BAC0 #275: Unable to read 'objectList' from device object

PouriaCh, 2021-05-12:
Hi Christian,
I have been struggling with some cases of reading the objectList of 'some' devices. I receive a timeout in return and I can't read the points. This only happens with some random devices. The weird part is that if I happen to know/guess one object of that device (say analogValue 12), I am able to read its presentValue. 
I have been dealing with couple of these issues. Would appreciate it if you could throw any thoughts here. Thanks.

Safrone, 2021-05-28:
I've had some issues when devices have a lot of points. Does it work when you specifically ask for `objectList 0` to get the point count? after that you can get the objects one by one with `objectList 1,` `objectList 2`, etc. you can combine X queries with a readMultiple where X will depend on the device

ChristianTremblay, 2021-05-28:
Thanks @Safrone, you just made me erase what I was writing :-)

PouriaCh, 2021-06-10:
Thanks, @Safrone. 

I actually considered this as well, but the problem is that the device does not respond (either time out, or no response from device) when the property we are reading is 'objectList'. I tried `objectList 0`, 'objectList 1', ... but none of them work.

If I happen to know a point (like AV 10), I can read the presentValue by `bacnet.read('192.168.1.100 analogValue 120 presentValue')`. But 'objectList' does not work.

ChristianTremblay, 2021-06-11:
Look here : BAC0/core/devices/mixins/read_mixin.py, line 133

closed 2021-08-19 by github-actions[bot]