ChristianTremblay/BAC0 #279: character_string

bbartling, 2021-06-15:
Hey Christian,

If I am trying to make some strings read from a json file as BACnet read only points from a simple app, can I get a tip on how to make the data come through?

```
import BAC0

from BAC0.core.devices.local.models import (
    analog_output,
    analog_value,
    binary_value,
    character_string
    )

from BAC0.tasks.RecurringTask import RecurringTask 
from bacpypes.primitivedata import Real

import json
import logging

logging.basicConfig(filename='bacnet_app.log', level=logging.ERROR)

# create string point
_new_objects = character_string(
        name="signal_duration",
        description="Signal Duration",
        presentValue="default sig duration string"
    )

# create string point
_new_objects = character_string(
        name="signal_start",
        description="Signal Start Time",
        presentValue="default sig start time string"

    )

STATIC_BACNET_IP = '192.168.0.105/24'
DEVICE_ID = '33333'
bacnet = BAC0.lite(ip=STATIC_BACNET_IP,deviceId=DEVICE_ID)
_new_objects.add_objects_to_application(bacnet)
bacnet._log.info("APP Created Success!")

def updater():
        
    with open('event_info.json', 'r') as fp:
        event_info_read = json.load(fp)
    print(f"EVENT INFO {event_info_read}")

    sig_dur = bacnet.this_application.get_object_name(f'signal_duration')
    new_sig_dur = event_info_read['duration']
    print(f"Signal Duration info {new_sig_dur}")
    sig_dur.presentValue = new_sig_dur

    sig_start = bacnet.this_application.get_object_name(f'signal_start')
    new_sig_start = event_info_read['dtstart']
    print(f"Signal Start info {new_sig_start}")
    sig_start.presentValue = new_sig_start

def main():
    task1 = RecurringTask(updater,delay=10)
    task1.start()

    
    while True:
        pass
        
        
if __name__ == '__main__':
    logging.info("Starting main loop")
    try:
        main()
    except Exception as e:
        logging.error("Unexpected error trying to run main loop:" + str(e))
```

Curious if I am missing something, at least I can remember using the `from bacpypes.primitivedata import Real` on floats or ints, would I need this for strings too?

What I notice when using a BACnet scanner tool on the app is I can discover the app and points but usuallyI can drill down a notch further can see the present value or even priority array if its an output. I cant drill down any further on these string points, any ideas to try?

[image]

ChristianTremblay, 2021-07-07:
please try : 
```
from bacpypes.primitivedata import CharacterString

# and cast the `presentValue`
sig_start.presentValue = CharacterString(new_sig_start)
```

Tell me if it helps

bbartling, 2021-07-17:
Still same issues, anything look odd?

```
from bacpypes.primitivedata import CharacterString
import BAC0

from BAC0.core.devices.local.models import (
    analog_output,
    analog_value,
    binary_value,
    character_string
    )

from BAC0.tasks.RecurringTask import RecurringTask
from bacpypes.primitivedata import Real

import json
import logging

logging.basicConfig(filename='bacnet_app.log', level=logging.ERROR)

# create string point
_new_objects = character_string(
        name="signal_duration",
        description="Signal Duration",
        presentValue="default sig duration string"
    )

# create string point
_new_objects = character_string(
        name="signal_start",
        description="Signal Start Time",
        presentValue="default sig start time string"

    )

STATIC_BACNET_IP = '192.168.0.105/24'
DEVICE_ID = '33333'
bacnet = BAC0.lite(ip=STATIC_BACNET_IP,deviceId=DEVICE_ID)
_new_objects.add_objects_to_application(bacnet)
bacnet._log.info("APP Created Success!")

def updater():

    with open('event_info.json', 'r') as fp:
        event_info_read = json.load(fp)
    print(f"EVENT INFO {event_info_read}")

    sig_dur = bacnet.this_application.get_object_name(f'signal_duration')
    new_sig_dur = event_info_read['duration']
    print(f"Signal Duration info {new_sig_dur}")
    sig_dur.presentValue = new_sig_dur

    sig_start = bacnet.this_application.get_object_name(f'signal_start')
    new_sig_start = event_info_read['dtstart']
    print(f"Signal Start info {new_sig_start}")

    # and cast the `presentValue`
    sig_start.presentValue = CharacterString(new_sig_start)

def main():
    task1 = RecurringTask(updater,delay=10)
    task1.start()

    while True:
        pass

if __name__ == '__main__':
    logging.info("Starting main loop")
    try:
        main()
    except Exception as e:
        logging.error("Unexpected error trying to run main loop:" + str(e))
```

ChristianTremblay, 2021-08-03:
@bbartling any update on that ?

bbartling, 2021-08-03:
Hey Christian, sorry I meant to say when I posted the code snip same issues with importing bacpypes directly. 

Is there anything odd in the code snip above I am doing wrong with trying to create bacnet discoverable string messages?

Also the code has a json read object, just fyi.

ChristianTremblay, 2021-08-03:
I made it work here and another instance of BAC0 was able to read the values.
The `character_string` model do not add priority array by default. You would need to specify `is_commandable` to True to add that.

That said, I'm not sure I answer the question...

bbartling, 2021-08-03:
Maybe its something odd with the Contemporary Controls BACnet tool. I can discover the BAC0 app and points with the Contemporary Controls tool but no present values come through.

Would you have a JACE or another tool to verify the point present values come through?

bbartling, 2021-08-04:
For some reason I think I originally noticed this issue trying to discover a BAC0 app in VOLTTRON that uses bacpypes under the hood. Same issue as the contemporary controls BACnet discovery tool, I could discover the BAC0 Lite app, the points, but no data comes forth for point present values `strings`.

idk, there could also be a large potential where there is high likely hood in the upper 97.5 percentile that this is all operator user error too.