ChristianTremblay/BAC0 #280: BAC0 on multithreading got stuck on disconnect

aDramaQueen, 2021-07-03:
Hi,
I already spammed the gitter chanel, but now I'm more sure this is a BAC0 problem.
I have strange behavior of BAC0. To be frank, I'm not quiet sure if it's a BAC0, a BACpypes or a network problem.
Quick introduction:
We want to read out a few devices in our university via BACnet. Since we are not the only ones, I wanted to keep the connection as short as possible. So I have now a separate thread per BAC controller, which should make a readout every 60 seconds.
For this readout it connects to the controller makes the readout and terminates the connection afterwards.
For the different controllers I open a BAC connection on the host side, where I simply count up the ports from 47808.
Currently I have exactly 2 devices. The problem is that the program only reads 1 device at a time. The other is also read a few times, but then it just stops. No exception, no error nothing..., the thread is still alive but it just doesn't do anything anymore.
This is the main function:
```
def __read_out__(self):
        self.__clockTrigger__.wait()
        try:
            print(f'CONNECTING TO BACNET ON PORT {self.__host_port__}...')
            self.__bacnet__ = BAC0.lite(ip=self.__host_ip__, port=self.__host_port__, bbmdAddress=self.__bbmd_address__, bbmdTTL=bacnet_broadcast_management_device_TTL_in_sec)
            print(f'CONNECTING TO BACNET ON PORT {self.__host_port__} SUCCESSFULLY ESTABLISHED')
        except Exception as exc:
            mainLogger.error(exc)
            if self.stopTrigger.is_set():
                return
            else:
                self.__read_out__()
        self.__time_stamp__ = self.__get_time__(datetime.now(timezone('UTC')))
        self.__answer__ = self.__bacnet__.readMultiple(self.__bbmd_address__, self.__request__)
        print(f'DISCONNECT ON PORT {self.__host_port__}...')
        self.__bacnet__.disconnect()
        print(f'DISCONNECTED SUCCESSFULLY ON PORT {self.__host_port__}')
        # self.__disassemble_answer__()
        # self.__send_data_to_db__()
        if self.stopTrigger.is_set():
            return
        else:
            self.__read_out__()
```
**The terminal output is this:**
CONNECTING TO BACNET ON PORT 47809...
2021-07-03 11:52:13,129 - INFO | Starting BAC0 version 21.02.25 (Lite)
2021-07-03 11:52:13,129 - INFO | Use BAC0.log_level to adjust verbosity of the app.
2021-07-03 11:52:13,129 - INFO | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2021-07-03 11:52:13,129 - INFO | Starting TaskManager
CONNECTING TO BACNET ON PORT 47808...
2021-07-03 11:52:13,129 - INFO | Starting BAC0 version 21.02.25 (Lite)
2021-07-03 11:52:13,129 - INFO | Use BAC0.log_level to adjust verbosity of the app.
2021-07-03 11:52:13,129 - INFO | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2021-07-03 11:52:13,129 - INFO | Using ip : 10.200.0.3
2021-07-03 11:52:13,129 - INFO | Using ip : 10.200.0.3:47809
2021-07-03 11:52:13,129 - INFO | Starting app...
2021-07-03 11:52:13,139 - INFO | Starting app...
2021-07-03 11:52:13,139 - INFO | BAC0 started
2021-07-03 11:52:13,139 - INFO | Registered as Simple BACnet/IP App
2021-07-03 11:52:13,139 - INFO | BAC0 started
2021-07-03 11:52:13,139 - INFO | Registered as Simple BACnet/IP App
2021-07-03 11:52:13,169 - INFO | Update Local COV Task started
CONNECTING TO BACNET ON PORT 47808 SUCCESSFULLY ESTABLISHED
2021-07-03 11:52:13,169 - INFO | Update Local COV Task started
CONNECTING TO BACNET ON PORT 47809 SUCCESSFULLY ESTABLISHED
DISCONNECT ON PORT 47808...
2021-07-03 11:52:13,229 - INFO | Stopping all tasks
2021-07-03 11:52:13,250 - INFO | Stopping TaskManager
DISCONNECT ON PORT 47809...
2021-07-03 11:52:13,254 - INFO | Stopping all tasks
2021-07-03 11:52:13,254 - INFO | Ok all tasks stopped
2021-07-03 11:52:13,269 - INFO | Ok all tasks stopped
2021-07-03 11:52:13,269 - INFO | BACnet stopped
DISCONNECTED SUCCESSFULLY ON PORT 47809
CONNECTING TO BACNET ON PORT 47809...
2021-07-03 11:52:18,150 - INFO | Starting BAC0 version 21.02.25 (Lite)
2021-07-03 11:52:18,150 - INFO | Use BAC0.log_level to adjust verbosity of the app.
2021-07-03 11:52:18,150 - INFO | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2021-07-03 11:52:18,150 - INFO | Starting TaskManager
2021-07-03 11:52:18,150 - INFO | Using ip : 10.200.0.3:47809
2021-07-03 11:52:18,154 - INFO | Starting app...
2021-07-03 11:52:18,157 - INFO | BAC0 started
2021-07-03 11:52:18,157 - INFO | Registered as Simple BACnet/IP App
2021-07-03 11:52:18,189 - INFO | Update Local COV Task started
CONNECTING TO BACNET ON PORT 47809 SUCCESSFULLY ESTABLISHED
DISCONNECT ON PORT 47809...
2021-07-03 11:52:18,280 - INFO | Stopping all tasks
2021-07-03 11:52:19,189 - INFO | Stopping TaskManager
2021-07-03 11:52:19,189 - INFO | Ok all tasks stopped
2021-07-03 11:52:19,209 - INFO | BACnet stopped
DISCONNECTED SUCCESSFULLY ON PORT 47809
CONNECTING TO BACNET ON PORT 47809...
2021-07-03 11:52:23,169 - INFO | Starting BAC0 version 21.02.25 (Lite)
2021-07-03 11:52:23,169 - INFO | Use BAC0.log_level to adjust verbosity of the app.
2021-07-03 11:52:23,169 - INFO | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2021-07-03 11:52:23,169 - INFO | Starting TaskManager
2021-07-03 11:52:23,169 - INFO | Using ip : 10.200.0.3:47809
2021-07-03 11:52:23,169 - INFO | Starting app...
2021-07-03 11:52:23,169 - INFO | BAC0 started
2021-07-03 11:52:23,169 - INFO | Registered as Simple BACnet/IP App
2021-07-03 11:52:23,209 - INFO | Update Local COV Task started
CONNECTING TO BACNET ON PORT 47809 SUCCESSFULLY ESTABLISHED
DISCONNECT ON PORT 47809...
2021-07-03 11:52:23,289 - INFO | Stopping all tasks
2021-07-03 11:52:24,209 - INFO | Stopping TaskManager
2021-07-03 11:52:24,229 - INFO | Ok all tasks stopped
2021-07-03 11:52:24,249 - INFO | BACnet stopped
DISCONNECTED SUCCESSFULLY ON PORT 47809
CONNECTING TO BACNET ON PORT 47809...
2021-07-03 11:52:28,189 - INFO | Starting BAC0 version 21.02.25 (Lite)
2021-07-03 11:52:28,189 - INFO | Use BAC0.log_level to adjust verbosity of the app.
2021-07-03 11:52:28,189 - INFO | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2021-07-03 11:52:28

As you can see port 47808 works the first time, but then it just stops and only port 47809 continues.

I twiddled a little bit with the source code & narrowed it down to following part:

[image]

**The Terminal output is following:**
.
.
.
CONNECTING TO BACNET ON PORT 47809...
CONNECTING TO BACNET ON PORT 47808...
CONNECTING TO BACNET ON PORT 47808 SUCCESSFULLY ESTABLISHED
CONNECTING TO BACNET ON PORT 47809 SUCCESSFULLY ESTABLISHED
DISCONNECT ON PORT 47809...
DISCONNECT ON PORT 47808...
WAIT FOR THREAD...
WAIT FOR THREAD...
CONTINUE
DISCONNECTED SUCCESSFULLY ON PORT 47809

Process finished with exit code -1

Soo..., the app thread seems not to terminate.

ChristianTremblay, 2021-07-07:
I'm not sure why you keep connecting then disconnecting. Being a BACnet device itself, BAC0 should not appear and disappear constantly.

Once you are connected, BAC0 will bind UDP port 47808 on the network interface but not so many traffic will be created by that.
I would keep the thing running, then create a task to read the points needed. For only 2 devices, with reading every 60 seconds... no need for 2 threads.

I know there is something with the task manager thread that will sometimes hang in somewhere, but I never found a root cause. It's intermittent and pretty hard to troubleshoot. In your particular case, I would think about some conflict caused by using two instances at the same time, with two ports... maybe.

If I would make a system like yours, I would : 

- start BAC0 once
- create a custom object list of the devices I need to monitor (just the points I need)
- create a BAC0.device for each will a polling delay of 60 and use the custom_object_list from above
- optional history max size limit (if the script will work for a long time, it's a good idea to limit this...)
- each point will have historical data in the form of `pandas.Series` (length = max history size)
- build a task to write the the DB using the `device['point'].lastValue` (this way, you will use the last value read in the point history and won't create another read on the network. (see. `ReccuringTask`)

That kind of script can run as a service

aDramaQueen, 2021-07-07:
Ah ok, I thought you were using TCP not UDP for the connection. That's why I closed the connection again and again, so that other people besides us can still use the device (except for the short time of readout it self).

Just to clarify: The waiting time will always be 60 seconds, regardless of the duration of the actual readout itself. Or does it do the readout and then waits 60 seconds?

ChristianTremblay, 2021-07-08:
The taskManager in BAC0 is very simple. It calls the function, wait for the response, then wait 60sec. You could have to play with the delay if it's required. But I don't think a few seconds will cause any issue... 
You could also use COV and some important point would be notified (and saved) everytime they change. 

Let me know if I can help

aDramaQueen, 2021-07-22:
OK...,
we tested it the last 2 weeks & it's looking good. I followed your advice and stoped opening & closing BAC0. Since then it runs.
About the timing: Personally, I don't really care, but the people for whom I write the program want to have relatively accurate reading points. The announcement was: "Every 60 seconds". I currently let then run a thread in parallel that simply counts to 60. So I get around the readout delay of BAC0.
Thank you very much good sir.

closed 2021-07-22 by aDramaQueen

ChristianTremblay, 2021-07-23:
@aDramaQueen 
Not sure I get what you (they) mean by relatively accurate. The timestamp you get is the exact moment where the reading was taken. As we rely on the network itself, all the reading will not be taken at the same moment. 

In the context of BACnet devices where we deal with temperatures and other stuff with an inertia... a couple ms (even sec) won't do any difference and technically, the timeseries analysis will implement some kind of aggregation to extract knowledge out of that.

The default polling value is 10sec in BAC0 and specifying a `poll=60` would set this delay to 60 seconds.

You can find the code for the Device Polling method here : 
https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/tasks/Poll.py#L56

It's nothing more than a "Task"
https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/tasks/TaskManager.py#L126

Let me know if you have idea to improve that