ChristianTremblay/BAC0 #281: Modifying Schedule Present Value

angelenriquecano, 2021-07-07:
Hi!
I'm developing a script to turn on/off a BACnet device using the schedule's present value property. To modify the schedule's present value, the device has to be out of service by writing a True to schedule's outOfService property.
After that, when I try to write schedule present value I get a TypeError: invalid constructor datatype.
Here I send you a piece of the code I'm testing:

```
import BAC0
from bacpypes.primitivedata import Atomic, Enumerated

BAC0.log_level('silence')
bacnet = BAC0.connect()

device = BAC0.device(address='1:0x0109', device_id=2098185, network=bacnet, segmentation_supported=False)

out_of_service = bacnet.read('1:0x0109 schedule 14 outOfService')
val = Enumerated()
val.value = 0

if out_of_service:
    device.write_property(('schedule', 14, 'presentValue'), value=val, priority=5)
else:
    device.write_property(('schedule', 14, 'outOfService'), value=True, priority=5)
    device.write_property(('schedule', 14, 'presentValue'), value=val, priority=5)
```

Has anyone has a problem like this? Can anybody give some information to know how to do it?
Thanks in advance!
Best regards, Angel

ChristianTremblay, 2021-07-18:
Hi !
Playing with out_of_service will disconnect the logic from the sequence and tell the device to "believe" the value is X or Y. If you are interacting on a device which is integrated to another system, it could bring some issues as other people will maybe not see that the point is actually out of service  and something else is playing with it...

That said, you could use :

```
# Set to True
bacet.out_of_service("1:0x109 schedule 14")
# Release
bacet.release("1:0x109 schedule 14")
```

I think you could also use
```
# Set out of service and write value
bacnet.sim("1:0x109 schedule 14 {}".format(val))
# Release
bacet.release("1:0x109 schedule 14")
```

ChristianTremblay, 2021-07-18:
Another way would be writing a new schedule.

https://bac0.readthedocs.io/en/latest/schedules.html?highlight=schedule

closed 2021-10-31 by github-actions[bot]