ChristianTremblay/BAC0 #283: Unable to communicate with devices behind gateway

KonstantinMasich, 2021-07-13:
Hello @ChristianTremblay 

Thank you for your work and support for BAC0, it's extremely helpful :)

I've encountered a strange issue: cannot establish BAC0 communication over Ethernet to devices behind BACnet gateway. The gateway itself is visible in BAC0 and I can communicate with it. However, BACnet devices (connected to this gateway) are not reachable via BAC0, although they are perfectly reachable from applications like YABE ("Yet Another Bacnet Explorer").

As you can see on the screenshot below, YABE is able to detect the 4 devices behind the gateway, as well as the gateway itself:

[image]

The 4 devices behind the gateway are sitting on the same IP address, as they are virtual devices. Nonetheless, communication with them should be possible, just as with any other BACnet device; it is possible using YABE.

However, scanning it with BAC0 brings up only the gateway:

```python
import BAC0
bacnet = BAC0.lite("192.168.172.3/24")
bacnet.whois()
# The result is [('192.168.172.2', 990064)] , which is a gateway
```

I've tried to use the `global_broadcast=True` option to detect devices in the other subnets, and I've got the following output in `stderr`:

[image]

Note that "path error" was printed exactly 4 times - that coincides with the 4 devices connected to BACnet gateway. 

I'm attaching a ZIP archive with two Wireshark captures - one for "whoise" with `global_broadcast=True`, and the other without that option.

[wireshark_captures.zip](https://github.com/ChristianTremblay/BAC0/files/6807968/wireshark_captures.zip)

ChristianTremblay, 2021-07-28:
Hi, this is very weird and I think we need to dig into bacpypes for that. I never saw that kind of issue.

@joelbender : have you ever saw something like this ?

JoelBender, 2021-07-28:
That thing that almost looks like a socket tuple isn't, socket numbers can't get that high, so there is some kind of address encoding/decoding error coming out of the network layer (SNET/SLEN/SADR) or it's getting mangled upstream.  I'll dig into these Wireshark captures and let you know what I can see.

JoelBender, 2021-07-29:
This device has a broken implementation that doesn't follow the BACnet standard, there is more than one device at the same BACnet address, so the careful "device address binding" relationship.  I was expecting it to present itself as an IPv4-to-virtual-network router by the way YABE is presenting those devices hanging off a little network, but that's deceiving.

Any BACnet application that issues more than one confirmed service (like a Read Property Request) to more than one of these virtual devices at a time could break because the returned ComplexACK will not have anything that distinguishes what device it came from.  Most BACnet applications do not reuse an invokeID, so there is some safety there, BACpypes will make sure that the invokeID doesn't collide with a current request to a device, but there's a chance it could.

Furthermore, when reading the value of an object like **analogValue:1** you will not know which device that is associated with, and I suspect that they way they keep their objects distinct is by enforcing that two virtual devices will not both have an analogValue:1.  You can test this by reading the **objectList** property from each device using the BACpypes **ReadProperty.py** sample application (like `read 192.168.172.2 device:990064 objectList`) and compare the results to see if they have any overlaps.

Humph.

ChristianTremblay, 2021-08-03:
We consider this an issue from the device. Closing the issue

closed 2021-08-03 by ChristianTremblay