ChristianTremblay/BAC0 #287: Passing of deviceId to "Base" class

igloodan, 2021-07-23:
Hello everyone,

I can see in the code that when calling `BAC0.lite()`, I can pass a variable number of arguments via `**params` but I don't think these are being used by `Base.__init__()`.

I'm looking to pass an argument in for `deviceId` so that I can specify what is used for communication on the network.

BAC0.lite() doesn't have a `deviceId` argument, just `**params`.
These `**params` are then passed to `Base.__init__()` but (my knowledge here is limited) this isn't parsed as `Base.__init__()` is awaiting a named argument called `deviceId` and doesn't specifically check the passed dict.

Is my understanding here correct, is it possible to specify a device ID with the code as is?

Thanks.

ChristianTremblay, 2021-07-23:
ex. : device_app = BAC0.lite(deviceId=101)

igloodan, 2021-07-23:
@ChristianTremblay 
Thanks for your quick response!
My apologies, I was trying this in the `BAC0.device()` function, not the `BAC0.lite()` function.

closed 2021-07-23 by igloodan