ChristianTremblay/BAC0 #289: Log Level

igloodan, 2021-07-27:
Hello,

I'm having a lot of trouble supressing the logs from BAC0, I've got the below in my program but it's still generating a long entry every 5 seconds when checking for COV events.

Program:
```
logging.getLogger("BAC0_Root.BAC0.scripts.Base.Base").disabled = True
BAC0.log_level("silence")
```

Log Output (Every 5 seconds):
```
27-Jul-21 18:21:40 - DEBUG - BAC0_Root.BAC0.tasks.UpdateCOV.Update_local_COV - Executing : update_local_cov_Update Local COV Task
27-Jul-21 18:21:40 - DEBUG - BAC0_Root.BAC0.tasks.TaskManager.Manager - Task 1907116432 | update_local_cov_Update Local COV Task executed. update_local_cov_Update Local COV Task   | Avg exec delay : 5.00 sec | Avg latency : 0.00 sec | last executed : 2021-07-27 18:21:40 | Next Time : 2021-07-27 18:21:45
```

I've been working through the `BAC0/core/utils/notes.py` file for hours but I just can't work out why this particular event would not be affected by the change to the log level.
Can anyone help?

Thanks.

igloodan, 2021-07-27:
I've managed to suppress these using the below...
`logging.getLogger("BAC0_Root.BAC0.tasks").propagate = False`

More research is required on my part to understand logging (especially with third party libraries).
I suspect that it may be something to do with the "lastResort" option of the logging hierarchy!

igloodan, 2021-07-27:
When some BACnet traffic actually starts to move on the network I get more logging activity, I've updated the above line to...
`logging.getLogger("BAC0_Root.BAC0").propagate = False`

ChristianTremblay, 2021-07-27:
Sorry about that.... logging is not an easy thing and I admit this could have been improved. I don't use DEBUG often so I haven't dealt with all the caveat.

igloodan, 2021-07-27:
@ChristianTremblay
I've done some research and I think I have a solution, I'll pull something together and send over a pill request. It may involve some large changes to the existing note_and_log decorator though.

igloodan, 2021-07-28:
I haven't managed to track this down yet but if I run the below...
```
    _log = logging.getLogger()
    logging.basicConfig(datefmt='%d-%b-%y %H:%M:%S',
                        format='%(asctime)s - %(levelname)s - %(name)s - %(message)s',
                        level=logging.INFO
                        )

    _bac0log = logging.getLogger("BAC0.core.utils.notes")
    print(_bac0log.getEffectiveLevel())
```
then it returns...
`10` which is DEBUG.

Most resources seem go suggest that handlers shouldn't be added to libraries and all logging should  use the `logging.getLogger(__name__)` naming convention which would allow all logs to propogate up the hierarchy and ultimately be handled by the root logger as part of the main application.

BAC0 adds a few handlers and also uses some hard coded logger names, I've tried changing these to `__name__` and adding 
```
_bac0log = logging.getLogger("BAC0.core.utils.notes")
#_bac0log.setLevel(logging.INFO)
```
to my application, which seems to control the log levels correctly. I just can't tell if it's onl the `BAC0.core.utils.notes` file which is getting the `DEBUG` level permanently.

closed 2021-10-04 by github-actions[bot]