ChristianTremblay/BAC0 #291: SQL save | Issue since PR#264

ChristianTremblay, 2021-07-28:
I need to investigate as tests are failing with proposal.

ChristianTremblay, 2021-07-28:
As stated in the commit, BAC0 will now create 2 columns for binary and multistate values. The default column gives the integer value and the supplemental column named with suffix `_str` will give the string interpretation of the value.

This is something I didn't want to loose especially when dealing with multistate values.

kushan-26, 2021-07-28:
This change would work well and allow better handling of multistate values. I have one question though; How will the resampling handle the supplemental column for string values?

ChristianTremblay, 2021-07-28:
It takes the last one. "Mean" would make no sense. 
Can you test it on your side and see if it works for you ?

Also, you can have a look to InfluxDB. I just add support for it and I think that for a lot of cases, this will bring a lot more than SQLite.

https://bac0.readthedocs.io/en/develop/database.html?highlight=influxdb

kushan-26, 2021-07-28:
I will test the update and let you know. Please allow me few days since I'll be travelling this week.

The influxDB support looks intriguing to me. I will definitely look into it. I've been looking into solutions on how to extract historicized data from remote devices.

ChristianTremblay, 2021-08-16:
Nice !

closed 2021-08-25 by ChristianTremblay