ChristianTremblay/BAC0 #297: Resampling Issue When Saving the History to Database

kushan-26, 2021-10-01:
Hello Christian,

I am getting the below errors after having an APDU timeout on a device and then trying to save the device to the database. I checked history and it was not empty. I have the resample frequency set for 15 minutes and had one entry in the History list. However, the save function fails to execute. 

Should there be a condition to execute the save function if the History contains only one entry to resample?

WARNING | APDU Abort Reject Reason : Timeout
WARNING | No response from controller Timeout
ERROR   | Error in resampling device240099/zone_temp_zs_1 : 69.50 degreesFahrenheit | Could not convert  to numeric (probably not enough points)
ERROR   | Super Mega Giga big error Could not convert to numeric. Removing task.

ChristianTremblay, 2021-10-14:
I did a real bunch of work to make BAC0 resilient to network failures in the dev branch. Can you (or did you) try it ?

kushan-26, 2021-10-18:
> I did a real bunch of work to make BAC0 resilient to network failures in the dev branch. Can you (or did you) try it ?

I have not yet tested it. I will perform some tests and update you.

Thanks!

closed 2021-12-26 by github-actions[bot]

kushan-26, 2022-01-12:
@ChristianTremblay,

I've been testing BAC0 resilience against offline devices and it seems to overcome many issues that I've seen in the past. 
However, I noticed that when a device is offline, BAC0 tries to ping the device rapidly. This might cause network failures if there are a lot of devices are offline simultaneously. Is there a way to make sure that the ping task would not bog down the BAS network? I have attached a text file containing ping task frequency for an offline device for reference.

[Offline_Device_Ping_Task.txt](https://github.com/ChristianTremblay/BAC0/files/7856081/Offline_Device_Ping_Task.txt)

ChristianTremblay, 2022-01-13:
Are those ping for 1 device or multiple devices ? I thought my delay of 5 minutes was ok... but if your file show pings for 1 device... I failed.

kushan-26, 2022-01-13:
This was for a single offline device. Is there a scenario where " Avg exec delay : 300.00 sec" would be ignored? I haven't seen this rapid ping for offline devices in previous versions though.

ChristianTremblay, 2022-01-13:
Let's move that to a new Issue...